arendst/Tasmota #13083: ESP32: some BLE commands not saved between restarts

traeu, 2021-09-06:
### PROBLEM DESCRIPTION
I am experimenting with xiaomi bluetooth thermometers and I found out that some commands I tried regarding BLE won't survive reboots.
If I try to assign aliases with BLEAlias command, they are gone after a reboot.
Same with the command MI32Key and MI32Option2.
Other commands I used, like static ip/gw/dns, setting another color theme or changing timezone keep their values between reboots. 
I tried unsuccessfully with the latest versions 9.5.0 and also 9.5.0.7 and 9.5.0.8.

Hardware-Info from Information-Page:
ESP Chip: Id16560268 (ESP32-D0WDQ6)
Flash Size: 4096 kB
Program Size: 1300 kB
Free Program Space: 1856 kB
Free Memory: 79.7 kB (frag. 26%)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 Dev Board
- [x] Tasmota binary firmware version number used: tasmota32-bluetooth.bin 9.5.0, 9.5.0.7, 9.5.0.8
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP_Flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:54:40.566 CMD: Backlog Template; Module; GPIO 255
00:54:40.632 MQT: stat/BLEGW/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:54:40.850 MQT: stat/BLEGW/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:54:41.125 MQT: stat/BLEGW/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
00:56:28.688 CMD: Status 0
00:56:28.705 MQT: stat/BLEXXXX/STATUS = {"Status":{"Module":1,"DeviceName":"BLE XXXXX","FriendlyName":["BLEXXXX"],"Topic":"XXXX","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:56:28.748 MQT: stat/BLEXXXX/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin","RestartReason":"Software reset CPU","Uptime":"0T00:33:35","StartupUTC":"2021-09-06T22:22:53","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-08-27T16:50:23","SaveCount":59}}
00:56:28.781 MQT: stat/BLEXXXX/STATUS2 = {"StatusFWR":{"Version":"9.5.0.8(bluetooth)","BuildDateTime":"2021-09-06T04:08:12","Core":"1_0_7_3","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"407/699"}}
00:56:28.804 MQT: stat/BLEXXXX/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXX",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000002"]}}
00:56:28.835 MQT: stat/BLEXXXX/STATUS4 = {"StatusMEM":{"ProgramSize":1300,"Free":1856,"Heap":77,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004080","00200000","00000024"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,38,50,79","Sensors":"1,2,3,5,6,52,62,127"}}
00:56:28.872 MQT: stat/BLEXXXX/STATUS5 = {"StatusNET":{"Hostname":"XXXX-4236","IPAddress":"192.168.XX.XX","Gateway":"192.168.XX.XX","Subnetmask":"255.255.255.0","DNSServer1":"192.168.XX.XX","DNSServer2":"0.0.0.0","Mac":"24:62:AB:FC:XX:XX","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
00:56:28.901 MQT: stat/BLEXXXX/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.XX.XX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FCB08C","MqttUser":"XXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:56:28.922 MQT: stat/BLEXXXX/STATUS7 = {"StatusTIM":{"UTC":"2021-09-06T22:56:28","Local":"2021-09-07T00:56:28","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:16","Sunset":"20:19"}}
00:56:28.952 MQT: stat/BLEXXXX/STATUS10 = {"StatusSNS":{"Time":"2021-09-07T00:56:28","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
00:56:28.971 MQT: stat/BLEXXXX/STATUS11 = {"StatusSTS":{"Time":"2021-09-07T00:56:28","Uptime":"0T00:33:35","UptimeSec":2015,"Heap":76,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXXXX","BSSId":"60:E3:XX:XX:XX:XX","Channel":6,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
-Set static ip, different tele-interval and different timezone via command line, reboot (with gui button) and verify that changed settings are still there (to verify that changes via command line can survive reboots in general --> works with default settings without problems)
-Enable BLE.
-Put XIAOMI thermometer (or another supported bluetooth-device I guess...) next to the ESP32, check that its values are shown on the status page.
-Set alias with "BLEAlias MACADRESS = ALIAS" command.
-Check on status page that alias is set correctly or verify with "BLEAlias" command that it returns some alias and is not empty.
-Restart with gui button
-Check with "BLEAlias" command or on status-page if settings are still there: BLEAlias is now empty again

The process can also be repeated with MI32Key and MI32Option2 (these are the ones I tried to use).

### EXPECTED BEHAVIOUR
I expected that the changes I made are still there after a reboot.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-09-06:
Hi,

You are right. Those commands don't save its settings to flash as they were designed. For those you need to make a rule to set your desired status at boot time like:

```lua
rule1 1
rule1 on system#boot do BLEAlias BLEAlias fred=1234567890/1 endon
```

If you want, you can make a feature request in Discussions/Ideas and wait for anyone interested on this to make a PR or you can propose a PR if you want.

closed 2021-09-06 by ascillato2

traeu, 2021-09-06:
Thanks for this information!
I don't really need that functionality as long as it forwards the values over mqtt. 
just wanted to report in case it's a bug.

ascillato2, 2021-09-06:
Yes, I understand. Not a bug. Just a feature that the authors of the driver had not implemented. Maybe so as to save flash settings space.

traeu, 2021-09-06:
Last question: Is it known which values are not saved to flash? Only BLE-stuff affected? or every "non-standard" esp32-command that only exists on esp32 but not on normal tasmota?
I guess this behavior is not to save storage, because adding rules for every custom setting will likely eat up more flash...

ascillato, 2021-09-06:
> Is it known which values are not saved to flash?

Yes, it is known. Not all commands save settings to flash. You have to go to the code to check that ( https://github.com/arendst/Tasmota/blob/development/tasmota/settings.h )