arendst/Tasmota #13098: Need help connecting to guest network w/ a captive portal

jschwalbe, 2021-09-09:
This isn't a bug, sorry if I'm doing this wrong.

I would like to connect my switches to a captive portal at work. It's a cisco-based guest network (at my job) that requires a website to be loaded and then click the "I agree" button. I've accomplished it on my desktop w/ curl, but I'm wondering if any such method exists within Tasmota.

Many thanks!

closed 2021-09-09 by close-issue-app[bot]

Jason2866, 2021-09-10:
No, not implemented.

jschwalbe, 2021-09-10:
I was just investigating further into the WebSend command. But that appears to only support http (not https), otherwise I think that would work. Bummer!

Jason2866, 2021-09-11:
You can try with Berry. Compile with support for https (only Berry!).
To enable you have to compile with `USE_WEBCLIENT_HTTPS`

jschwalbe, 2021-09-11:
> You can try with Berry. Compile with support for https (only Berry!).
> To enable you have to compile with `USE_WEBCLIENT_HTTPS`

@Jason2866 Thanks for the idea - and forgive me if I'm misunderstanding something - but it seems like this is for esp32 only, is that correct?

Jason2866, 2021-09-11:
Ahh, yes. So no way.