arendst/Tasmota #13102: Tasmota restarting after toggling on/off

MoLow, 2021-09-10:
### PROBLEM DESCRIPTION
I installed tasmota on a shelly EM (using [this](https://templates.blakadder.com/shelly_EM.html) template), and it sometimes restarts after toggeling on/off state.
I have factory reset it and set only minimal setting (Wifi credentials, template and module) and it continues to reproduce.
more details and logs below

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly EM
- [X] Tasmota binary firmware version number used: 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: https://github.com/arendst/mgos-to-tasmota
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:20:48.889 CMD: Backlog Template; Module; GPIO 255
13:20:48.910 RSL: RESULT = {"NAME":"Shelly EM","GPIO":[0,0,0,0,0,0,0,0,640,3456,608,224,0,1],"FLAG":0,"BASE":18}
13:20:49.136 RSL: RESULT = {"Module":{"0":"Shelly EM"}}
13:20:49.391 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"3456":"ADE7953 IRQ"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:21:48.757 CMD: Backlog Rule1; Rule2; Rule3
13:21:48.778 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:21:49.005 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:21:49.256 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:21:23.340 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1FAEB7","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:21:23.368 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:04:47","StartupUTC":"2021-09-10T12:16:36","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-09-10T13:11:00","SaveCount":23,"SaveAddress":"F5000"}}
13:21:23.399 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"331/699"}}
13:21:23.418 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Atlow_LR",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
13:21:23.448 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1620C2","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:21:23.479 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_1FAEB7-3767","IPAddress":"10.127.16.14","Gateway":"10.127.16.118","Subnetmask":"255.255.255.0","DNSServer":"10.127.16.113","Mac":"98:CD:AC:1F:AE:B7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:21:23.501 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-09-10T12:21:23","Local":"2021-09-10T13:21:23","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:20","Sunset":"19:13"}}
13:21:23.518 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:21:23.535 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-09-10T13:21:23","ENERGY":{"TotalStartTime":"2021-09-10T13:11:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":235,"Current":[0.000,0.000]}}}
13:21:23.560 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-09-10T13:21:23","Uptime":"0T00:04:47","UptimeSec":287,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Atlow_LR","BSSId":"60:38:E0:AB:B8:24","Channel":13,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:21:23.597 RSL: STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4000bf80","00000000","00000000"],"EXCVADDR":"08002818","DEPC":"00000000","CallChain":["40101b73","4023cbf5","40201772","40207a0a","4020a9ff","4024f544","4024f501","4024f5b8","40207eac","402080f4","40286a3c","40240054","402028c2","40101b73","4028d5f1","4022ddfe","4024f500","401018eb","40101b73","4023817c","40101268","4022e674","4024f068","40230ae9","40240000","4021e4fc","40280031","402483d4","4028ce7d","4021e4fc","4021e4fc"]}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:02.006 RSL: INFO1 = {"Info1":{"Module":"Shelly EM","Version":"9.5.0(tasmota)","FallbackTopic":"cmnd/DVES_1FAEB7_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:02.015 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_1FAEB7-3767","IPAddress":"10.127.16.14"}}
00:00:02.035 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["4020176c","00000000","00000000"],"EXCVADDR":"08002818","DEPC":"00000000","CallChain":["402481a3","402079ff","4020a9ff","4024f544","4024f501","4024f5b8","40207eac","402080f4","40286a3c","40240054","402028c2","40101b73","4028d5f1","4022ddfe","4025a0b8","40253000","401018eb","40101b73","4023817c","40101268","4022e674","4024f068","40230ae9","40240000","4021e4fc","40280031","402483d4","4028ce7d","4021e4fc","4021e4fc","40230fc8"]}}}
00:00:02.072 RSL: RESULT = {"POWER":"ON"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure module to Shelly em after configuring Shelly EM template, toggle off/on. might not happen every time but for me it is very often

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The device should not restart :(

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-09-10:
Hi,

Your `RestartReason` is `Exception:0`, so there is a problem in your firmware.

Please, try to update to latest Tasmota from dev Branch: http://ota.tasmota.com/tasmota/tasmota.bin.gz using OTAurl Update, and after that, reset the device to defaults and configure it again so as to start from skratch.

You can do all that by using the WebUI or in the Tasmota console as following:

```lua
OTAurl http://ota.tasmota.com/tasmota/tasmota.bin.gz
upgrade 1
```

After it completes the update in about 2 minutes, you should reset the device. The following command will retain Wi-Fi and MQTT settings.

```lua
reset 6
```

And finally, after that power down the device for 2 seconds and give energy to it again.

MoLow, 2021-09-10:
@ascillato thanks, I have performed that,
but now when turning on/off the same thing happens, and I see this in the logs:
```
14:05:23.039 MQT: tele/tasmota_1FAEB7/INFO1 = {"Info1":{"Module":"Shelly EM","Version":"9.5.0.8(tasmota)","FallbackTopic":"cmnd/DVES_1FAEB7_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:05:23.048 MQT: tele/tasmota_1FAEB7/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-1FAEB7-3767","IPAddress":"10.127.16.14"}}
14:05:23.061 MQT: tele/tasmota_1FAEB7/INFO3 = {"Info3":{"RestartReason":"External System"}}
```

ascillato, 2021-09-10:
Now your `RestartReason` is `External System` that is that the device was left without power.

Do you see any exception when the device boot?

mefisto22, 2021-09-10:
Today my shelly 1 started freezing after toogle. I tried 9.5.0 and newest dev firmware, but the result is same. After downgrading 9.1.0 shelly 1 works fine.

ascillato, 2021-09-10:
@mefisto22 

Hi,

The problem you are explaining is different to this issue. Yours is a Shelly 1 freezing and this issue is about a Shelly EM that have restarts by exception 0.

Please, open a new issue with all the required information about your problem (so as to not asking for your status 0, console logs, etc. and mixing 2 different problems in the same issue) or address this to the Tasmota Support Chat if you prefer. Thanks.

ascillato2, 2021-09-21:
> Do you see any exception when the device boot?

Hi, any news on this?

MoLow, 2021-09-22:
> > Do you see any exception when the device boot?
> 
> Hi, any news on this?

Hi, I have changed the electrical wiring so the shelly device wont be connected directly at the contactor - and it seemed to solve the issue - so it was probably some contact issues.
Thanks for your help!

closed 2021-09-22 by MoLow