arendst/Tasmota #13104: Loss of settings after restart due to mqtt broker being unavailable

Elix-g, 2021-09-10:
### PROBLEM DESCRIPTION
I'm using a self-compiled Tasmota version along with an SSL-secured mosquitto broker. In case the broker isn't available, Tasmota tries to reconnect frequently. After a while (> 1 hour), Tasmota restarts the device. During the startup, there's a console message "Some settings were set back to defaults" and indeed the device name and -configurations are gone.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): JoyIt NODEMCU ESP8266, ObiSocket2
- [X] Tasmota binary firmware version number used: 9.4.0, 9.5.0.,2
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: esptool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"8266","GPIO":[32,3680,1,3648,640,608,0,0,225,227,226,224,1,0],"FLAG":0,"BASE":18}
{"Module":{"18":"Generic"}}
{"GPIO0":{"32":"Button1"},"GPIO1":{"3648":"iBeacon TX"},"GPIO2":{"0":"None"},"GPIO3":{"3680":"iBeacon RX"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"227":"Relay4"},"GPIO14":{"226":"Relay3"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
{"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":273,"Free":238,"Rules":"ON ibeacon#uid DO IF (%value%=_ibeaconid/maj/min/rssi_) var1 %value% ENDIF ENDON ON ibeacon#major DO IF (%value%=000D) var2 %value% ENDIF ENDON ON ibeacon#minor DO IF (%value%=0007) var3 %value% ENDIF ENDON ON ibeacon#rssi DO Backlog var4 %value%;event shoot ENDON "}}
{"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":247,"Free":264,"Rules":"ON event#shoot DO IF (%var1%%var2%%var3%=_ibeaconid/maj/min/rssi_) backlog publish privnet/%topic%/tele/SENSOR5 {\"IBEACON\":{\"UID\":\"%var1%\",\"MAJOR\":\"%var2%\",\"MINOR\":\"%var3%\",\"RSSI\":\"%var4%\"}};var1 \";var2 \";var3 \"; var4 \" ENDIF ENDON"}}
{"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":107,"Free":404,"Rules":"ON tele-bh1750#illuminance DO publish privnet/%topic%/tele/SENSOR1 {\"BH1750\":{\"Illuminance\":%value%}} ENDON"}}
```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
{"Status":{"Module":18,"DeviceName":"Tasmo8266","FriendlyName":["Tasmo8266","Tasmo8266-2","Tasmo8266-3","Tasmo8266-4"],"Topic":"tasmo8266","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
{"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"_OTA URL_","RestartReason":"Software/System restart","Uptime":"3T21:02:02","StartupUTC":"2021-09-06T18:14:15","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-06-12T17:58:08","SaveCount":719,"SaveAddress":"FC000"}}
{"StatusFWR":{"Version":"9.5.0.7(tasmota)","BuildDateTime":"2021.08.27 21:23:16","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"517/699"}}
{"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"_syslog server_","LogPort":514,"SSId":["_SSID_",""],"TelePeriod":120,"Resolution":"558180C0","SetOption":["00008019","2805C8000100060000005A0A000000000000","00000088","00206000","00004000"]}}
{"StatusMEM":{"ProgramSize":797,"Free":1248,"Heap":14,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F8A84C7","04248001","00001CCF","010013C0","44007885","00004084","00200000","00000020"],"Drivers":"1,2,3,4,5,6,8,9,10,16,22,24,29,37,38,50","Sensors":"1,2,3,4,5,6,9,10,52,62"}}
{"StatusNET":{"Hostname":"tasmo-8266","IPAddress":"192.168.179.246","Gateway":"192.168.179.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.179.23","DNSServer2":"192.168.179.24","Mac":"2C:F4:32:78:20:E9","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
{"StatusMQT":{"MqttHost":"_mqtt broker_","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7820E9","MqttUser":"tasmo8266","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
{"StatusTIM":{"UTC":"2021-09-10T15:16:17","Local":"2021-09-10T17:16:17","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:59","Sunset":"19:57"}}
{"StatusSNS":{"Time":"2021-09-10T17:16:17","BH1750":{"Illuminance":53}}}
{"StatusSTS":{"Time":"2021-09-10T17:16:17","Uptime":"3T21:02:02","UptimeSec":334922,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"fingerwech","BSSId":"44:4E:6D:40:CB:AD","Channel":6,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Configure to use an SSL-secured MQTT broker and turn it off (not just reject connection attempts, make it unavailable entirely) 

### EXPECTED BEHAVIOUR
No reset of settings. A restart without loss of settings is acceptable of course.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-09-10:
Seems that your device have entered in a bootloop or it has been power cycled several times before reaching 10 seconds of uptime. Because of that, Tasmota disables some settings to be able to start.

You can disable those protections by:

```lua
SetOption36 0
SetOption65 1
```

Elix-g, 2021-09-10:
You assume a boot loop due to the lack of connection to the broker? Ok, sounds reasonable. But the five steps of boot loop protection are the wrong reactions to a device restart caused by a faulty connection to the broker. So from my point of view, either the restart probably caused by trying to re-connect tro the broker shouldn't happen or the boot loop protection shouln't jump in upon such type of restart.

I will increase BOOT_LOOP_TIME and report back - please give me some time :-)

ascillato, 2021-09-10:
> You assume a boot loop due to the lack of connection to the broker?

No, as there are no logs, I can't tell if was:
1- a bootloop due to a problem in the firmware configuration that makes Tasmota not to start
OR
2- a power failure with consecutive cycles (7 or more) that don't allow Tasmota to reach an uptime of 10 seconds.

So, with the 2 commands posted above, you can disable both protections so as to not loose any configuration.
As there are no logs, I'm just guessing that it is most probably that your problem was just a power failure not related to TLS nor the broker.

Elix-g, 2021-09-10:
I will.

sfromis, 2021-09-10:
And do consider the reason *why* it is boot "too many" times.

ascillato2, 2021-09-21:
Hi, any news on this?

Jason2866, 2021-09-22:
Closing since there is no feedback.

closed 2021-09-22 by Jason2866