arendst/Tasmota #13117: Issue with EZOHUM driver

DolfLund, 2021-09-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): __ESP8266 nodemcu___
- [ ] Tasmota binary firmware version number used: __9.5.0___
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: ____NodeMCU PYFlasher_
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:07:48.065 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:07:48.290 RSL: RESULT = {"Module":{"0":"Generic"}}
07:07:48.545 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 10
00:00:00.055 QPC: Count 1
00:00:00.061 Project tasmota Tasmota Version 9.5.0(TasmoCompiler-esp8266generic)-2_7_4_9(2021-09-13T05:58:15)
00:00:00.550 WIF: Connecting to AP1 GRwifi Channel 1 BSSId DA:6A:91:7C:E0:A6 in mode 11n as tasmota_E4A900-2304...
00:00:02.753 WIF: Connected
00:00:03.004 HTP: Web server active on tasmota_E4A900-2304 with IP address 10.0.1.31
07:04:17.029 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"9.5.0(TasmoCompiler-esp8266generic)","FallbackTopic":"cmnd/DVES_E4A900_fb/","GroupTopic":"cmnd/tasmotas/"}}
07:04:17.035 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_E4A900-2304","IPAddress":"10.0.1.31"}}
07:04:17.045 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
07:04:19.423 QPC: Reset
07:04:21.387 RSL: STATE = {"Time":"2021-09-13T07:04:21","Uptime":"0T00:00:10","UptimeSec":10,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"GRwifi","BSSId":"DA:6A:91:7C:E0:A6","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:07:48.043 CMD: Backlog Template; Module; GPIO 255
07:07:48.065 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:07:48.290 RSL: RESULT = {"Module":{"0":"Generic"}}
07:07:48.545 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
07:08:20.919 CMD: Status 0
07:08:20.925 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E4A900","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:08:20.951 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:09","StartupUTC":"2021-09-13T06:04:11","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-09-13T07:00:18","SaveCount":11,"SaveAddress":"F9000"}}
07:08:20.984 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(TasmoCompiler-esp8266generic)","BuildDateTime":"2021-09-13T05:58:15","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"347/699"}}
07:08:20.005 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GRwifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
07:08:20.034 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FC0659E","04260001","00000003","00000000","C000E981","00000004","00001004","00000000"],"Drivers":"1,2,4,6,8,9,10,12,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2,78"}}
07:08:20.062 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_E4A900-2304","IPAddress":"10.0.1.31","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"3C:61:05:E4:A9:00","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:08:20.083 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E4A900","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:08:20.105 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-09-13T06:08:20","Local":"2021-09-13T07:08:20","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:24","Sunset":"19:07"}}
07:08:20.124 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-09-13T07:08:20"}}
07:08:20.130 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-09-13T07:08:20","Uptime":"0T00:04:09","UptimeSec":249,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"GRwifi","BSSId":"DA:6A:91:7C:E0:A6","Channel":1,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Issue is constant. Device not discovered/recognized
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile firmware with #define USE_EZOHUM
Connect EZO HUM device, i2cscan shows found device, but no device is detected/shown at top of web gui

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should be recognized and begin reporting data from sensor

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

i2cscan:
07:10:04.009 CMD: i2cscan
07:10:04.031 RSL: RESULT = {"I2CScan":"Device(s) found at 0x6f"}
### ADDITIONAL CONTEXT
_Add any other context about the problem here._

i2cdriver shows:
07:11:26.307 CMD: i2cdriver
07:11:26.313 RSL: RESULT = {"I2CDriver":"55"}

I have quite a few EZO devices but only one single EZO HUM,  my other EZO devices (EC and PH) work as expected with same firmware build just changing "#define USE_EZOHUM to USE_EZOEC

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-10-16:
Have you considered that your sensor may be damaged or not correctly connected ?
Have you tested it with code other than Tasmota ?

ascillato2, 2021-11-01:
Great to know that now your issue is solved.

closed 2021-11-01 by ascillato2