arendst/Tasmota #13127: Sensor53 OBIS Script seems to match substrings, not the exact string

nibblerrick, 2021-09-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The string matching mechanism of sensor53/OBIS seems to match on substrings, not the exact string.

My meter puts out this when dumping:

```
13:26:50.698 : 0.0.0(000000009)
13:26:51.065 : 0.1.0(00)
13:26:51.766 : 1.8.0(00000108*kWh)
13:26:52.465 : 2.8.0(00000264*kWh)
13:26:53.032 : 1.25(-13.35*kW)
13:26:53.598 : 31.25(005.75*A)
13:26:54.165 : 51.25(031.04*A)
13:26:54.732 : 71.25(020.66*A)
13:26:55.265 : 32.25(237.0*V)
13:26:55.799 : 52.25(239.7*V)
```

When trying to capture these with

```
1,1.8.0(@1,Total consumption,KWh,Total_in,0
1,2.8.0(@1,Total Feed,KWh,Total_out,0
1,1.25(@1,Current usage,kW,curr_use,2
1,31.25(@1,A L1,A,a_l1,2
1,51.25(@1,A L2,A,a_l2,2
1,71.25(@1,A L3,A,a_l3,2
1,32.25(@1,V L1,V,v_l1,1
1,52.25(@1,V L2,V,v_l2,1
```

it works for almost all string, only `1,1.25(@1,Current usage,kW,curr_use,2` gets overridden with 31.25, 51.25 and then 71.25, so it reports mostly the same numer as 71.25, as it's the last read value in the cycle.

Therefore it looks to me like it's matching not the exact string but part or last part of it.
I haven't found any possibilities like with regex ^1.25 to tell it should match beginning of the line and then 1.25.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemod D1 mini
- [X] Tasmota binary firmware version number used: 9.5.0.8(tasmota)
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: ESPHomeflasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"18":"Generic"}}
 {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
{"Status":{"Module":18,"DeviceName":"PM","FriendlyName":["PM"],"Topic":"tasmota_EABFF5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":300,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"2T01:00:52","StartupUTC":"2021-09-12T10:20:08","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2021-09-11T11:38:54","SaveCount":75,"SaveAddress":"F9000"},"StatusFWR":{"Version":"9.5.0.8(tasmota)","BuildDateTime":"2021-09-11T09:31:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"337/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00004000"]},"StatusMEM":{"ProgramSize":545,"Free":456,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87400586","042A8001","00000007","000000C0","00003008","00000000","00000000","00000020"],"Drivers":"1,2,3,4,6,9,10,12,16,18,19,22","Sensors":"1,2,3,4,53"},"StatusNET":{"Hostname":"tasmota-EABFF5-6565","IPAddress":"10.10.10.100","Gateway":"10.10.10.1","Subnetmask":"255.255.0.0","DNSServer1":"10.10.10.1","DNSServer2":"","Mac":"68:C6:3A:EA:FF:FF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"10.10.10.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EABFF5","MqttUser":"client","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-09-14T11:21:00","Local":"2021-09-14T13:21:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:25","Sunset":"20:05"},"StatusSNS":{"Time":"2021-09-14T13:21:00","PM_BB":{"Total_in":30108,"Total_out":66263,"a_l1":5.91,"a_l2":32.09,"a_l3":20.91,"v_l1":237.0,"v_l2":240.0,"v_l3":239.0,"Meter_number":"07474279"}},"StatusSTS":{"Time":"2021-09-14T13:21:00","Uptime":"2T01:00:52","UptimeSec":176452,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"Wifi":{"AP":1,"SSId":"SSID","BSSId":"####","Channel":11,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":4,"Downtime":"0T00:00:25"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
 {"WebLog":4}
 Saved to flash at F8, Count 76, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Wait a moment for the next readout-cycle

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Should match the exact string, so I can grab all values

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-09-19:
attached is an sml version that should fix this issue.
add #define SML_OBIS_LINE to user config
if it works for you i will do a pr.

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/7191074/xsns_53_sml.ino.zip)

nibblerrick, 2021-09-22:
Hi, thank you.
I compiled your version and that works as expected, the value doesn't get overwritten anymore!

PR #13198 (2021-09-22): fix sml obis
## Description:

**Related issue (if applicable):** fixes #13127

fixes obis decoder with rarely used short ids.
needs 

#define SML_OBIS_LINE

to do a line compare instead of a pattern compare

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.1.0.7.4
  - [x] I accept [... cut]
closed 2021-09-22 by arendst