arendst/Tasmota #13158: PowerOnState option to read the current state from switches when available

anishsane, 2021-09-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Consider a situation:
There is an extended power cut (few minutes to an hour).
If I change the switch position from on to off after the power cut, the firmware does not read the switch position after the power is restored.
So, with available PowerOnState options, the best I can do is to restore the relay1 state to the last state set when the esp was still powered.
However, it will be a good to have an option to follow the switch1 position on power on for relay1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 dev kit
- [X] Tasmota binary firmware version number used: 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: ESP-Flasher-Windows-x64.exe
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
    17:01:21.332 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
    17:01:21.545 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
    17:01:21.814 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"224":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"160":"Switch1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
  Rules not used
```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
    17:02:22.818 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_47E5F8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
    17:02:22.858 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:12:24","StartupUTC":"2021-09-17T15:49:58","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2021-09-17T16:47:48","SaveCount":28}}
    17:02:22.886 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:29:40","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"377/699"}}
    17:02:22.904 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sanes",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
    17:02:22.942 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1457,"Free":1856,"Heap":212,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","00084052","20200000","0000002C"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
    17:02:22.983 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_47E5F8-1528","IPAddress":"192.168.0.34","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"7C:9E:BD:47:E5:F8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
    17:02:23.006 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_47E5F8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
    17:02:23.025 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-09-17T16:02:23","Local":"2021-09-17T17:02:23","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:30","Sunset":"18:58"}}
    17:02:23.044 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-09-17T17:02:23","Switch1":"ON","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
    17:02:23.058 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-09-17T17:02:23","Uptime":"0T00:12:25","UptimeSec":745,"Heap":210,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"hidden","BSSId":"84:D8:1B:47:8B:A3","Channel":10,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
config: GPIO2 = relay(1), GPIO15 = switch(1).
PowerOnState: 3
SwitchMode1: 2

1. Set the relay state to some value using switch or web console (or alexa/mqtt etc)
2. Disconnect the power to esp module
3. Toggle the switch
4. Restore the power to module.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
There should be a way that relay should follow current position of the switch.

Even better if the code can differentiate if the last value (relay1) was set via the hardware switch (switch1) or from software (e.g. web).
That way, we can set the relay to the last value set across power failure as well.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-09-17:
Hi,

That is already available. you need to use a rule to be triggered at boot time like:

```js
Rule1 ON Switch1#Boot DO Power1 %value% ENDON
```

closed 2021-09-17 by ascillato2

anishsane, 2021-09-17:
Thanks. This works.
I read the Rules page, but I was not able to understand that %value% would reflect the value of the switch even for Switch#boot event.
I thought it's only relevant for Switch#State trigger.

Thanks again for the support.

anishsane, 2021-11-03:
I have 8 switches and 8 relays.
Is there any limit for rules?
I understand that there are 3 rulesets and each ruleset can contain approx 1000 bytes of rules. So, can I add different rules for different switches in the same ruleset?

barbudor, 2021-11-03:
Yes
If it becomes complicated to enter all rules into a single command line you use multiple lines such as
```
Rule1 on..... endon
Rule1 +on..... endon 
Rule1 +on..... endon
```