arendst/Tasmota #13162: MQTT does not connect to broker.emqx.io

sanchosk, 2021-09-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I use external MQTT server (broker.emqx.io), tasmota does not connect.
In random cases, MQTT connects after few retries.
Connection from mosquitto_pub/sub, node red and other tasmota works OK.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wamos D1 mini
- [X] Tasmota binary firmware version number used: tasmota-display.bin 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:03:37.500 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:03:37.729 RSL: RESULT = {"Module":{"18":"Generic"}}
21:03:37.936 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"608":"I2C SCL"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:04:38.131 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["ESPdisplay"],"Topic":"display","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:04:38.156 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:03:44","StartupUTC":"2021-09-17T20:00:54","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2021-09-17T16:30:16","SaveCount":82,"SaveAddress":"F9000"}}
21:04:38.189 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.8(display)","BuildDateTime":"2021-09-17T12:45:58","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
21:04:38.208 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wlan_home2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
21:04:38.236 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":602,"Free":400,"Heap":28,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","401F8000","00000000"],"Drivers":"1,2,4,5,8,9,10,12,13,16","Sensors":"1,2,5,6"}}
21:04:38.264 RSL: STATUS5 = {"StatusNET":{"Hostname":"display-4654","IPAddress":"192.168.11.225","Gateway":"192.168.11.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.11.254","DNSServer2":"0.0.0.0","Mac":"8C:AA:B5:77:F2:2E","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:04:38.289 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"broker.emqx.io","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_77F22E","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:04:38.312 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-09-17T20:04:38","Local":"2021-09-17T21:04:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:30","Sunset":"18:58"}}
21:04:38.330 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-09-17T21:04:38"}}
21:04:38.336 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-09-17T21:04:38","Uptime":"0T00:03:44","UptimeSec":224,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wlan_home2","BSSId":"90:EF:68:7E:BC:48","Channel":11,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 83
00:00:00.056 QPC: Count 1
00:00:00.058 CFG: CR 372/699, Busy 0
00:00:00.062 SRC: Display
00:00:00.063 RSL: RESULT = {"POWER":"ON"}
00:00:00.065 RSL: POWER = ON
00:00:00.065 SRC: Restart
00:00:00.067 Project tasmota - Tasmota Version 9.5.0.8(display)-2_7_4_9(2021-09-17T12:45:58)
00:00:00.218 WIF: Checking connection...
00:00:00.219 WIF: Attempting connection...
00:00:00.555 WIF: Connecting to AP1 wlan_home2 Channel 11 BSSId 90:EF:68:7E:BC:48 in mode 11n as display-4654...
00:00:01.752 WIF: Checking connection...
00:00:01.753 WIF: Connected
00:00:02.003 HTP: Web server active on display-4654 with IP address 192.168.11.225
00:00:02.207 HTP: Main Menu
00:00:02.412 NTP: Synch time...
00:00:02.451 RTC: UTC 2021-09-17T20:05:45, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
21:05:45.000 NTP: Synched
21:05:46.004 MQT: Attempting connection...
21:05:46.288 MQT: Connect failed to broker.emqx.io:1883, rc -2. Retry in 10 sec
21:05:46.496 HTP: Console
21:05:49.448 QPC: Reset
21:05:51.402 APP: Boot Count 27
21:05:51.579 CFG: Saved to flash at F7, Count 84, Bytes 4096
21:05:57.300 MQT: Attempting connection...
21:05:57.513 MQT: Connect failed to broker.emqx.io:1883, rc -2. Retry in 20 sec

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install tasmota on clean wemos, configure wifi and set mqtt broker to broker.emqx.io, port 1883, username and password not needed.
Restart tasmota.
Tasmota seems to report MQTT status -2 and retries. Sometimes it connects after 3-4 retries, but usually not.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota will connect to MQTT the same way as all other clients.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The same device works 100% reliably with local mosquitto server.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-09-17:
Error -2 means tcp connection error. It is a connectivity issue. And Tasmota can't do anything about it.

sanchosk, 2021-09-18:
Is there some way how to debug this problem? It's strange especially if 2 tasmotas connect, 2 PCs connect and 1 tasmota does not with this error message :(

s-hadinger, 2021-09-18:
Best way would be to use Wireshark and try to find what's going on.

Jason2866, 2021-09-18:
Closing since it is answered and the issue is not from Tasmota

closed 2021-09-18 by Jason2866