arendst/Tasmota #13167: Unable to update the Topic Field keeps reverting to default

Tipalo, 2021-09-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I a unable to update the Topic Field on Versions 9.5.0.0 and 9.5.0.8
The field is reset back to default  I am trying to set topic as P81_BenchLamp or P1-BenchLamp.

Changing the topic to   sexybeast   DOES WORK.    But that DOESN'T work for me.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _LITESUN LA-WF7____
- [ ] Tasmota binary firmware version number used: 9.5.0 also 9.5.0.8_____
  - [ ] Pre-compiled YES
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
06:03:49.896 MQT: stat/sexybeast/RESULT = {"NAME":"LITESUN LA-WF7","GPIO":[0,320,0,32,2720,2656,0,0,2624,321,224,0,0,0],"FLAG":0,"BASE":18}
06:03:50.105 MQT: stat/sexybeast/RESULT = {"Module":{"0":"LITESUN LA-WF7"}}
06:03:50.361 MQT: stat/sexybeast/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"320":"Led_i1"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
05:53:44.516 CMD: Status 0`: ```lua
05:53:44.524 MQT: stat/tasmota_B10C83/STATUS = {"Status":{"Module":0,"DeviceName":"P81_BenchLamp","FriendlyName":["P81_BenchLamp"],"Topic":"tasmota_B10C83","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
05:53:44.529 MQT: stat/tasmota_B10C83/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:47","StartupUTC":"2021-09-19T04:48:57","Sleep":50,"CfgHolder":4617,"BootCount":204,"BCResetTime":"2020-08-03T07:01:19","SaveCount":1062,"SaveAddress":"F8000"}}
05:53:44.534 MQT: stat/tasmota_B10C83/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"382/699"}}
05:53:44.539 MQT: stat/tasmota_B10C83/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ToEjAm","Step31"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
05:53:44.552 MQT: stat/tasmota_B10C83/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
05:53:44.562 MQT: stat/tasmota_B10C83/STATUS5 = {"StatusNET":{"Hostname":"P81_BenchLamp","IPAddress":"192.168.1.81","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"DC:4F:22:B1:0C:83","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
05:53:44.568 MQT: stat/tasmota_B10C83/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.126","MqttPort":1883,"MqttClientMask":"P81_BenchLamp","MqttClient":"P81_BenchLamp","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
05:53:44.574 MQT: stat/tasmota_B10C83/STATUS7 = {"StatusTIM":{"UTC":"2021-09-19T04:53:44","Local":"2021-09-19T05:53:44","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:33","Sunset":"18:54"}}
05:53:44.579 MQT: stat/tasmota_B10C83/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
05:53:44.586 MQT: stat/tasmota_B10C83/STATUS10 = {"StatusSNS":{"Time":"2021-09-19T05:53:44","ENERGY":{"TotalStartTime":"2020-07-25T05:45:47","Total":10.048,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
05:53:44.592 MQT: stat/tasmota_B10C83/STATUS11 = {"StatusSTS":{"Time":"2021-09-19T05:53:44","Uptime":"0T00:04:47","UptimeSec":287,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ToEjAm","BSSId":"24:5A:4C:1A:66:59","Channel":6,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
update the MQTT Topic field to something with an underscore, or a dash and caps

the result is field reset to default

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected to be able to use the device name as the topic. If the device name is legal then it should also be fine for topic.  I was using P81_BenchLamp as topic. It conveys to me that it is a Plug Module at 192.168.1.81 with a friendly name.
tasmota_2a4d13 is not useful when configuring subscriptions.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I am using this in Homeseer as well as Home Assistant
Thanks for the great work !  Even though I am bitchin I am grateful for your talents.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-09-19:
Usually this is due to people breaking rule about having MqttClient and Topic different, which will cause Tasmota to reset them.

Tipalo, 2021-09-21:
Thanks. I was not aware of this.

Tipalo, 2021-09-21:
I've read through the documentation  "Getting Started" page on Tasmota.com and could not find any mention of this.  I have been using Client=Topic for some time on many devices.  I would be grateful if you can point me to a doc that explains this.

ascillato2, 2021-09-21:
From https://tasmota.github.io/docs/Commands/

[image]

closed 2021-09-21 by ascillato2

Tipalo, 2021-09-21:
Just an FYI I can not find anything that mentions this on the link provided using the search tool.  I did due diligence before posting here so I don't want to come off the wrong way.  If this is by design, it might be good idea to add a note under the getting started section in the DOC.
[image]

ascillato, 2021-09-21:
If you press `CTRL+F` and then you type `topic can not be`, you obtain:

[image]

pmknowles, 2021-11-20:
I have to ask. When did this change and why? I have several devices which have the same MQTT client and topic since V6. It's now getting vital that when things change the docs need to say 'changed from Vx onwards'. On the principle stated on one of the upgrade pages 'if it ain't broke' I have devices on different versions. I upgraded a device to V10. It carried on with the same client and topic. I then moved my Mosquitto to a different device and the change on the MQTT configuration page broke several devices because it SILENTLY changed the topic setting.

pmknowles, 2021-11-20:
By the way, there is nothing in the MQTT specification that requires the ClientID and Topic to be different. The only requirement is that each device must have a unique ClientID. Moreover I have searched the release notes and have been unable to find a notification as to when this changed. It HAS changed because I still have devices which I have not migrated to my new Mosquitto server with identical ClientID and topics.

sfromis, 2021-11-20:
The coding in Tasmota checking the sameness is not 100%, it is not impossible to sneak through a configuration likely to suddenly cause you trouble later.

barbudor, 2021-11-20:
I've been able to track back this test to 6.2.1.14 On Oct 10th 2018
As there have been a major refactoring of MQTT module at that time, it is not easy to check if it was there even before.

While the MQTT standard do not enforce that, it may also be a limitation of the PubSubClient library that we are using.
Many reasons could be behind this although it doesn't seems to have been documented why.

Personnaly I consider it is a good habit to have the ClientD tighted to the physical layer (the hardware, based on MAC address) and the Topic tight to the logical layer (what it is used for).

pmknowles, 2021-11-20:
It looks like it’s further back than that

Incorrect generation of MQTT FullTopic queue name · Issue #1528 · arendst/Tasmota (github.com)<https://github.com/arendst/Tasmota/issues/1528>

With fallback topics

I started with V5.2 in 2017

Regards

Phil K