arendst/Tasmota #13196: MQT: Connect failed to <mqtt host IP>:1883, rc -2. Retry in XX sec

sym0nd0, 2021-09-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Connection to broker was working but now it is unable to connect show rc -2 error.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Athom 16A UK Power Monitoring Plug (Model reference: PG04-UK16A)
- [X] Tasmota binary firmware version number used: 9.5.0.8
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: None, pre-flashed - updated to latest version using web interface
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:41:02.960 CMD: Backlog Template; Module; GPIO 255
09:41:03.038 RSL: RESULT = {"NAME":"Athom PG04-UK16A","GPIO":[0,0,0,32,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":18}
09:41:03.245 RSL: RESULT = {"Module":{"0":"Athom PG04-UK16A"}}
09:41:03.499 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:41:39.061 CMD: Status 0
09:41:39.067 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"OctoPi Plug","FriendlyName":["OctoPi Plug"],"Topic":"tasmota_5603E4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:41:39.094 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:36","StartupUTC":"2021-09-22T08:39:03","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-09-22T08:43:56","SaveCount":26,"SaveAddress":"FA000"}}
09:41:39.127 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.8(tasmota)","BuildDateTime":"2021-09-21T12:38:47","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"368/699"}}
09:41:39.146 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00004000"]}}
09:41:39.176 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":609,"Free":392,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:41:39.206 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-5603E4-0996","IPAddress":"192.168.1.25","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:56:03:E4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:41:39.230 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5603E4","MqttUser":"ha_mqtt","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:41:39.253 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-09-22T08:41:39","Local":"2021-09-22T09:41:39","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:37","Sunset":"18:47"}}
09:41:39.271 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:41:39.287 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-09-22T09:41:39","ENERGY":{"TotalStartTime":"2021-09-22T08:43:56","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
09:41:39.308 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-09-22T09:41:39","Uptime":"0T00:02:36","UptimeSec":156,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WiFi,"BSSId":"42:37:86:2E:A3:97","Channel":1,"Mode":"11n","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
09:44:00.949 MQT: Connect failed to 192.168.1.3:1883, rc -2. Retry in 80 sec
09:44:03.420 RSL: HASS_STATE = {"Version":"9.5.0.8(tasmota)","BuildDateTime":"2021-09-21T12:38:47","Module or Template":"Athom PG04-UK16A","RestartReason":"Software/System restart","Uptime":"0T00:05:00","Hostname":"tasmota-5603E4-0996","IPAddress":"192.168.1.25","RSSI":"100","Signal (dBm)":"-42","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":0,"LoadAvg":19}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure MQTT, same details as numerous other Tasmota devices.
Connects to broker.
_Some time later..._
Loses connection and won't reconnect

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Connect to MQTT broker and stay connected.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I've the same problem affecting another plug, same make and model, and also my OctoPrint (running OctoPi OS) instance also appears to be unable to connect. All other MQTT/Tasmota devices are working as expected, with no connection problems.

Broker is Mosquitto Broker v6.0.1, running as an addon and subsequent integration in Home Assistant core-2021.9.7.

Home Assistant Discovery enabled, via `SetOption19 1` 

Also running Tasmota Home Assistant integration and Zigbee2mqtt addon.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-09-22:
Do not open an issue when you just have network setup trouble. That rc -2 simply means network connection failed, reasons not known to Tasmota.

sym0nd0, 2021-09-22:
That's helpful.

MoLow, 2021-09-22:
I have had issues with setting mqtt password via UI - the ui seems to turn password into `****` - When setting the password directly via the console I was able to connect

sym0nd0, 2021-09-22:
> I have had issues with setting mqtt password via UI - the ui seems to turn password into `****` - When setting the password directly via the console I was able to connect - this happend with the latest Tasmota version

THANK YOU! That sorted it right away.

closed 2021-09-22 by sym0nd0

steffenberthold, 2023-11-11:
What will be the command to be entered in the console for this? Is this a one time solution or remains the connetcion forever (how long)? Thx

barbudor, 2023-11-11:
RTFM
https://tasmota.github.io/docs/Commands/#mqttpassword

barbudor, 2023-11-11:
https://tasmota.github.io/docs/Commands/#mqttpassword

barbudor, 2023-11-11:
And don't come here to wake up old issues
Open an new discussion or better got to discord for so simple questions

Finners101, 2025-01-05:
Fuckwit