arendst/Tasmota #13201: Switch delay when turned on, but not when off. Sometimes no Events triggered.

ReviloEgros, 2021-09-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am experiencing a delay of roughly one second when i turn on the switch. All events or rules run after one second. When i turn the switch off, there is no delay and all actions are almost instant. If i turn the switch on for less than a second, i get no events at all. I am using a Wemos D1 mini and wired my switch to GPIO 2 which is tied to the onboard led. If i turn the switch on, the onboard led is instantaneously lit but as i said all actions sadly follow one second later.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.5.0.8(sensors)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:21:24.310 CMD: Backlog Template; Module; GPIO 255
20:21:24.313 SRC: WebConsole from 192.168.188.47
20:21:24.315 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
20:21:24.329 SRC: Backlog
20:21:24.332 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
20:21:24.340 MQT: stat/QuestionBlock/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:21:24.559 SRC: Backlog
20:21:24.561 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
20:21:24.567 MQT: stat/QuestionBlock/RESULT = {"Module":{"18":"Generic"}}
20:21:24.786 SRC: Backlog
20:21:24.788 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
20:21:24.799 MQT: stat/QuestionBlock/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"160":"Switch1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"2144":"MP3 Player"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
20:22:24.895 CMD: Backlog Rule1; Rule2; Rule3
20:22:24.897 SRC: WebConsole from 192.168.188.47
20:22:24.899 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
20:22:24.913 SRC: Backlog
20:22:24.916 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
20:22:24.924 MQT: stat/QuestionBlock/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":87,"Free":424,"Rules":"ON System#Boot DO Backlog MP3EQ 5; Delay 10; MP3Volume 70; Delay 10; MP3Track 121 ENDON"}}
20:22:25.150 SRC: Backlog
20:22:25.153 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
20:22:25.160 MQT: stat/QuestionBlock/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":101,"Free":410,"Rules":"on switch1#state=1 do publish wled/mario on endon on switch1#state=0 do publish wled/mario off endon "}}
20:22:25.379 SRC: Backlog
20:22:25.382 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
20:22:25.389 MQT: stat/QuestionBlock/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:22:56.950 CMD: status 0
20:22:56.953 SRC: WebConsole from 192.168.188.47
20:22:56.955 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
20:22:56.962 MQT: stat/QuestionBlock/STATUS = {"Status":{"Module":18,"DeviceName":"QuestionBlock","FriendlyName":["QuestionBlock"],"Topic":"QuestionBlock","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:22:56.980 MQT: stat/QuestionBlock/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T02:01:21","StartupUTC":"2021-09-22T16:21:35","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2021-09-22T00:24:06","SaveCount":91,"SaveAddress":"F8000"}}
20:22:57.000 MQT: stat/QuestionBlock/STATUS2 = {"StatusFWR":{"Version":"9.5.0.8(sensors)","BuildDateTime":"2021-09-21T08:59:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"364/699"}}
20:22:57.017 MQT: stat/QuestionBlock/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Umbrella2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A00B","2805C8000100060000005A0A000000000000","00800480","00006000","00006001"]}}
20:22:57.045 MQT: stat/QuestionBlock/STATUS4 = {"StatusMEM":{"ProgramSize":712,"Free":1332,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","000840D2","20200000","00002020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,38,50","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,91"}}
20:22:57.065 MQT: stat/QuestionBlock/STATUS5 = {"StatusNET":{"Hostname":"QuestionBlock-5416","IPAddress":"192.168.188.181","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.188.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:DC:35:28","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:22:57.081 MQT: stat/QuestionBlock/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.200","MqttPort":1883,"MqttClientMask":"QuestionBlock","MqttClient":"QuestionBlock","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:22:57.099 MQT: stat/QuestionBlock/STATUS7 = {"StatusTIM":{"UTC":"2021-09-22T18:22:57","Local":"2021-09-22T20:22:57","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:37","Sunset":"19:47"}}
20:22:57.117 MQT: stat/QuestionBlock/STATUS10 = {"StatusSNS":{"Time":"2021-09-22T20:22:57","Switch1":"OFF"}}
20:22:57.134 MQT: stat/QuestionBlock/STATUS11 = {"StatusSTS":{"Time":"2021-09-22T20:22:57","Uptime":"0T02:01:22","UptimeSec":7282,"Heap":20,"SleepMode":"Normal","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Umbrella2G","BSSId":"76:83:C2:15:7D:0E","Channel":4,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:16"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
20:26:03.905 CMD: mp3track 1
20:26:03.907 SRC: WebConsole from 192.168.188.47
20:26:03.909 CMD: Grp 0, Cmnd 'MP3TRACK', Idx 1, Len 1, Data '1'
20:26:04.926 MQT: stat/QuestionBlock/RESULT = {"MP3Track":1}
20:26:04.988 RUL: SWITCH1#STATE=1 performs "publish wled/mario on"
20:26:04.990 SRC: Rule
20:26:04.992 CMD: Grp 0, Cmnd 'PUBLISH', Idx 1, Len 13, Data 'wled/mario on'
20:26:04.998 MQT: wled/mario = on
20:26:05.007 MQT: stat/QuestionBlock/RESULT = {"Switch1":{"Action":"ON"}}
20:26:08.875 RUL: SWITCH1#STATE=0 performs "publish wled/mario off"
20:26:08.877 SRC: Rule
20:26:08.879 CMD: Grp 0, Cmnd 'PUBLISH', Idx 1, Len 14, Data 'wled/mario off'
20:26:08.884 MQT: wled/mario = off
20:26:08.890 MQT: stat/QuestionBlock/RESULT = {"Switch1":{"Action":"OFF"}}

```
### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure one GPIO as switch and use rules to publish MQTT events and turn it on and off, and turn it on and of within one second. In my example i use rules to trigger another esp via MQTT. The on event is delayed, the off event is instant. If i do a manual publish wled/mario on in the console it reacts instant too. So it has nothing to do with my wifi or mqtt communication.
SetOption1 ON
SetOption13 ON
SetOption73 ON
SetOption114 ON
SetOption127 ON

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Send events instantaneously without any delay and  send events if the on/off period is less than a second.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-09-22:
Since it is a Switch, not a Button (in configuration), most of your SetOption values do not even apply. Since you publish a message to another device, a question is exactly where in the flow you see a delay. From push to rule firing? From rule firing to publish command? From publish command to MQTT message published? From message published to received by the other device? From reception by the other device to action seen? When you experience swift off-action, this does not preclude the possibility that "on" could be processed differently. "Insignificant" details matter a lot. 

Instead of wiring the switch to the onboard led, you could also config it as a relay and have a rule like `on Switch1#state do Power1 %value% endon`, or include it in your rules. That gives a very effective feedback on the Tasmota-side processing time.

And you can not at all judge speed issues in the Tasmota web console, better look at a MQTT subscription (with a speedy tool) or serial console.

ascillato2, 2021-09-23:
Moving to discussion.

closed 2021-09-23 by ascillato2