arendst/Tasmota #13203: UPNPDescription with Belkin WeMo Emulation - UPNP SCPD Description is wrong

simann, 2021-09-22:
### PROBLEM DESCRIPTION
I have an Tasmota Device with Belkin WeMo Emulation in my network . Besides that an Program that scans for UPNP Devices and there configurations. The SCPD Description  of this device with WeMo Emulation seems to be wrong. This is the url:  http://deviceip/metainfoservice.xml

IMHO: In the `<argumentList>`  section is an `<argument>`  node missing.

According to the UPNP Specification (ref: http://www.upnp.org/specs/arch/UPnP-arch-DeviceArchitecture-v1.0-20080424.pdf, page 32): 

[image]

`<argument>`  is required and should be direct child of `<argumentList>`

Screenshot Device:
[image]

Screenshot XML:
[image]

The "eventservice" XML follows this specification:
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic):  Sonoff T1EU1C
- [X ] Tasmota binary firmware version number used: 8.1.0.1
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: tuya-convert
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:50:15 MQT: stat/switch/simon_wand/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:50:15 MQT: stat/switch/simon_wand/RESULT = {"Module":{"1":"Sonoff Basic"}}
22:50:15 MQT: stat/switch/simon_wand/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:50:55 CMD: status 0
22:50:55 MQT: stat/switch/simon_wand/STATUS = {"Status":{"Module":1,"FriendlyName":["Simon_Lichtschalter_Wand"],"Topic":"switch/simon_wand","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"simon_licht_schalter_wand","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:50:55 MQT: stat/switch/simon_wand/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"46T12:00:28","StartupUTC":"2021-08-07T09:50:27","Sleep":50,"CfgHolder":4617,"BootCount":70,"SaveCount":4998,"SaveAddress":"F6000"}}
22:50:55 MQT: stat/switch/simon_wand/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(d627de5-tasmota)","BuildDateTime":"2019-12-29T12:00:11","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"452/699"}}
22:50:55 MQT: stat/switch/simon_wand/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pretty fly for a Wifi",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00028009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
22:50:55 MQT: stat/switch/simon_wand/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:50:55 MQT: stat/switch/simon_wand/STATUS5 = {"StatusNET":{"Hostname":"SImon-Lichtschalter-Tasmota","IPAddress":"192.168.178.54","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"D8:F1:5B:82:DE:2E","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
22:50:55 MQT: stat/switch/simon_wand/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.168","MqttPort":1883,"MqttClientMask":"SIMON_LICHT_SCHALTER_WAND","MqttClient":"SIMON_LICHT_SCHALTER_WAND","MqttUser":"sonoff","MqttCount":142,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:50:55 MQT: stat/switch/simon_wand/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 22 21:50:55 2021","Local":"Wed Sep 22 22:50:55 2021","StartDST":"Sun Mar 28 02:00:00 2021","EndDST":"Sun Oct 31 03:00:00 2021","Timezone":"+01:00","Sunrise":"06:37","Sunset":"18:48"}}
22:50:55 MQT: stat/switch/simon_wand/STATUS10 = {"StatusSNS":{"Time":"2021-09-22T22:50:55"}}
22:50:55 MQT: stat/switch/simon_wand/STATUS11 = {"StatusSTS":{"Time":"2021-09-22T22:50:55","Uptime":"46T12:00:28","UptimeSec":4017628,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":142,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Pretty fly for a Wifi","BSSId":"DC:15:C8:62:D3:A9","Channel":11,"RSSI":100,"Signal":-42,"LinkCount":112,"Downtime":"0T00:07:32"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
 Activate WeMo Emulation, and navigate to the url mentioned above, e.g.

`http://192.168.178.54/metainfoservice.xml`

### EXPECTED BEHAVIOUR
The XML Description should have an child node `<Argument>`  and that should wrap around the information for every parameter.

### SCREENSHOTS
Screenshots can be seen above.

### ADDITIONAL CONTEXT
Currently im using Tasmota version 8.1.0.1, but i will update in a few days and check if the issue still persists.

Besides that Tasmota is a great software and im really glad you folks are putting that much effort into it! :)

Sincerely Simon

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-09-23:
Thanks for you detailed description. Take into account that ESP8266 resources are **VERY** limited. Wemo emulation is stripped down to the minimum required for Alexa to work.

It's definitely missing some arguments, but adding them would be a waste of Flash resource and would not have any impact on Alexa.

Unless there is a good reason to spend flash space for this feature, it will not be added.

simann, 2021-09-23:
IMO there is no good reason. I wasnt sure if this is intended or a bug, tho.

I already made some changes in my program to ignore this edge case. :)

closed as "wont fix" or "working as intended"

closed 2021-09-23 by simann