arendst/Tasmota #13214: Cant connect to Wifi with 63 Chars

chlorophyll-zz, 2021-09-24:
### PROBLEM DESCRIPTION
Cant connect to Wifi with 63 Chars:
Wifi Password (randomized):
B947hwZS&*/Zk]pGFn-,4^K+S(7'nTgLpXeYzp%r;XpJ>B7)B-WtU&tFoH+h>?R

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used: 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used:Webinterface
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
`17:36:32.255 CMD: Backlog Template; Module; GPIO 255
17:36:32.277 MQT: stat/tasmota_581106/RESULT = {"NAME":"LED-RGBW","GPIO":[0,0,0,0,416,420,0,0,417,419,418,0,0,0],"FLAG":0,"BASE":18}
17:36:32.481 MQT: stat/tasmota_581106/RESULT = {"Module":{"0":"LED-RGBW"}}
17:36:32.687 MQT: stat/tasmota_581106/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"420":"PWM5"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"419":"PWM4"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}` 

```lua

```
17:37:47.864 CMD: lua
17:37:47.872 MQT: stat/tasmota_581106/RESULT = {"Command":"Unknown"}
```

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
```
17:39:45.483 CMD: Status 0
17:39:45.491 MQT: stat/tasmota_581106/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_581106","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:39:45.520 MQT: stat/tasmota_581106/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:05:47","StartupUTC":"2021-09-24T16:33:58","Sleep":50,"CfgHolder":4617,"BootCount":98,"BCResetTime":"2021-09-23T09:57:37","SaveCount":1650,"SaveAddress":"FA000"}}
17:39:45.555 MQT: stat/tasmota_581106/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"390/699"}}
17:39:45.578 MQT: stat/tasmota_581106/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["meins","FRITZ!Box Gastzugang"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:39:45.612 MQT: stat/tasmota_581106/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:39:45.646 MQT: stat/tasmota_581106/STATUS5 = {"StatusNET":{"Hostname":"tasmota_581106-4358","IPAddress":"192.168.188.24","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.1","Mac":"CC:50:E3:58:11:06","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:39:45.670 MQT: stat/tasmota_581106/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.188.1","MqttPort":1883,"MqttClientMask":"DVES_581106","MqttClient":"DVES_581106","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:39:45.694 MQT: stat/tasmota_581106/STATUS7 = {"StatusTIM":{"UTC":"2021-09-24T16:39:45","Local":"2021-09-24T17:39:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:40","Sunset":"18:43"}}
17:39:45.712 MQT: stat/tasmota_581106/STATUS10 = {"StatusSNS":{"Time":"2021-09-24T17:39:45"}}
17:39:45.723 MQT: stat/tasmota_581106/STATUS11 = {"StatusSTS":{"Time":"2021-09-24T17:39:45","Uptime":"0T00:05:47","UptimeSec":347,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"000000BB44","HSBColor":"0,0,0","White":100,"CT":246,"Channel":[0,0,0,73,27],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"FRITZ!Box Gastzugang","BSSId":"5C:49:79:7F:E1:DF","Channel":11,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:
Set Wifi Password:B947hwZS&*/Zk]pGFn-,4^K+S(7'nTgLpXeYzp%r;XpJ>B7)B-WtU&tFoH+h>?R
Connect with Tasmota

### EXPECTED BEHAVIOUR
Tasmota connect to Wifi

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-09-24:
From the docs about password:
**Do not use special characters or white spaces in the password.**

chlorophyll-zz, 2021-09-24:
I want to add this as Feature Request.

ascillato2, 2021-09-24:
>I want to add this as Feature Request.

Ok, Moving this to discussions/Ideas.

Thanks.

closed 2021-09-24 by ascillato2