arendst/Tasmota #13224: No data on Website or console. Modbus i.O. Response over Modbus->serial (SDM630) plausible

Chriss312, 2021-09-27:
### PROBLEM DESCRIPTION
First. I have flashed Tasmota 9.5.0 (sensors).
Second. I configured my Wemos d1 clone to Generic with GPIO 1 SDM630 TX an GPIO 3 to SDM630 RX.
I see TX and RX Led flashing..

I see over my oscilloscope a request and a plausible response from my SDM630. 
Response: 0x( 01 04 04 43 60 25 88 F4 E8)  --> 224.146606445 Volt

But no data is displayed in the console as well as on the website.

Console:
09:40:27.423 RSL: SENSOR = {"Time":"2021-09-27T09:40:27","ENERGY":{"TotalStartTime":"2021-09-27T01:35:28","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":[0,0,0],"ApparentPower":[0,0,0],"ReactivePower":[0,0,0],"Factor":[0.00,0.00,0.00],"Voltage":[0,0,0],"Current":[0.000,0.000,0.000]}}

It looks like Tasmota is not processing the response.

Have you a tip for me to resolve the problem?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 wifi (clone?)
- [ ] Tasmota binary firmware version number used: 9.5.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: https://arendst.github.io/Tasmota-firmware/
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:49:35.145 CMD: Backlog Template; Module; GPIO 255`: ```luaBacklog Template; Module; GPIO 255`: ```lua
09:49:35.147 SRC: WebConsole from 192.168.179.23
09:49:35.150 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 78, Data 'Template; Module; GPIO 255`: ```luaBacklog Template; Module; GPIO 255`: ```lua'
09:49:35.216 SRC: Backlog
09:49:35.219 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
09:49:35.225 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:49:35.421 SRC: Backlog
09:49:35.423 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
09:49:35.427 RSL: RESULT = {"Module":{"0":"Generic"}}
09:49:35.626 SRC: Backlog
09:49:35.629 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 28, Data '255`: ```luaBacklog Template'
09:49:35.637 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1984":"SDM630 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2016":"SDM630 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
09:49:35.830 SRC: Backlog
09:49:35.832 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
09:49:35.835 RSL: RESULT = {"Module":{"0":"Generic"}}
09:49:36.034 SRC: Backlog
09:49:36.036 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 12, Data '255`: ```lua'
09:49:36.044 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1984":"SDM630 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2016":"SDM630 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
09:49:42.245 WIF: Checking connection...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
09:51:40.371 CMD: Backlog Rule1; Rule2; Rule3`: ```lua
09:51:40.373 SRC: WebConsole from 192.168.179.23
09:51:40.376 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 28, Data 'Rule1; Rule2; Rule3`: ```lua'
09:51:40.443 SRC: Backlog
09:51:40.445 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
09:51:40.450 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:51:40.647 SRC: Backlog
09:51:40.650 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
09:51:40.654 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:51:40.851 SRC: Backlog
09:51:40.853 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 9, Data '`: ```lua'
09:51:40.857 RUL: Stored uncompressed, would compress from 9 to 10 (-0%)
09:51:40.860 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":9,"Free":502,"Rules":"`: ```lua"}}
09:51:41.297 CFG: Saved to flash at F6, Count 22, Bytes 4096

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

09:48:10.869 CMD: Status 0
09:48:10.871 SRC: WebConsole from 192.168.179.23
09:48:10.874 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
09:48:10.879 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_BDA8E8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:48:10.886 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T07:42:54","StartupUTC":"2021-09-27T01:05:16","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2021-09-27T01:35:28","SaveCount":21,"SaveAddress":"F7000"}}
09:48:10.891 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(sensors)","BuildDateTime":"2021-06-17T08:26:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
09:48:10.896 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["hn_g",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00400080","00006000","00000000"]}}
09:48:10.913 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":661,"Free":340,"Heap":23,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840EF","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
09:48:10.921 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_BDA8E8-2280","IPAddress":"192.168.179.15","Gateway":"192.168.179.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.179.1","Mac":"5C:CF:7F:BD:A8:E8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:48:10.924 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BDA8E8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:48:10.930 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-09-27T08:48:10","Local":"2021-09-27T09:48:10","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:44","Sunset":"18:37"}}
09:48:10.934 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:48:10.943 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-09-27T09:48:10","ENERGY":{"TotalStartTime":"2021-09-27T01:35:28","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0,0],"ApparentPower":[0,0,0],"ReactivePower":[0,0,0],"Factor":[0.00,0.00,0.00],"Voltage":[0,0,0],"Current":[0.000,0.000,0.000]}}}
09:48:10.950 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-09-27T09:48:10","Uptime":"0T07:42:54","UptimeSec":27774,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"hn_g","BSSId":"E2:28:6D:67:E2:2E","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}
09:48:22.251 WIF: Checking connection...

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)*
[image]
*

Chriss312, 2021-09-27:
I found this in the console.

10:35:41.506 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"9.5.0(sensors)","FallbackTopic":"cmnd/DVES_BDA8E8_fb/","GroupTopic":"cmnd/tasmotas/"}}
10:35:41.510 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_BDA8E8-2280","IPAddress":"192.168.179.15"}}
10:35:41.514 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["40257022","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}}
10:35:44.189 QPC: Reset
**10:35:45.158 NRG: Energy reset by invalid data**
10:35:46.161 APP: Boot Count 11

sfromis, 2021-09-27:
You should read the docs for [Smart Meter Interface](https://tasmota.github.io/docs/Smart-Meter-Interface/) and compile a build as described.

Chriss312, 2021-09-27:
Thank you for your reply. But the answer is not enough for me. Why should I do this?
Why should i compile my own build? 
Is there any problem with the pre compiled "sensors" version or is something missing there?
If something is missing in the precompiled sensors version, why can I select the SDM630?

sfromis, 2021-09-27:
Read the page I linked. 
**This feature is not included in precompiled binaries**

Chriss312, 2021-09-27:
Ok. I'll try. But I think that is not nice.

In my opinion, I expect that when I select a "sensor version" in which I can configure the corresponding pins, everything is implemented to be able to read out the device via the pins.

I get after the configuration also offered a web interface in which all the values that the device provides are mapped. 

Otherwise, the implementation makes no sense to me.

 I think this is a gap. 
-I can configure the device for connection in Tasmota
-The values the device sends are known
-the web interface is customized for thease values after the configuration

but the connection is missing between the read data and the web interface?

I can't close this gap with the "sensor version" either because the scripting is missing?

I do not understand this.

Chriss312, 2021-09-27:
I have found a Note in the Smart Meter Interface description.

If no >M section is found in the script or if the scripting language is not compiled, the driver reverts to the default hardcoded #define definitions. **If no meter script is defined, the default harcoded descriptor interface (deprecated) uses RX GPIO3 for receiving data** and thus may interfere with other Tasmota Defintions without warning. 

I think it should work.

arendst, 2021-09-27:
SDM630 is enabled in the sensors binary.

Connecting the SDM630 modbus interface directly to the wemos won't work. You'll need a RS485 converter. Look for the correct one in the Tasmota documentation.

closed 2021-09-27 by ascillato2