arendst/Tasmota #13236: BME280 returns values that are too low if deepsleep is set up

sohnvonoff, 2021-09-28:
### PROBLEM DESCRIPTION
Have a battery operated ESP8266 with BME280 (and DS18b20) as a climate sensor with Tasmota. 
Deepsleep is active. every 15 minutes the climate data of the BME280 is transmitted with MQTT. 
Since the update from 8.4 to 9.5, the BME280 returns values that are too low. This only happens after waking up from deep sleep. 
If Deepsleep is deactivated, the BME280 runs stably again. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [v] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [v] Tasmota binary firmware version number used: _____
  - [v] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:27:33.679 CMD: Backlog Template; Module; GPIO 255
19:27:33.681 SRC: WebConsole from 192.168.10.25
19:27:33.684 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
19:27:33.765 SRC: Backlog
19:27:33.768 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
19:27:33.775 MQT: /BRE-ET161/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:27:33.981 SRC: Backlog
19:27:33.983 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
19:27:33.988 MQT: /BRE-ET161/stat/RESULT = {"Module":{"18":"Generic"}}
19:27:34.232 SRC: Backlog
19:27:34.235 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
19:27:34.248 MQT: /BRE-ET161/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1312":"DS18x20"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4704":"ADC Input"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: /BRE-ET161/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":33,"Free":478,"Rules":"ON System#Save DO Delay 150 ENDON"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: STATUS0 = {"Status":{"Module":18,"DeviceName":"BRE-ET161","FriendlyName":["Klima-161"],"Topic":"BRE-ET161","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:04:25","StartupUTC":"2021-09-28T18:14:07","Sleep":50,"CfgHolder":4617,"BootCount":73,"BCResetTime":"2020-09-02T15:14:14","SaveCount":4905,"SaveAddress":"F4000"},"StatusFWR":{"Version":"9.5.0(sensors)","BuildDateTime":"2021-06-17T08:26:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"439/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.10.5","LogPort":514,"SSId":["Fuchsbau",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008209","2805C8000000060000005A0A000000000000","01000080","00006000","00000000"]},"StatusMEM":{"ProgramSize":661,"Free":340,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"},"StatusNET":{"Hostname":"BRE-ET161-1478","IPAddress":"192.168.10.161","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.5","Mac":"BC:DD:C2:24:45:C6","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.10.136","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2445C6","MqttUser":"Mqtt-User","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-09-28T18:18:32","Local":"2021-09-28T19:18:32","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:45","Sunset":"18:35"},"StatusSNS":{"Time":"2021-09-28T19:18:32","ANALOG":{"A0":764},"DS18B20-1":{"Id":"000005A022B3","Temperature":14.4},"DS18B20-2":{"Id":"03168348C8FF","Temperature":16.6},"BME280":{"Temperature":14.7,"Humidity":100.0,"DewPoint":14.7,"Pressure":1017.1},"PressureUnit":"hPa","TempUnit":"C"},"StatusSTS":{"Time":"2021-09-28T19:18:32","Uptime":"0T00:04:25","UptimeSec":265,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Fuchsbau","BSSId":"DC:39:6F:86:2B:53","Channel":11,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
--> The output between waking up and falling asleep 

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 4907
00:00:00.056 CFG: CR 439/699, Busy 0
00:00:00.060 CFG: CR 439/699, Busy 0
00:00:00.063 SRC: Restart
00:00:00.064 Project tasmota BRE-ET161 Version 9.5.0(sensors)-2_7_4_9(2021-06-17T08:26:50)
00:00:00.097 DSB: Sensors found 2
00:00:00.109 SHT: Sensor did not ACK command
00:00:00.148 I2C: BME280 found at 0x77
00:00:00.465 WIF: Checking connection...
00:00:00.466 WIF: Attempting connection...
00:00:00.802 WIF: Connecting to AP1 Fuchsbau Channel 11 BSSId DC:39:6F:86:2B:53 in mode 11n as BRE-ET161-1478...
00:00:01.754 WIF: Checking connection...
00:00:01.755 WIF: Attempting connection...
00:00:02.754 WIF: Checking connection...
00:00:02.755 WIF: Attempting connection...
00:00:03.754 WIF: Checking connection...
00:00:03.755 WIF: Attempting connection...
00:00:04.754 WIF: Checking connection...
00:00:04.755 WIF: Attempting connection...
00:00:05.757 WIF: Checking connection...
00:00:05.758 WIF: Connected
00:00:06.010 HTP: Web server active on BRE-ET161-1478 with IP address 192.168.10.161
00:00:06.242 RTC: UTC 2021-09-28T18:23:09, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
19:23:10.028 MQT: Attempting connection...
19:23:10.112 MQT: Connected
19:23:10.116 MQT: /BRE-ET161/tele/LWT = Online (retained)
19:23:10.119 MQT: /BRE-ET161/cmnd/POWER = 
19:23:10.120 MQT: Subscribe to /BRE-ET161/cmnd/#
19:23:10.122 MQT: Subscribe to /tasmotas/cmnd/#
19:23:10.123 MQT: Subscribe to cmnd/DVES_2445C6_fb/#
19:23:10.125 MQT: Unsubscribe from homeassistant/status
19:23:10.334 SRC: MQTT
19:23:10.336 CMD: Grp 0, Cmnd 'DEEPSLEEPTIME', Idx 1, Len 2, Data '30'
19:23:10.341 MQT: /BRE-ET161/stat/RESULT = {"DeepSleepTime":30}
19:23:11.145 DSL: Reset wrong settings wakeup: 0, slip 10000
19:23:11.150 MQT: /BRE-ET161/stat/RESULT = {"DeepSleep":{"Time":"2021-09-28T19:23:30","Epoch":1632853410}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure a esp8266 (D1 mini) w. BME2800 and DS18B20 
Configure Deppsleep (e.g. 15min)

without deepsleep are all values ok

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Before the update from 8.4 to 9.5, the measured values were always plausible 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-10-04:
@sohnvonoff How would you qualify "too low" ? By 0.1° by 1° by 10° ?
I can't see much code changes in recently in the BMP/BME driver that could be impacted by DeepSleep.
Is your BME280 always powered or are you using some circuitry to switch it off between the measures ?

Milbert-R, 2021-10-05:
I made a test with  connected DHT22 and BME280.
The screenshoots will show, that output for BME280 in webUI will change during NodeMCU is in deepsleep.
Wrong data is _NOT_ sent via MQTT.

NodeMCU is waked up:
[image]

NodeMCU is in deepsleep:
[image]

Weblog 4 about this time from wakeup to deepsleep: 

`00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: aus Flash geladen am F6, Anzahl 514
00:00:00.055 CFG: CR 553/699, Busy 0
00:00:00.060 MSH: Initialized
00:00:00.061 SRC: Restart
00:00:00.062 Projekt RkM - RkM_NodeMCU_Deepsleep_Test Version 9.5.0.4(tasmota)-2_7_4_9(2021.08.02 17:55:13)
00:00:00.064 DHT: (v5) Sensor gefunden 1
00:00:00.076 I2C: BME280 gefunden bei 0x76
00:00:00.183 WIF: Prüfe Verbindung...
00:00:00.184 WIF: Verbindungsversuch...
00:00:00.520 WIF: verbinden mit AP1 RKMWLAN Channel 5 BSSId 44:4E:6D:A0:7F:20 in Modus 11n als RkM-NodeMCU-Deepsleep-Test-7681...
00:00:01.753 WIF: Prüfe Verbindung...
00:00:01.754 WIF: verbunden
00:00:02.006 HTP: Web-Server aktiv bei RkM-NodeMCU-Deepsleep-Test-7681 mit IP-Adresse 192.168.0.219
00:00:02.469 RTC: UTC 2021-10-05T15:23:59, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
17:24:00.048 MQT: Verbindungsversuch...
17:24:00.065 MQT: verbunden
17:24:00.068 MQT: tele/RkM_NodeMCU_Deepsleep_Test/LWT = Online (beibehalten)
17:24:00.071 MQT: cmnd/RkM_NodeMCU_Deepsleep_Test/POWER = 
17:24:00.072 MQT: abonniere cmnd/RkM_NodeMCU_Deepsleep_Test/#
17:24:00.074 MQT: abonniere cmnd/tasmotas/#
17:24:00.077 MQT: abonniere cmnd/RkM_NodeMCU_Deepsleep_Test_fb/#
17:24:00.078 MQT: löse abo. von homeassistant/status
17:24:04.494 MQT: tele/RkM_NodeMCU_Deepsleep_Test/STATE = {"Time":"2021-10-05T17:24:04","Uptime":"0T00:00:09","UptimeSec":9,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"RKMWLAN","BSSId":"44:4E:6D:A0:7F:20","Channel":5,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:24:04.530 MQT: tele/RkM_NodeMCU_Deepsleep_Test/SENSOR = {"Time":"2021-10-05T17:24:04","Switch1":"OFF","AM2301":{"Temperature":24.2,"Humidity":56.1,"DewPoint":14.9},"BME280":{"Temperature":23.6,"Humidity":48.2,"DewPoint":12.0,"Pressure":966.8,"SeaPressure":1012.3},"Global":{"Temperature":23.6,"Humidity":48.2,"DewPoint":12.0,"Pressure":966.8,"SeaPressure":1012.3},"PressureUnit":"hPa","TempUnit":"C"}
17:24:09.442 MQT: tasmota/discovery/3C6105CFDE01/config = {"ip":"192.168.0.219","dn":"RkM_NodeMCU_Deepsleep_Test","fn":["RkM_NodeMCU_Deepsleep_Test",null,null,null,null,null,null,null],"hn":"RkM-NodeMCU-Deepsleep-Test-7681","mac":"3C6105CFDE01","md":"Generic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.5.0.4","t":"RkM_NodeMCU_Deepsleep_Test","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0],"swc":[1,-1,-1,-1,-1,-1,-1,-1],"swn":["Switch1",null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":1,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (beibehalten)
17:24:09.453 MQT: tasmota/discovery/3C6105CFDE01/sensors = {"sn":{"Time":"2021-10-05T17:24:09","Switch1":"OFF","AM2301":{"Temperature":24.2,"Humidity":56.0,"DewPoint":14.9},"BME280":{"Temperature":23.6,"Humidity":48.2,"DewPoint":12.0,"Pressure":966.8,"SeaPressure":1012.3},"Global":{"Temperature":23.6,"Humidity":48.2,"DewPoint":12.0,"Pressure":966.8,"SeaPressure":1012.3},"PressureUnit":"hPa","TempUnit":"C"},"ver":1} (beibehalten)
17:24:18.780 WIF: Prüfe Verbindung...
17:24:38.802 WIF: Prüfe Verbindung...
17:24:58.797 WIF: Prüfe Verbindung...
17:25:04.423 MQT: tele/RkM_NodeMCU_Deepsleep_Test/STATE = {"Time":"2021-10-05T17:25:04","Uptime":"0T00:01:09","UptimeSec":69,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"RKMWLAN","BSSId":"44:4E:6D:A0:7F:20","Channel":5,"Mode":"11n","RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:25:04.454 MQT: tele/RkM_NodeMCU_Deepsleep_Test/SENSOR = {"Time":"2021-10-05T17:25:04","Switch1":"OFF","AM2301":{"Temperature":24.2,"Humidity":55.8,"DewPoint":14.8},"BME280":{"Temperature":23.6,"Humidity":48.2,"DewPoint":12.0,"Pressure":966.7,"SeaPressure":1012.3},"Global":{"Temperature":23.6,"Humidity":48.2,"DewPoint":12.0,"Pressure":966.7,"SeaPressure":1012.3},"PressureUnit":"hPa","TempUnit":"C"}
17:25:04.501 MQT: stat/RkM_NodeMCU_Deepsleep_Test/RESULT = {"DeepSleep":{"Time":"2021-10-05T17:25:00","Epoch":1633447500}}`

barbudor, 2021-10-05:
I still don't understand
There would be no data displayed on the web page when the NodeMCU is in deepsleep because the ESP is sleeping
And you just said that the correct values are send over MQTT

sfromis, 2021-10-05:
Those air pressure numbers in your last screenshot match what I see when the sensor is not in a correct "state". Haven't explored details of what happens, but I suspect that it may be more likely to happen with some sort of power issue, which easily could be the case with a battery setup. When I've seen similar, the readings have usually been fixed.

Not sure what you mean with "NodeMCU is in deepsleep".

Milbert-R, 2021-10-05:
I just flashed precompiled **[http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz](http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz)**

Teleperiod set to 300

Same result as  above...

The second screenshot is the display of the website during deepsleep.

The unit is not battery powered but powered by a 5v power supply.

Unit is in deepsleep means the time between

`MQT: stat/RkM_NodeMCU_Deepsleep_Test/RESULT = {"DeepSleep":{"Time":"2021-10-05T18:24:00","Epoch":1633454640}}
`

and next
```00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 544
00:00:00.055 CFG: CR 553/699, Busy 0
00:00:00.060 SRC: Restart
```

Weblog 4:

```
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 544
00:00:00.055 CFG: CR 553/699, Busy 0
00:00:00.060 SRC: Restart
00:00:00.062 Project tasmota RkM_NodeMCU_Deepsleep_Test Version 9.5.0(sensors)-2_7_4_9(2021-06-17T08:26:50)
00:00:00.064 DHT: (v5) Sensors found 1
00:00:00.075 SHT: Sensor did not ACK command
00:00:00.116 I2C: BME280 found at 0x76
00:00:00.460 WIF: Checking connection...
00:00:00.460 WIF: Attempting connection...
00:00:00.797 WIF: Connecting to AP1 RKMWLAN Channel 5 BSSId 44:4E:6D:A0:7F:20 in mode 11n as RkM_NodeMCU_Deepsleep_Test-7681...
00:00:01.752 WIF: Checking connection...
00:00:01.753 WIF: Connected
00:00:02.005 HTP: Web server active on RkM_NodeMCU_Deepsleep_Test-7681 with IP address 192.168.0.219
00:00:02.221 RTC: UTC 2021-10-05T17:23:02, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
18:23:03.093 MQT: Attempting connection...
18:23:03.123 MQT: Connected
18:23:03.127 MQT: tele/RkM_NodeMCU_Deepsleep_Test/LWT = Online (retained)
18:23:03.129 MQT: cmnd/RkM_NodeMCU_Deepsleep_Test/POWER = 
18:23:03.130 MQT: Subscribe to cmnd/RkM_NodeMCU_Deepsleep_Test/#
18:23:03.134 MQT: Subscribe to cmnd/tasmotas/#
18:23:03.135 MQT: Subscribe to cmnd/RkM_NodeMCU_Deepsleep_Test_fb/#
18:23:03.137 MQT: Unsubscribe from homeassistant/status
18:23:07.169 MQT: tele/RkM_NodeMCU_Deepsleep_Test/STATE = {"Time":"2021-10-05T18:23:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"RKMWLAN","BSSId":"44:4E:6D:A0:7F:20","Channel":5,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:23:07.198 MQT: tele/RkM_NodeMCU_Deepsleep_Test/SENSOR = {"Time":"2021-10-05T18:23:07","AM2301":{"Temperature":25.1,"Humidity":56.1,"DewPoint":15.7},"BME280":{"Temperature":25.5,"Humidity":46.2,"DewPoint":13.0,"Pressure":967.3,"SeaPressure":1012.9},"PressureUnit":"hPa","TempUnit":"C"}
18:23:07.227 MQT: stat/RkM_NodeMCU_Deepsleep_Test/RESULT = {"DeepSleep":{"Time":"2021-10-05T18:24:00","Epoch":1633454640}}
```

sfromis, 2021-10-05:
Tasmota does not update the web page display (or anything else) while in deep sleep. It cannot be "between" as you describe. Is it first update after wakeup, or last update before sleep?

My comment about potential power issue was not restricted to battery operation, but generally.

TD-er, 2021-10-20:
Please be aware that the Bosch sensors have a moving average filter.
This can be configured in the chip itself.
It does cause some delay in response and especially on the first N samples the value can be (far) too low, depending on this filter setting.

No idea how Tasmota does use the Bosch sensors, but I guess that if you need a short measurement interval right after boot, you may need to perform N readings before accepting the value. (or change the filter setting :) )