arendst/Tasmota #13251: Sonoff Dual R3 reboot after memory heap leak.. (4 pcs)

itProfi, 2021-09-30:
### PROBLEM DESCRIPTION
My 4 psc Sonoff Dual R3 reboot after memoty leak..

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
18:59:44.320 RUL: Rule1 on system#boot do Backlog RuleTimer1 60; RuleTimer2 20; RuleTimer3 70 endon on Time#Initialized do Backlog var2 0; var3 0; var10 0; mem1; mem2; mem3; mem10; MI32Option2 1 endon on Rules#Timer=1 do Backlog RuleTimer1 60 endon on Rules#Timer=2 do Backlog RuleTimer2 20; var2=%mem2%+3; var3=%mem3%-3 endon on Rules#Timer=3 do if (var10==0) restart 1 else RuleTimer3 70 endif endon on MJ_HT_V131425e#temperature do var10=%value% endon on MJ_HT_V131425e#temperature<=%mem3% do if ((mem1==1) and (%value%>var3))
18:59:44.328 RUL: Rule1 +power1 1 endif endon on MJ_HT_V131425e#temperature>=%mem2% do if ((mem1==1) and (%value%<var2)) power1 0 endif endon
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
8:51:20.552 MQT: stat/tasmota_2B8300/STATUS = {"Status":{"Module":0,"DeviceName":"1-floor-thermostat-42:5e","FriendlyName":["1-floor-thermostat-42:5e",""],"Topic":"tasmota_2B8300","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:51:20.632 MQT: stat/tasmota_2B8300/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T02:39:45","StartupUTC":"2021-09-30T13:11:35","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-09-30T09:33:16","SaveCount":304}}
18:51:20.694 MQT: stat/tasmota_2B8300/STATUS2 = {"StatusFWR":{"Version":"9.5.0.9(bluetooth)","BuildDateTime":"2021-09-29T18:47:41","Core":"1_0_7_4","SDK":"v3.3.5-245-g95097b38d","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"450/699"}}
18:51:20.751 MQT: stat/tasmota_2B8300/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"192.168.1.36","LogPort":514,"SSId":["MALINOVAYA",""],"TelePeriod":50,"Resolution":"558180C0","SetOption":["04208008","2805C8000100060000005A00000000000000","00000200","00406000","00000002"]}}
18:51:20.830 MQT: stat/tasmota_2B8300/STATUS4 = {"StatusMEM":{"ProgramSize":1157,"Free":1856,"Heap":63,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8D800583","00000001","00000007","000000C0","00000000","00004080","00200000","00000024"],"Drivers":"1,2,3,4,8,9,10,12,38,50,79,85","Sensors":"1,2,3,52,62,127"}}
18:51:20.897 MQT: stat/tasmota_2B8300/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2B8300-0768","IPAddress":"192.168.1.65","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"08:3A:F2:2B:83:00","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":17.0}}
18:51:20.954 MQT: stat/tasmota_2B8300/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1885,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2B8300","MqttUser":"my_mqtt_client_1","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:51:21.033 MQT: stat/tasmota_2B8300/STATUS7 = {"StatusTIM":{"UTC":"2021-09-30T15:51:21","Local":"2021-09-30T18:51:21","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+03:00"}}
18:51:21.091 MQT: stat/tasmota_2B8300/STATUS9 = {"StatusPTH":{"PowerDelta":[102,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:51:21.162 MQT: stat/tasmota_2B8300/STATUS10 = {"StatusSNS":{"Time":"2021-09-30T18:51:21","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2021-09-29T16:50:35","Total":0.010,"Yesterday":0.000,"Today":0.010,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":225,"Current":[0.000,0.000]},"ESP32":{"Temperature":55.0},"TempUnit":"C"}}
18:51:21.241 MQT: stat/tasmota_2B8300/STATUS11 = {"StatusSTS":{"Time":"2021-09-30T18:51:21","Uptime":"0T02:39:46","UptimeSec":9586,"Heap":65,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":69,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MMM","BSSId":"67:FF:30:0F:EE:96","Channel":11,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}
18:51:21.318 MQT: stat/tasmota_2B8300/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"LoadProhibited","EPC":"40091db9","EXCVADDR":"00000000","CallChain":["40091db6","4012c711","401354ce","4013550a","40120053","401200f2","400d8b46","400e0ca2","400f4da4","400f5242","400f5398","400f56e3","400e280f","400fd630","400fd693","40126b69"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-10-01:
Pls try latest dev release available after 2 hours and report back.

I disabled a recently enabled bluetooth feature which might cause the memory leak.

itProfi, 2021-10-02:
Thank you for your answer. The size of the firmware image decreased by 9 KB, but, apparently, the leakage did not stop ..
[image]

itProfi, 2021-10-02:
After harware reset and 2 hours -same result

itProfi, 2021-10-02:
[image]

arendst, 2021-10-03:
I see a decline in heap size too:

[image]

I guess it's time for the blutooth guru's to kick in.

arendst, 2021-10-03:
Logging a bit longer reveals no heap memory leak though:

[image]

arendst, 2021-10-03:
I just enabled your rules and see a gradual decline in heap space. Looks like (continuous calls to) rules is causing heap issues.

Will test further.

itProfi, 2021-10-03:
I see..( In esp8266 these rules have not resulteddecline in heap space ..

itProfi, 2021-10-03:
> I just enabled your rules and see a gradual decline in heap space. Looks like (continuous calls to) rules is causing heap issues.
> 
> [image]
> 
> Will test further.

It's my small thermostats :)

itProfi, 2021-10-04:
In my case 4-th experimental Sonoff Dual R3 - haven't any rules, but ble turn on and i see heap leak

itProfi, 2021-10-04:
With this commit, in my case leak was fater.
[image]
.