arendst/Tasmota #13262: Compile error with LVGL on and Berry off

mikep1998, 2021-10-02:
### PROBLEM DESCRIPTION
Compile error with LVGL on and Berry off with latest Tasmota code.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: ___VERSION = 0x09050009__
  - [] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Archiving .pio\build\tasmota32-userconfig_thermostat\libf22\libAdafruit_SGP30-1.2.0.a
In file included from lib\libesp32\Berry\default\be_lvgl_cb.c:5:0:
lib\libesp32\Berry\src/be_constobj.h:113:24: error: 'be_const_str_lv_cb' undeclared here (not in a function)
     .name = (bstring*)&be_const_str_##_name_                    \
                        ^
lib\libesp32\Berry\default\../generate/be_fixed_be_lvgl_cb.h:15:20: note: in expansion of macro 'be_define_const_class'
 BE_EXPORT_VARIABLE be_define_const_class(
                    ^
lib\libesp32\Berry\src/be_constobj.h:113:24: error: 'be_const_str_lv_group_focus_cb' undeclared here (not in a function)
     .name = (bstring*)&be_const_str_##_name_                    \
                        ^
lib\libesp32\Berry\default\../generate/be_fixed_be_lv_group_focus_cb.h:12:20: note: in expansion of macro 'be_define_const_class'
 BE_EXPORT_VARIABLE be_define_const_class(
                    ^
lib\libesp32\Berry\src/be_constobj.h:113:24: error: 'be_const_str_lv_event_cb' undeclared here (not in a function)
     .name = (bstring*)&be_const_str_##_name_                    \
                        ^
lib\libesp32\Berry\default\../generate/be_fixed_be_lv_event_cb.h:12:20: note: in expansion of macro 'be_define_const_class'
 BE_EXPORT_VARIABLE be_define_const_class(
                    ^
lib\libesp32\Berry\src/be_constobj.h:113:24: error: 'be_const_str_lv_signal_cb' undeclared here (not in a function)
     .name = (bstring*)&be_const_str_##_name_                    \
                        ^
lib\libesp32\Berry\default\../generate/be_fixed_be_lv_signal_cb.h:12:20: note: in expansion of macro 'be_define_const_class'
 BE_EXPORT_VARIABLE be_define_const_class(
                    ^
lib\libesp32\Berry\src/be_constobj.h:113:24: error: 'be_const_str_lv_design_cb' undeclared here (not in a function)
     .name = (bstring*)&be_const_str_##_name_                    \
      
### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-10-02:
As I understand it, the LVGL support is based on using Berry. Thus it makes sense that you cannot remove this.

s-hadinger, 2021-10-02:
Well. You can still use lvgl without Berry if you're using LVGL's C API. 

I will look at it

mikep1998, 2021-10-02:
@s-hadinger  It compiled fine with the first pre-release I downloaded. I did not get a chance to test the second update you did. That should narrow it down to a few files.

s-hadinger, 2021-10-03:
I can't reproduce. `be_const_str_lv_cb` doesn't exist anymore. I believe you have old stuff remaining in your directories. Please pull a fresh new copy from github.

mikep1998, 2021-10-03:
Sorry about that. I cleaned and got a fresh copy and its working now.
User error.

closed 2021-10-03 by mikep1998