arendst/Tasmota #13282: New Sonoff RF Bridge Board Layout (R2 v2.2)

andrewmeepos, 2021-10-06:
### PROBLEM DESCRIPTION
There is a new configuration of the Sonoff RF bridge board, v2.2, which as a different configuration than past versions. While I was able to flash Tasmota using this guide:
https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/

I was not able to figure out which pins to connect to flash Portisch.

Here are images of the new board.

[image]
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _Sonoff RF Bridge____
- [ ] Tasmota binary firmware version number used: __9.5.0___
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _ESP-flasher____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
N/A

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
N/A

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
N/A

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-10-06:
It has been identified that this new version is more than a layout change as they change the whole MCU part.
It is no more an EFM processor and Portish firmware may not be compatible as-is. 
Also no one as yet digged into the communication between the ESP and this new MCU which may be I2C instead of serial.

You are currently on a totally new and unexplored land until someone with the device, the skills, the time and the willingness dives into this.

sedlacekdavid, 2021-10-06:
Same problem here. 

It's a new design without EFM processor. Seems like the first chip has remained same and the second one is different. Maaan I needed an RF gateway so ordered this :/

closed 2021-10-06 by blakadder