arendst/Tasmota #13303: Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1) when SR04 enabled.

mikep1998, 2021-10-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32___
- [x] Tasmota binary firmware version number used: _9.0.5.9____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:41:27.492 MQT: stat/jacuzzi_Tub_Down/RESULT = {"NAME":"Jacuzzi_Tub_Down","GPIO":[1,1,1,1,193,1,0,0,1,160,416,1,194,1,352,195,0,1312,1856,1888,0,258,259,260,0,0,0,0,256,257,4704,1,4737,0,0,1],"FLAG":0,"BASE":1}
18:41:27.731 MQT: stat/jacuzzi_Tub_Down/RESULT = {"Module":{"0":"Jacuzzi_Tub_Down"}}
18:41:27.993 MQT: stat/jacuzzi_Tub_Down/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"193":"Switch_n2"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"160":"Switch1"},"GPIO14":{"416":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"194":"Switch_n3"},"GPIO17":{"0":"None"},"GPIO18":{"352":"Counter1"},"GPIO19":{"195":"Switch_n4"},"GPIO20":{"0":"None"},"GPIO21":{"1312":"DS18x20"},"GPIO22":{"1856":"SR04 Tri/TX"},"GPIO23":{"1888":"SR04 Ech/RX"},"GPIO24":{"0":"None"},"GPIO25":{"258":"Relay_i3"},"GPIO26":{"259":"Relay_i4"},"GPIO27":{"260":"Relay_i5"},"GPIO32":{"256":"Relay_i1"},"GPIO33":{"257":"Relay_i2"},"GPIO34":{"4704":"ADC Input1"},"GPIO35":{"0":"None"},"GPIO36":{"4737":"ADC Temp2"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:42:26.412 CMD: status 0
18:42:26.429 MQT: stat/jacuzzi_Tub_Down/STATUS = {"Status":{"Module":0,"DeviceName":"jacuzzi_Tub_Down","FriendlyName":["jacuzzi_Tub_Down","jacuzzi_Tub_Down2","jacuzzi_Tub_Down3","jacuzzi_Tub_Down4",""],"Topic":"jacuzzi_Tub_Down","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"jacuzzi_Tub_Down_switch","SwitchMode":[2,15,15,15,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:42:26.444 MQT: stat/jacuzzi_Tub_Down/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-minimal.bin","RestartReason":"Software reset CPU","Uptime":"0T00:48:43","StartupUTC":"2021-10-09T00:53:43","Sleep":250,"CfgHolder":9889,"BootCount":1,"BCResetTime":"2021-10-08T17:53:39","SaveCount":54}}
18:42:26.458 MQT: stat/jacuzzi_Tub_Down/STATUS2 = {"StatusFWR":{"Version":"9.5.0.9(tasmota)","BuildDateTime":"2021-10-08T17:52:38","Core":"1_0_7_4","SDK":"v3.3.5-245-g95097b38d","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"542/699"}}
18:42:26.470 MQT: stat/jacuzzi_Tub_Down/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mike1.1.1","mike1.1.2"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A0108","2805C8000100060000005A00000000000000","000022C0","00006000","00004000"]}}
18:42:26.491 MQT: stat/jacuzzi_Tub_Down/STATUS4 = {"StatusMEM":{"ProgramSize":951,"Free":1856,"Heap":143,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F50C786","00100001","020000C3","00000000","40000901","00000080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,21,24,26,27,29,38,50,91","Sensors":"1,2,5,6,22,93,127"}}
18:42:26.510 MQT: stat/jacuzzi_Tub_Down/STATUS5 = {"StatusNET":{"Hostname":"jacuzzi-Tub-Down","IPAddress":"192.168.1.105","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"24:6F:28:9E:A6:EC","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
18:42:26.527 MQT: stat/jacuzzi_Tub_Down/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9EA6EC","MqttUser":"pi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:42:26.544 MQT: stat/jacuzzi_Tub_Down/STATUS7 = {"StatusTIM":{"UTC":"2021-10-09T01:42:26","Local":"2021-10-08T18:42:26","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"-07:00","Sunrise":"07:11","Sunset":"18:41"}}
18:42:26.658 MQT: stat/jacuzzi_Tub_Down/STATUS10 = {"StatusSNS":{"Time":"2021-10-08T18:42:26","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","COUNTER":{"C1":0},"ANALOG":{"A1":1075,"Temperature2":65.9},"DS18B20-1":{"Id":"3C01D6074A0D","Temperature":68.4},"DS18B20-2":{"Id":"01131C1E0C73","Temperature":67.9},"Jacuzzi":{"Drain":"Open","Level":0,"Target_Temp":0,"HotWaterTemp":68,"TubWaterTemp":68,"StateNum":"1","FillStateNum":"1","State":"Idle","FillState":"Idle","Depth":0.000000},"ESP32":{"Temperature":128.0},"TempUnit":"F"}}
18:42:26.684 MQT: stat/jacuzzi_Tub_Down/STATUS11 = {"StatusSTS":{"Time":"2021-10-08T18:42:26","Uptime":"0T00:48:43","UptimeSec":2923,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":67,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","PWM":{"PWM1":0},"Wifi":{"AP":2,"SSId":"mike1.1.2","BSSId":"9C:D3:6D:9E:41:3C","Channel":10,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:11"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:53:00.112 >
16:53:00.113 > 00:00:00.003 HDW: ESP32-D0WDQ6
16:53:00.168 > 00:00:00.057 UFS: FlashFS mounted with 296 kB free
16:53:00.236 > 00:00:00.125 CFG: Loaded from File, Count 10
16:53:00.248 > 00:00:00.138 QPC: Count 1
16:53:00.257 > 00:00:00.147 Project tasmota - jacuzzi_Tub_Down Version 9.5.0.9(tasmota)-1_0_7_4(2021-10-08T16:49:27)
16:53:00.267 > 00:00:00.154 RSL: RESULT = {"AdcParam2":[2,33000,8700,3350]}
16:53:01.579 > Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
16:53:01.586 > Core 1 register dump:
16:53:01.588 > PC      : 0x400827e3  PS      : 0x00050034  A0      : 0x40089148  A1      : 0x3ffbf8a0
16:53:01.597 > A2      : 0x3ffbdef0  A3      : 0x00000002  A4      : 0x00000001  A5      : 0x4009091c
16:53:01.603 > A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ff6e000
16:53:01.611 > A10     : 0x00000000  A11     : 0x00000180  A12     : 0x00000000  A13     : 0x3ffb1e30
16:53:01.619 > A14     : 0x3ffca2e0  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000006
16:53:01.627 > EXCVADDR: 0x00000000  LBEG    : 0x4008b4a8  LEND    : 0x4008b4bb  LCOUNT  : 0x00000000
16:53:01.636 > Core 1 was running in ISR context:
16:53:01.638 > EPC1    : 0x4011096f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400827e3
16:53:01.646 >
16:53:01.646 > ELF file SHA256: 0000000000000000
16:53:01.650 >
16:53:01.650 > Backtrace: 0x400827e0:0x3ffbf8a0 0x40089145:0x3ffbf8d0 0x4011096c:0x3ffb1e90 0x4010becd:0x3ffb1eb0 0x4010494c:0x3ffb1ed0 0x40104961:0x3ffb1ef0 0x400e6e76:0x3ffb1f10 0x400e6edd:0x3ffb1f30 0x400faad8:0x3ffb1f50 0x400e7786:0x3ffb1f70 0x400fbe09:0x3ffb1f90 0x400fbe27:0x3ffb1fb0 0x40111249:0x3ffb1fd0
16:53:02.061 >   #0  0x400827e0:0x3ffbf8a0 in _uart_isr at C:\Users\mike\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-uart.c:568
[... 51 lines cut ...]
      sr04_type = 2;
    }
  } else {
    sr04_type = 1;
  }
  if (sr04_type < 2) {
    delete sonar_serial;
    sonar_serial = nullptr;
    if (-1 == sr04_trig_pin) {
      sr04_trig_pin = Pin(GPIO_SR04_ECHO);  // if GPIO_SR04_TRIG is not configured use single PIN mode with GPIO_SR04_ECHO only
    }

    sonar = new NewPing(sr04_trig_pin, sr04_echo_pin, SR04_MAX_SENSOR_DISTANCE);
  } else {
    if (sonar_serial->hardwareSerial()) {
      ClaimSerial();
    }
  }

  AddLog(LOG_LEVEL_INFO,PSTR("SR4: Mode %d"), sr04_type);
  return sr04_type;
}
```

**(Please, remember to close the issue when the problem has been addressed)**

mikep1998, 2021-10-09:
After using the prebuild tasmota code and adding just the SR04_TRIG and SR04_ECHO the system boots fine with no panics. So the panic may be in my code. But the backtrace is not indicating the correct locals.
I'm using Visual Code and running device monitor to get the backtrace decoded. 
As you can see above the trace is not completely decoding to the correct files as in:
'''
Sr04TModeDetect() at C:/Users/mike/Documents/PlatformIO/Projects/Tasmota-development/tasmota/xdrv_04_light.ino:277
'''
says 'Sr04TModeDetect() ' routine is in 'xdrv_04_light.ino:277' which is not correct. 
Anyone know how to resolve this issue? Or how to properly resolve the panic?

Thanks

mikep1998, 2021-10-09:
yes, sr04 is mode 1. and both trig and echo are enabled in the config. 
I have not made any changes to any of the serial  or ping code. I have added my own driver for the jacuzzi logic to drive all the sensors.

The problem is intermittent it seems to come and go. It was working and the delay I added but then it started failing again. 

Here is another output of the serial log when it happens.

```
E (397) psram: PSRAM ID read error: 0xffffffff
E (397) spiram: SPI RAM enabled but initialization failed. Bailing out.

00:00:00.002 HDW: ESP32-D0WDQ6 
00:00:00.031 UFS: FlashFS mounted with 296 kB free
00:00:00.063 CFG: Loaded from File, Count 15
00:00:00.067 FRC: Some settings have been reset (2)
00:00:00.076 Project tasmota - jacuzzi_Tub_Down Version 9.5.0.9(tasmota)-1_0_7_4(2021-10-09T01:13:48)
00:00:00.084 RSL: RESULT = {"AdcParam2":[2,33000,8700,3350]}
00:00:00.211 SR4: Reading CRC error.
00:00:04.610 SR4: Reading CRC error.
00:00:08.760 SR4: Reading CRC error.
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40082789  PS      : 0x00050034  A0      : 0x40089148  A1      : 0x3ffbf8a0  
A2      : 0x3ffbdec8  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4009091c  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ff40000  A9      : 0x3ff6e000  
A10     : 0x000002bc  A11     : 0x000002bc  A12     : 0x00000000  A13     : 0x3ffb1e30  
A14     : 0x3ffca2e0  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008c285  LEND    : 0x4008c295  LCOUNT  : 0xfffffff9  
Core 1 was running in ISR context:
EPC1    : 0x401109af  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082789

ELF file SHA256: 0000000000000000

Backtrace: 0x40082786:0x3ffbf8a0 0x40089145:0x3ffbf8d0 0x401109ac:0x3ffb1e90 0x4010bf0d:0x3ffb1eb0 0x4010498c:0x3ffb1ed0 0x401049a1:0x3ffb1ef0 0x400e6e8e:0x3ffb1f10 0x400e6efd:0x3ffb1f30 0x400fab1c:0x3ffb1f50 0x400e77a1:0x3ffb1f70 0x400fbe55:0x3ffb1f90 0x400fbe73:0x3ffb1fb0 0x40111289:0x3ffb1fd0

Core 0 register dump:
PC      : 0x401826d6  PS      : 0x00060b34  A0      : 0x8013eeba  A1      : 0x3ffbc4f0  
A2      : 0x00000000  A3      : 0x40086ce0  A4      : 0x00060920  A5      : 0x3ffbbb80  
A6      : 0x00000000  A7      : 0x3ffbc1e0  A8      : 0x8013dfba  A9      : 0x3ffbc4c0  
A10     : 0x00000000  A11     : 0x3ffbe050  A12     : 0x8008eb08  A13     : 0x3ffbc3f0  
A14     : 0x00000000  A15     : 0x3ffbc1e0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401826d3:0x3ffbc4f0 0x4013eeb7:0x3ffbc510 0x4008fab9:0x3ffbc530

Rebooting...
ets Jun  8 2016 00:22:57

``` 

This is the first error. It continues to reboot and after awhile it boots up without the GPIOs configured. 
If I remove the SR04_TRIG and SR04_ECHO everything works fine.

Any ideas would much appreciated.

mikep1998, 2021-10-09:
@barbudor yes I understand. I will start a discussion. Does it make a difference that it works fine with tasmota code dated 10/3/2021 4:28 AM but fails with tasmota code dated 10/7/2021 1:02 PM.  Something has changed in the tasmota code to make this fail.

mikep1998, 2021-10-10:
@barbudor today the issue is gone. Now it works with the latest level of tasmota and works with the 10/7/2021 level??? flaky.
thanks for you help

mikep1998, 2021-10-10:
Well it came back and I discovered the problem was from just enabling the SR04 in tasmota without actually connecting an SR04 to the pins. So I used a resister to on the ECHO pin to pull it up. Now everything is fine. I suppose the 'sonor_serial' is using a the input without the pull up. So just noise on the wire was causing the check sum errors and the panics. 
So no corrupt memory.

barbudor, 2021-10-10:
I will look further in the coming days but it looks like delete the TasmotaSerial object leave the ISR enabled. It shouldn't.
No sure if the problem is in Tasmota or in Arduino or in Espressif

mikep1998, 2021-10-10:
The one thing I noticed from the logs was on a failure boot log indicated 1 or 2 CRC Errors just prior to the panic
```
00:00:00.055 RSL: RESULT = {"AdcParam2":[2,33000,8700,3350]}
00:00:01.843 SR4: Reading CRC error.
E (14461) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
```

When the boot sometimes completed sucessfully the log had 3 'CRC error' then 'SR04: Mode 1'

I wondered why the CRC errors. So I added the pullup to the ECHO pin.
Then the logs read no CRC errors just the Mode 1.

```
00:00:00.002 HDW: ESP32-D0WDQ6 
00:00:00.054 UFS: FlashFS mounted with 296 kB free
00:00:00.108 CFG: Loaded from File, Count 932
00:00:00.122 QPC: Count 1
00:00:00.131 Project tasmota - jacuzzi_Tub_Down Version 9.5.0.9(tasmota)-1_0_7_4(2021-10-09T21:02:36)
00:00:00.139 RSL: RESULT = {"AdcParam2":[2,33000,8700,3350]}
00:00:01.031 SR4: Mode 1
```
Good luck in finding the problem. I don't have access to source code for low level code.