arendst/Tasmota #13316: Cannot set certain topic or MqttClient.

revk, 2021-10-10:
### PROBLEM DESCRIPTION

I have several Shelly1 running tasmota 9.5.0, all good, except...

On one of them I am unable to set the topic I want. I am trying to set `U2Kitchen`.
I can set `U2Kitche` or `U2Kitchens`, or, well, almost anything else, but trying U2Kitchen comes back as `tasmota_%06X`.

```
cmnd/U2Kitchens/Topic U2Kitchen
stat/U2Kitchens/RESULT {"Topic":"tasmota_%06X"}
```

I have a similar problem on a couple of others where I am unable to set an `MqttClient` of `Office2`, it just comes back as `DVES_%06X`.

I'm at a loss...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

I tried several topics, and was able to set them, e.g. `U2Kitchens` worked, but not `U2Kitchen`.

```
cmnd/U2Kitchens/Topic U2Kitchen
stat/U2Kitchens/RESULT {"Topic":"tasmota_%06X"}
```

I tried on some other units (Shelly1) and could not change the topic at all on several of them, which is rather odd.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect to be able to set a Topic or MqttClient

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-10-10:
Reading the docs, you could have seen the stipulation that Topic and MqttClient should not have the same value, causing the reset you see.

revk, 2021-10-10:
I'll go read the docs more, I did try searching for this before raising the issue, but I did not realise it was a "can't be the same" issue, more of a "can't save a specific value, on just a couple of my devices", hence not finding it.

It does seem odd though, the device has a "name" that I use for the topic, and for hostname for DHCP, and insisting it has a different name for mqtt client name just seems crazy. I'll see if I can find a better explanation in the docs.

revk, 2021-10-10:
So I am trying to change to be different,  but some of them will not let me change the Topic or MqttClient from what they are even to something totally different, even when they are currently the same as each other. All seems a big confused. At the very least it would be good if trying this returned a message saying you can't do it rather than just refusing or setting to a default, and if it did allow they to be set different once they are set the same!