arendst/Tasmota #13317: Slower responses to MQTT in 9.5.0

revk, 2021-10-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have made some simple tools to manage config, and I send a sequence of messages via MQTT to get values.

Running 9.3 the responses are very quick, I can tens of settings per second, so quickly go through a lot of settings on each device.

Running 9.5 the responses seem to all be 200ms or much more, making the whole process very slow.

I have tried option127 and sleep 0, no help.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Simple example.

```
2021-10-10 11:56:27.604250:cmnd/Office2/Topic 
2021-10-10 11:56:27.965310:stat/Office2/RESULT {"Topic":"Office2"}
2021-10-10 11:56:27.968255:cmnd/Office2/MqttHost 
2021-10-10 11:56:28.271866:stat/Office2/RESULT {"MqttHost":"mqtt.iot"}
```

The timestamps show 361ms and 303ms for those two commands to get a response. On 9.3 the time is much lower (tens of ms).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Would be nice to be as fast as it used to be, or some option to make it so.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-10-10:
I cannot reproduce on v9.5.0.9 using this script:

```
#!/bin/sh
SLEEP=0.0
DEVICE=demo
mosquitto_pub -t cmnd/$DEVICE/topic -n
sleep $SLEEP
mosquitto_pub -t cmnd/$DEVICE/mqtthost -n
sleep $SLEEP
mosquitto_pub -t cmnd/$DEVICE/topic -n
sleep $SLEEP
mosquitto_pub -t cmnd/$DEVICE/grouptopic -n
```
responses on both esp32 and esp8266 are swift:
```
13:08:05.232 SRC: MQTT
13:08:05.233 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 0, Data ''
13:08:05.245 MQT: stat/esp32d/RESULT = {"Topic":"esp32d"}
13:08:05.283 SRC: MQTT
13:08:05.284 CMD: Grp 0, Cmnd 'MQTTHOST', Idx 1, Len 0, Data ''
13:08:05.290 MQT: stat/esp32d/RESULT = {"MqttHost":"mqttbroker"}
13:08:05.333 SRC: MQTT
13:08:05.334 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 0, Data ''
13:08:05.343 MQT: stat/esp32d/RESULT = {"Topic":"esp32d"}
13:08:05.385 SRC: MQTT
13:08:05.386 CMD: Grp 0, Cmnd 'GROUPTOPIC', Idx 1, Len 0, Data ''
13:08:05.395 MQT: stat/esp32d/RESULT = {"GroupTopic1":"tasmotas","GroupTopic2":"","GroupTopic3":"","GroupTopic4":""}

14:09:43.480 SRC: MQTT
14:09:43.482 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 0, Data ''
14:09:43.486 MQT: stat/demo/RESULT = {"Topic":"%hostname%"}
14:09:43.530 SRC: MQTT
14:09:43.532 CMD: Grp 0, Cmnd 'MQTTHOST', Idx 1, Len 0, Data ''
14:09:43.536 MQT: stat/demo/RESULT = {"MqttHost":"mqttbroker"}
14:09:43.580 SRC: MQTT
14:09:43.581 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 0, Data ''
14:09:43.586 MQT: stat/demo/RESULT = {"Topic":"%hostname%"}
14:09:43.631 SRC: MQTT
14:09:43.632 CMD: Grp 0, Cmnd 'GROUPTOPIC', Idx 1, Len 0, Data ''
14:09:43.637 MQT: stat/demo/RESULT = {"GroupTopic1":"tasmotas","GroupTopic2":"","GroupTopic3":"","GroupTopic4":""}
```

arendst, 2021-10-10:
Try to update one with the latest dev release and see how it performs.

revk, 2021-10-10:
Is there a suitable pre-built binary otaurl for that, or do I need to build?

arendst, 2021-10-10:
Look here http://ota.tasmota.com/tasmota/ for pre-builds

arendst, 2021-10-10:
Again, cannot reproduce.

```
14:36:10.910 SRC: MQTT
14:36:10.912 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 0, Data ''
14:36:10.916 MQT: stat/demo/RESULT = {"Topic":"%hostname%"}
14:36:10.961 SRC: MQTT
14:36:10.962 CMD: Grp 0, Cmnd 'MQTTHOST', Idx 1, Len 0, Data ''
14:36:10.967 MQT: stat/demo/RESULT = {"MqttHost":"mqttbroker"}
14:36:11.011 SRC: MQTT
14:36:11.012 CMD: Grp 0, Cmnd 'SETOPTION', Idx 0, Len 0, Data ''
14:36:11.016 MQT: stat/demo/RESULT = {"SetOption0":"ON"}
14:36:11.061 SRC: MQTT
14:36:11.062 CMD: Grp 0, Cmnd 'SETOPTION', Idx 1, Len 0, Data ''
14:36:11.065 MQT: stat/demo/RESULT = {"SetOption1":"ON"}
14:36:11.111 SRC: MQTT
14:36:11.113 CMD: Grp 0, Cmnd 'SETOPTION', Idx 3, Len 0, Data ''
14:36:11.116 MQT: stat/demo/RESULT = {"SetOption3":"ON"}
14:36:11.161 SRC: MQTT
14:36:11.163 CMD: Grp 0, Cmnd 'SETOPTION', Idx 4, Len 0, Data ''
14:36:11.166 MQT: stat/demo/RESULT = {"SetOption4":"OFF"}
```
It's time you report your result from the command ``status 0``.

revk, 2021-10-10:
Yes, sorry, here...

```
cmnd/Office1/status0 
stat/Office1/STATUS0 {"Status":{"Module":46,"DeviceName":"Office Lights Door","FriendlyName":["Door (Office)"],"Topic":"Office1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:04","StartupUTC":"2021-10-10T12:30:10","Sleep":50,"CfgHolder":4617,"BootCount":159,"BCResetTime":"2021-03-20T08:08:13","SaveCount":3501,"SaveAddress":"F9000"},"StatusFWR":{"Version":"9.5.0.9(tasmota)","BuildDateTime":"2021-10-10T09:23:45","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"353/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["C00AA00B","0A05C8000100060000005A0A000000000000","00000280","00006100","00004000"]},"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"Office1","IPAddress":"10.22.107.3","Gateway":"10.22.107.1","Subnetmask":"255.255.255.0","DNSServer1":"10.22.107.1","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:91:90:49","Webserver":1,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"mqtt.iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_919049","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-10-10T12:39:14","Local":"2021-10-10T13:39:14","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:27","Sunset":"18:28"},"StatusSNS":{"Time":"2021-10-10T13:39:14","Switch1":"ON"},"StatusSTS":{"Time":"2021-10-10T13:39:14","Uptime":"0T00:09:04","UptimeSec":544,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","BSSId":"9C:8C:D8:16:7F:01","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```

arendst, 2021-10-10:
Try this https://ota.tasmota.com/tasmota/release-9.3.1/ 

9.3.0 was replaced by 9.3.1 due too major issues.

arendst, 2021-10-10:
I see no issues in your ``status 0`` response.

revk, 2021-10-10:
OK, 9.3.1, thank you.

As you can see, responses now 47ms instead, way better.

```
2021-10-10 13:55:31.122308:cmnd/Office1/SetOption1 
2021-10-10 13:55:31.169062:stat/Office1/RESULT {"SetOption1":"ON"}
2021-10-10 13:55:31.172836:cmnd/Office1/SetOption3 
2021-10-10 13:55:31.219419:stat/Office1/RESULT {"SetOption3":"ON"}
2021-10-10 13:55:31.223434:cmnd/Office1/SetOption4 
2021-10-10 13:55:31.270975:stat/Office1/RESULT {"SetOption4":"OFF"}
2021-10-10 13:55:31.274048:cmnd/Office1/SetOption8 
2021-10-10 13:55:31.320710:stat/Office1/RESULT {"SetOption8":"OFF"}
```

status0 I assume is not there, I get

```
stat/Office1/STATUS {"Status":{"Module":46,"DeviceName":"Office Lights Door","FriendlyName":["Door (Office)"],"Topic":"Office1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
```

revk, 2021-10-10:
```
cmnd/Office1/status 0
stat/Office1/STATUS {"Status":{"Module":46,"DeviceName":"Office Lights Door","FriendlyName":["Door (Office)"],"Topic":"Office1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
stat/Office1/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:33","StartupUTC":"2021-10-10T12:59:39","Sleep":50,"CfgHolder":4617,"BootCount":161,"BCResetTime":"2021-03-20T08:08:13","SaveCount":3513,"SaveAddress":"FA000"}}
stat/Office1/STATUS2 {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"346/699"}}
stat/Office1/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["C00AA00B","0A05C8000100060000005A0A000000000000","00000280","00006100","00004000"]}}
stat/Office1/STATUS4 {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
stat/Office1/STATUS5 {"StatusNET":{"Hostname":"Office1","IPAddress":"10.22.107.3","Gateway":"10.22.107.1","Subnetmask":"255.255.255.0","DNSServer":"10.22.107.1","Mac":"48:3F:DA:91:90:49","Webserver":1,"WifiConfig":4,"WifiPower":17.0}}
stat/Office1/STATUS6 {"StatusMQT":{"MqttHost":"mqtt.iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_919049","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
stat/Office1/STATUS7 {"StatusTIM":{"UTC":"2021-10-10T13:17:12","Local":"2021-10-10T14:17:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:27","Sunset":"18:28"}}
stat/Office1/STATUS10 {"StatusSNS":{"Time":"2021-10-10T14:17:12","Switch1":"ON"}}
stat/Office1/STATUS11 {"StatusSTS":{"Time":"2021-10-10T14:17:12","Uptime":"0T00:17:33","UptimeSec":1053,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","BSSId":"9C:8C:D8:16:7F:01","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

revk, 2021-10-10:
```
cmnd/Office1/status 0
stat/Office1/STATUS {"Status":{"Module":46,"DeviceName":"Office Lights Door","FriendlyName":["Door (Office)"],"Topic":"Office1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
stat/Office1/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release-9.3.1/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:07","StartupUTC":"2021-10-10T13:18:57","Sleep":50,"CfgHolder":4617,"BootCount":161,"BCResetTime":"2021-03-20T08:08:13","SaveCount":3516,"SaveAddress":"FB000"}}
stat/Office1/STATUS2 {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"352/699"}}
stat/Office1/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["C00AA00B","0A05C8000100060000005A0A000000000000","00000280","00006100","00004000"]}}
stat/Office1/STATUS4 {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
stat/Office1/STATUS5 {"StatusNET":{"Hostname":"Office1","IPAddress":"10.22.107.3","Gateway":"10.22.107.1","Subnetmask":"255.255.255.0","DNSServer":"10.22.107.1","Mac":"48:3F:DA:91:90:49","Webserver":1,"WifiConfig":4,"WifiPower":17.0}}
stat/Office1/STATUS6 {"StatusMQT":{"MqttHost":"mqtt.iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_919049","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
stat/Office1/STATUS7 {"StatusTIM":{"UTC":"2021-10-10T13:19:04","Local":"2021-10-10T14:19:04","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:27","Sunset":"18:28"}}
stat/Office1/STATUS10 {"StatusSNS":{"Time":"2021-10-10T14:19:04","Switch1":"ON"}}
stat/Office1/STATUS11 {"StatusSTS":{"Time":"2021-10-10T14:19:04","Uptime":"0T00:00:07","UptimeSec":7,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","BSSId":"9C:8C:D8:16:7F:01","Channel":6,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```

revk, 2021-10-10:
Thanks for looking.

sfromis, 2021-10-10:
Doing a small test, with timestamping MQTT subscribe results, and using a 3rd client to publish commands (in this case a simple `Time` command) to a plug with Tasmota 9.5.0.9, I'm seeing much shorter lags of 40-50 ms from command to response. The timestamps in the Tasmota console from `SRC: MQTT` to the result message being published was a bit shorter than 10 ms. No signs of slowness at all.

revk, 2021-10-10:
My time stamps are from a separate mqtt client on the same broker, but as I say it works much better with 9.3.1 so I do not suspect the broker in any way.

sfromis, 2021-10-10:
It seems pretty clear that Tasmota is able to run without inducing the delays you see. Don't know what may be "different" in your setup..... 

Did you try to match with the time stamps seen in the Tasmota console? Activate `Weblog 3` to get it all.

revk, 2021-10-10:
For now, I'll change my code to use backlog.

Jason2866, 2021-10-10:
Have you tried to setup one device from scratch? -> Erase whole flash and do a clean serial flash.
Do not import any old settings. Configure everything "new"

revk, 2021-10-10:
If anyone is interested in the tool, allowing backup of all tasmota config to sql table, and loading config in to tasmota from an sql table. [tasmota-sql on GitHub](https://github.com/revk/tasmota-sql)

ascillato2, 2021-10-10:
> For now, I'll change my code to use backlog0.[P.S. this is working well and way better anyway, so the delay is more of a non issue for me now]

Great yo know that you have found a workaround to your issue.

closed 2021-10-10 by ascillato2