arendst/Tasmota #13344: Zigbee SONOFF ZBMINI: getting tele/SENSOR msg twice (duplicate messages)

ortegafernando, 2021-10-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have joined a ZBMINI to my zigbee system (PIR, water, temp sensors....) and I am getting an strange behaviour. I am getting twice the tele/SENSOR messages from zbbridge (a SONOFF bridge with tasmota firmware 9.5.0)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): SONOFF ZBBBRIDGE and ZBMINI
- [x] Tasmota binary firmware version number used: 9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotozer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:26:24.973 CMD: Backlog Template; Module; GPIO 255
08:26:25.641 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
08:26:26.494 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
08:26:27.273 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:27:22.020 CMD: Backlog Rule1; Rule2; Rule3
08:27:22.695 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
08:27:23.312 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
08:27:23.912 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:19:30.076 MQT: ch/domo/ZBBRIDGE/stat/STATUS = {"Status":{"Module":75,"DeviceName":"ZBBRIDGE","FriendlyName":["ZBBRIDGE"],"Topic":"ZBBRIDGE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:19:30.696 MQT: ch/domo/ZBBRIDGE/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz","RestartReason":"Software/System restart","Uptime":"3T10:59:43","StartupUTC":"2021-10-08T20:19:47","Sleep":50,"CfgHolder":4617,"BootCount":147,"BCResetTime":"2021-03-14T19:43:40","SaveCount":182,"SaveAddress":"1FB000"}}
08:19:31.311 MQT: ch/domo/ZBBRIDGE/stat/STATUS2 = {"StatusFWR":{"Version":"9.5.0(zbbridge)","BuildDateTime":"2021-06-17T08:27:09","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"355/699"}}
08:19:31.915 MQT: ch/domo/ZBBRIDGE/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00206000","00000000"]}}
08:19:32.537 MQT: ch/domo/ZBBRIDGE/stat/STATUS4 = {"StatusMEM":{"ProgramSize":589,"Free":1200,"Heap":21,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
08:19:33.155 MQT: ch/domo/ZBBRIDGE/stat/STATUS5 = {"StatusNET":{"Hostname":"ZBBRIDGE-5314","IPAddress":"192.168.43.234","Gateway":"192.168.43.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.43.1","Mac":"84:CC:A8:AA:54:C2","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
08:19:33.770 MQT: ch/domo/ZBBRIDGE/stat/STATUS6 = {"StatusMQT":{"MqttHost":"XXXXXXXXX","MqttPort":XXXXX,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA54C2","MqttUser":"XXXX","MqttCount":5,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:19:34.395 MQT: ch/domo/ZBBRIDGE/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-10-12T07:19:33","Local":"2021-10-12T08:19:33","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:06","Sunset":"18:06"}}
08:19:35.031 MQT: ch/domo/ZBBRIDGE/stat/STATUS10 = {"StatusSNS":{"Time":"2021-10-12T08:19:34"}}
08:19:35.635 MQT: ch/domo/ZBBRIDGE/stat/STATUS11 = {"StatusSTS":{"Time":"2021-10-12T08:19:35","Uptime":"3T10:59:48","UptimeSec":298788,"Vcc":3.499,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"Wifi":{"AP":1,"SSId":"XXXXXXXXX","BSSId":"4C:4E:03:00:22:AC","Channel":6,"Mode":"11n","RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
08:21:39.405 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"WebLog":4}
08:21:40.811 ZIG: {"ZbEZSPReceived2":"70E79001450000040100050101000100007FFFE1AD8AFFFF0919F90001000001000002"}
08:21:40.813 ZIG: ZbEZSPSentRaw 80
08:21:40.814 ZIG: {"ZbEZSPReceived":"450000040100050101000100007FFFE1AD8AFFFF0919F90001000001000002"}
08:21:40.816 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0500","srcaddr":"0x8AAD","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":147,"securityuse":0,"seqnumber":127,"fc":"0x19","frametype":1,"direction":1,"disableresp":1,"manuf":"0x0000","transact":249,"cmdid":"0x00","payload":"010000010000"}}
08:21:40.818 ZIG: ZbZCLRawReceived: {"0x8AAD":{"0500<00":"010000010000","ZoneStatusChange":1,"ZoneStatusChangeZone":1,"0406/0000":1,"Endpoint":1,"LinkQuality":147}}
08:21:41.432 MQT: ch/domo/ZBBRIDGE/tele/SENSOR = {"ZbReceived":{"0x8AAD":{"Device":"0x8AAD","Name":"PIR","0500<00":"010000010000","ZoneStatusChange":1,"ZoneStatusChangeZone":1,"Occupancy":1,"Endpoint":1,"LinkQuality":147}}}
08:21:47.737 WIF: Checking connection...
08:22:00.249 ZIG: {"ZbEZSPReceived2":"00E79001590084F72D69F9FEFFF9E3B478BA00"}
08:22:00.250 ZIG: ZbEZSPSentRaw 81
08:22:00.251 ZIG: {"ZbEZSPReceived":"590084F72D69F9FEFFF9E3B478BA00"}
08:22:00.258 ZIG: {"ZbEZSPReceived2":"10E79001590084F72D69F9FEFFF9E3B478BA00"}
08:22:00.259 ZIG: ZbEZSPSentRaw 82
08:22:00.260 ZIG: {"ZbEZSPReceived":"590084F72D69F9FEFFF9E3B478BA00"}
08:22:00.290 ZIG: {"ZbEZSPReceived2":"20E79001450000040106000101400100001678BA84F7FFFF0708080A0000280002"}
08:22:00.291 ZIG: ZbEZSPSentRaw 83
08:22:00.292 ZIG: {"ZbEZSPReceived":"450000040106000101400100001678BA84F7FFFF0708080A0000280002"}
08:22:00.294 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0xF784","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":45,"securityuse":0,"seqnumber":22,"fc":"0x08","frametype":0,"direction":1,"disableresp":0,"manuf":"0x0000","transact":8,"cmdid":"0x0A","payload":"00002800"}}
[... 49 lines cut ...]
_Steps to reproduce the behavior:_
use zbsend  command to ON/OFF switch, or use the button in the ZBMINI. In this case, I have used the BUTTON in ZBMINI

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Recieve only I message in TELE/SENSOR

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
With all other sensors in my zigbee bridge I only get 1 message.

```
08:08:45.397 CMD: zbstatus
08:08:46.008 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"ZbStatus1":[{"Device":"0x3218","Name":"Temp"},{"Device":"0xBE97","Name":"Water"},{"Device":"0x8AAD","Name":"PIR"},{"Device":"0xDEA3","Name":"Temp2"},{"Device":"0xF784","Name":"Switch1"}] [... cut ...]
08:09:25.973 MQT: ch/domo/ZBBRIDGE/stat/RESULT = {"ZbStatus3":[{"Device":"0xF784","Name":"Switch1","IEEEAddr":"0xB4E3F9FFFEF9692D","ModelId":"01MINIZB","Manufacturer":"SONOFF","Endpoints":[1],"Config":["O01"],"Power":0,"Reachable":true,"LastSeen":54,"LastSeenEpoch":1634022511,"LinkQuality":58}]}

08:04:33.656 MQT: ch/domo/ZBBRIDGE/tele/SENSOR = {"ZbReceived":{"0xF784":{"Device":"0xF784","Name":"Switch1","Power":0,"Endpoint":1,"LinkQuality":68}}}
08:04:35.095 MQT: ch/domo/ZBBRIDGE/tele/SENSOR = {"ZbReceived":{"0xF784":{"Device":"0xF784","Name":"Switch1","Power":0,"Endpoint":1,"LinkQuality":71}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-11-01:
Hi,

Yes, you are right but that is not a Bug. It is the Sonoff ZbMini that sends data twice. I can see the same with mine.

closed 2021-11-01 by ascillato2

ortegafernando, 2021-11-01:
Thanks, will it worth to solve in future (without any priority) by "filtering" in bridge these messages?

ascillato, 2021-11-02:
I don't think so because it is being sent by the device. It can contain different information from the previous message.