arendst/Tasmota #13527: Smart Meter Interface SML fails in general

Pewibe, 2021-10-30:
### PROBLEM DESCRIPTION
Smart Meter Interface is not working correct in actual version.
I read my EasyMeter Q3A once a second with self compiled version 9.5.0 (see 1st screenshot). Therefore file "user_config_override.h" has the additional lines:
```
#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
#ifndef USE_SML_SCRIPT_CMD
#define USE_SML_SCRIPT_CMD
#endif
```
As new version 10.0.0 was published I copied the development source, edited the file "user_config_override.h" and updated the device by web interface. The display of values on webpage is missing, same by MQTT value transfer.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Basic R3
- [ x] Tasmota binary firmware version number used: 10.0.0
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: _____
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:26:16.479 CMD: Backlog Template; Module; GPIO 255
16:26:16.481 SRC: WebConsole from 192.168.0.39
16:26:16.484 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
16:26:16.546 SRC: Backlog
16:26:16.549 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
16:26:16.555 MQT: stat/H.StromZaehler/RESULT = {"NAME":"Sonoff BasicR3","GPIO":[32,1,1,1,1,1,1,1,224,320,1,1,1,1],"FLAG":0,"BASE":18}
16:26:16.951 SRC: Backlog
16:26:16.954 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
16:26:16.958 MQT: stat/H.StromZaehler/RESULT = {"Module":{"0":"Sonoff BasicR3"}}
16:26:17.200 SRC: Backlog
16:26:17.203 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
16:26:17.212 MQT: stat/H.StromZaehler/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:36:27.277 CMD: status 0
16:36:27.280 SRC: WebConsole from 192.168.0.xxx
16:36:27.282 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
16:36:27.289 MQT: stat/H.StromZaehler/STATUS = {"Status":{"Module":0,"DeviceName":"H.StromZaehler.106","FriendlyName":[".106"],"Topic":"H.StromZaehler","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"0004","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:36:27.607 MQT: stat/H.StromZaehler/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T14:33:45","StartupUTC":"2021-10-30T00:02:42","Sleep":50,"CfgHolder":4617,"BootCount":35,"BCResetTime":"2021-06-24T22:08:26","SaveCount":57,"SaveAddress":"FB000"}}
16:36:28.000 MQT: stat/H.StromZaehler/STATUS2 = {"StatusFWR":{"Version":"10.0.0.96(tasmota)","BuildDateTime":"2021.10.30 01:56:29","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"397/699"}}
16:36:28.249 MQT: stat/H.StromZaehler/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":4,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["Helios",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","000000C0","00006000","00004000"]}}
16:36:28.565 MQT: stat/H.StromZaehler/STATUS4 = {"StatusMEM":{"ProgramSize":641,"Free":360,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","87DAC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"!1,!2,!3,!4,!5,!6,!53"}}
16:36:29.002 MQT: stat/H.StromZaehler/STATUS5 = {"StatusNET":{"Hostname":"H-StromZaehler-xxx","IPAddress":"192.168.0.xxx","Gateway":"192.168.0.xxx","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.xxx","DNSServer2":"0.0.0.0","Mac":"...","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
16:36:29.325 MQT: stat/H.StromZaehler/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.xxx","MqttPort":1883,"MqttClientMask":"H.StromZaehler.xxx","MqttClient":"H.StromZaehler.xxx","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:36:29.613 MQT: stat/H.StromZaehler/STATUS7 = {"StatusTIM":{"UTC":"2021-10-30T14:36:29","Local":"2021-10-30T16:36:29","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"08:34","Sunset":"18:32"}}
16:36:29.855 MQT: stat/H.StromZaehler/STATUS10 = {"StatusSNS":{"Time":"2021-10-30T16:36:29"}}
16:36:29.955 MQT: stat/H.StromZaehler/STATUS11 = {"StatusSTS":{"Time":"2021-10-30T16:36:29","Uptime":"0T14:33:47","UptimeSec":52427,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":3,"POWER":"---","Wifi":{"AP":1,"SSId":"...","BSSId":"...","Channel":13,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
[... 58 lines cut ...]
### ADDITIONAL CONTEXT
_Add any other context about the problem here._
 Content of script:
```
>D
>B
=>sensor53 r
tper=30
>M 1
+1,3,s,0,9600,ENO
1,77070100000009ff@#,Server-ID,,MeterID,0
1,77070100010800ff@1000,Verbrauch,kWh,EnergyTotal,3
1,=h<hr/>
1,=d 2 86400 @1,Leistung / Tag,W,PowerAvgDay,0
1,=d 2 3600 @1,Leistung / Std,W,PowerAvgHour,0
1,=d 2 60 @1,Leistung / Min,W,PowerAvgMin,0
1,77070100100700ff@1,Leistung aktuell,W,PowerAct,0
1,=h<hr/>
1,77070100240700ff@1,Leistung-Ph1,W,PowerPh1,0
1,77070100380700ff@1,Leistung-Ph2,W,PowerPh2,0
1,770701004c0700ff@1,Leistung-Ph3,W,PowerPh3,0
#
```

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-10-31:
i just updated my 4 meters to 10.01 with no issues. (SML,OBIS,MODBUS,EBUS)
i also pasted your script into one meter and got at least a web and MQTT output. (i have no Q3A)
so i guess something else is wrong with your configuration.

gemu2015, 2021-10-31:
what is your console output when you save the script ?

Pewibe, 2021-10-31:
Thanks @gemu2015 for this early sunday morning answer and your effort. 
1. My script is OK because it works pretty well in version 9.5.0.
2. Other standard functions are working in version 10.0.0 in several devices at my home.
3. Only the self compiled (visual studio code with platformio extension) version 10.0.0 with the change described under "PROBLEM DESCRIPTION" (enable script) runs faulty.
4. As you requested I tried to get a console snapshot between this "ghost commands" appearing every second / each smart meter output. 
```
12:42:22.781 SRC: Serial
12:42:22.784 CMD: Grp 0, Cmnd 'KB', Idx 1, Len 0, Data ''
12:42:22.794 MQT: stat/H.StromZaehler/RESULT = {"Command":"Unknown"}
12:42:23.380 script compressed to 563 bytes = 63 %
12:42:23.382 HTP: Konsolen
12:42:23.780 CMD: kb
12:42:23.782 SRC: Serial
12:42:23.785 CMD: Grp 0, Cmnd 'KB', Idx 1, Len 0, Data ''
12:42:23.794 MQT: stat/H.StromZaehler/RESULT = {"Command":"Unknown"}
12:42:25.349 HTP: Skript konfigurieren
12:42:29.774 script compressed to 563 bytes = 63 %
12:42:29.776 Script: nv=0, tv=0, vns=0, vmem=16, smem=2560
12:42:29.779 Script: performs "sensor53 r"
12:42:29.805 SRC: Rule
12:42:29.808 CMD: Grp 0, Cmnd 'SENSOR', Idx 53, Len 1, Data 'r'
12:42:29.814 MQT: stat/H.StromZaehler/RESULT = {"Command":"Error"}
12:42:29.878 HTP: Konsolen
12:42:30.021 CMD: kb
12:42:30.023 SRC: Serial
12:42:30.025 CMD: Grp 0, Cmnd 'KB', Idx 1, Len 0, Data ''
12:42:30.032 MQT: stat/H.StromZaehler/RESULT = {"Command":"Unknown"}
12:42:30.780 CMD: kb
12:42:30.782 SRC: Serial
12:42:30.785 CMD: Grp 0, Cmnd 'KB', Idx 1, Len 0, Data ''
12:42:30.791 MQT: stat/H.StromZaehler/RESULT = {"Command":"Unknown"}
12:42:31.546 HTP: Konsole
12:42:34.073 WIF: Prüfe Verbindung...
12:42:36.779 CMD: kb
12:42:36.782 SRC: Serial
12:42:36.784 CMD: Grp 0, Cmnd 'KB', Idx 1, Len 0, Data ''
12:42:36.790 MQT: stat/H.StromZaehler/RESULT = {"Command":"Unknown"}
```
5. By the way, commands "Sensor53 d1" and "... d0" show no dump.
6. Look to the console output: What does "SRC: Serial" mean after "CMD: kb"? I'm using Rx/Tx ports (GPIO1/3) as connection to smart meter.
7. During all your thoughts, keep in mind that I changed the Tasmota source (I tried development and release), but kept unchanged hardware device, cable, WLAN, script, development environment, code change (enable script), language version DE.

gemu2015, 2021-10-31:
this is extremely strange. you should definitely get a web display of the meter definition.
there also has been no change in SML driver that could result in this behavior.
i assume SML is no active at all in your build.

gemu2015, 2021-10-31:
look at the drivers definition:
"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45"
driver 53 is NOT there!

Pewibe, 2021-10-31:
Ok, it's hard for me to follow you and to interpret not documented information like "drivers" and "sensors". For a working SML Driver 53 should be there. So what is the purpose of "Sensors !53" in same status 4 line? 
Or in other words, what is important for a working SML?
In file "", is there something more needed than the 4 definitions (#define USE_SCRIPT, #define USE_SML_M, #undef USE_RULES, #define USE_SML_SCRIPT_CMD)?
What is the mechanism to enable Driver 53?

gemu2015, 2021-10-31:
sorry i am very busy with other tasks and confounded that, my mistake.
you are right it is a sensor driver and it is there.
i can not explain this behavior as i can not reproduce that in any way with 4 different devices.

Pewibe, 2021-10-31:
I understand your lack of time. I like to get further by myself and to close with you (for the next days). Can you send your SML-enabled tasmota binary firmware file version 10.0.0 to me? Here a temporary mail address: sml@trash-mail.com.

Pewibe, 2021-10-31:
Sorry coming up again, but I have found a difference between working version 9.5.0:
MQT: stat/H.StromZaehler/STATUS4 = {"StatusMEM":{"ProgramSize":591, ... ,"Sensors":"1,2,3,4,5,6,**53**"}}
and not working version 10.0.0:
MQT: stat/H.StromZaehler/STATUS4 = {"StatusMEM":{"ProgramSize":641, ... ,"Sensors":"!1,!2,!3,!4,!5,!6,**!53**"}}
I cannot find an explanation for the Sensors-part of "Status 4". What does the "!" mean?

arendst, 2021-10-31:
Every exclamation mark means that sensor driver is disabled. So it makes well sense that the latter won't work as all sensors are disabled.

Use command ``setsensor`` to control individual sensors. Like ``setsensor53 1``

gemu2015, 2021-10-31:
that's it ! sensor 53 and all others are DISABLED!
there is a tasmota command to disable or enable sensors.
dont know why that is active in your build. see tasmota docs.

SetSensor<x> | Enable / Disable individual sensor driver(x= 1..127)

Pewibe, 2021-10-31:
Yes, you got it. After "setsensorX 1" the values appear in the web interface.
So 2 items remain:
1. What is responsible not enabling the sensors?
2. The periodically sending of values by the Smart Meter should not erroneously result in these "ghost commands" kb! See last command screen shot.

Pewibe, 2021-10-31:
To deliver additional info: I switched back and forth between my self compiled versions 9.5.0 and 10.0.0 and sensors keep enabled. So it is not worth to think further about who or what disabled them. I can say only that I never used that command "setsensorX y" to switch off because even I didn't know it.

ascillato2, 2021-11-01:
> that's it ! sensor 53 and all others are DISABLED!

Good to know that now your issue is solved.

closed 2021-11-01 by ascillato2