arendst/Tasmota #13530: When send Buzzer command, Sleep Settings change value without control

mkonsel, 2021-10-30:
### PROBLEM DESCRIPTION
On a Lolin32 and Wroom32 Device, with Tasmota 10.0.0.1, with factory settings, if i send command on "Buzzer1", the sleep settings modify, going to "0" or "10"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Lolin32 and Wroom32
- [ x] Tasmota binary firmware version number used: 10.0.0.1 
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: Web installer
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:07:55.517 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:07:55.714 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:07:55.939 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"512":"Buzzer_i"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:08:31.629 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D1952C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:08:31.667 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:05:27","StartupUTC":"2021-10-30T23:03:04","Sleep":250,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2021-10-30T23:58:23","SaveCount":17}}
00:08:31.699 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0.1(tasmota)","BuildDateTime":"2021-10-30T17:59:18","Core":"1_0_7_5","SDK":"v3.3.5-254-g9a6567289","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"393/699"}}
00:08:31.713 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wifi82",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
00:08:31.754 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1273,"Free":1856,"Heap":134,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000482D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52,59","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,127"}}
00:08:31.799 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-D1952C-5420","IPAddress":"192.168.2.163","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:D1:95:2C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:08:31.824 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D1952C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:08:31.840 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-10-30T23:08:31","Local":"2021-10-31T00:08:31","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:36","Sunset":"17:31"}}
00:08:31.867 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-10-31T00:08:31","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
00:08:31.876 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-10-31T00:08:31","Uptime":"0T00:05:27","UptimeSec":327,"Heap":133,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"Wifi":{"AP":1,"SSId":"wifi82","BSSId":"F0:B0:14:5B:03:00","Channel":1,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":2,"Downtime":"0T00:00:09"}}}

```
- [ x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:09:23.141 CMD: weblog 4
00:09:23.143 SRC: WebConsole from 192.168.2.138
00:09:23.146 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '4'
00:09:23.150 RSL: RESULT = {"WebLog":4}
00:09:24.867 CMD: state
00:09:24.869 SRC: WebConsole from 192.168.2.138
00:09:24.872 CMD: Grp 0, Cmnd 'STATE', Idx 1, Len 0, Data ''
00:09:24.882 RSL: RESULT = {"Time":"2021-10-31T00:09:24","Uptime":"0T00:06:20","UptimeSec":380,"Heap":132,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"wifi82","BSSId":"F0:B0:14:5B:03:00","Channel":1,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":2,"Downtime":"0T00:00:09"}}
00:09:29.382 CMD: buzzer1 0
00:09:29.385 SRC: WebConsole from 192.168.2.138
00:09:29.388 CMD: Grp 0, Cmnd 'BUZZER', Idx 1, Len 1, Data '0'
00:09:29.393 BUZ: Count 0(0), Time 1/1, Tune 0x00000000(0x00000000), Size 0, Mode 0
00:09:29.396 RSL: RESULT = {"Buzzer":"Done"}
00:09:34.202 WIF: Checking connection...
00:09:39.186 CMD: state
00:09:39.190 SRC: WebConsole from 192.168.2.138
00:09:39.192 CMD: Grp 0, Cmnd 'STATE', Idx 1, Len 0, Data ''
00:09:39.200 RSL: RESULT = {"Time":"2021-10-31T00:09:39","Uptime":"0T00:06:35","UptimeSec":395,"Heap":131,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":42,"MqttCount":0,"Wifi":{"AP":1,"SSId":"wifi82","BSSId":"F0:B0:14:5B:03:00","Channel":1,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":2,"Downtime":"0T00:00:09"}}
00:09:54.200 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1) send command "sleep 250"
2) send command "buzzer1 0"
3) check sleep status with command "state"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
the sleep settings is modified when send "buzzer1" command...whY?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-10-31:
The buzzer function requires resetting Sleep to a short interval to work, due to the timing control needed for buzzing. In this case, it seems that your nonsensical use of "Buzzer1 0" while the buzzer is not running causes an unusual situation. Normally, the buzzer will return to the prior state after a "Buzzer1 0", but in with no prior state to return to, it gets out of whack.

You can avoid this by not turning off buzzer when not relevant, or by activating the Sleep command afterwards, and turn on/off the buzzer.

ascillato2, 2021-10-31:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-10-31 by ascillato2

mkonsel, 2021-10-31:
> The buzzer function requires resetting Sleep to a short interval to work, due to the timing control needed for buzzing. In this case, it seems that your nonsensical use of "Buzzer1 0" while the [...]

Hi, the step are with nonsensical use, but in the real life i created a rule for activate buzzer when Counter1 > 2, and disactivate buzzer when Counter1 < 3:

Rule1 on counter#c1 > 2 do Buzzer1 -1,3,4 endon on counter#c1 < 3 do Buzzer1 0 endon (example)

when tasmota boot or restart, execute the rule1, and if counter1 < 3, execute the command "Buzzer1 0" , modifying the Sleep Settings (set to 0 with release 10.0.0.0, set to 10 with release 10.0.0.1)

It's a non sense use of rule?

PS: I need to do blinking led when the counter > 2, it's correct to use a buzzer function? Alternatives for blinking led? thanks

when tasmota boot or restart, if counter

sfromis, 2021-10-31:
Anything (very much including the rules you post now) turning off buzzer when it has not been turned on is nonsencical. And it does appear that such nonsencial use triggers an anomalous behavior of the code to save/restore sleep across the buzzing action. 

The simplest way of having a blinking led is to config it as a relay, and use Power Blink after setting BlinkCount/BlinkTime once.

mkonsel, 2021-10-31:
> Anything (very much including the rules you post now) turning off buzzer when it has not been turned on is nonsencical. And it does appear that such nonsencial use triggers an anomalous behavior of the code to save/restore sleep across the buzzing action.
> 

The tasmota execute the rules after reboot/restart  it's logical behaviour. If the user execute a command for disable a buzzer, and the result is set to zero the sleep time without a scope (power consumption > 1 watt, loadavg>100), it's a BUG. 

> The simplest way of having a blinking led is to config it as a relay, and use Power Blink after setting BlinkCount/BlinkTime once.

If i need to Blink a led for infinite time, how to set blinktime/blinkcount? Blinkcount max is 32.000, exist a value for set infinite time?

barbudor, 2021-11-01:
> it's a BUG.

You are welcome to provide a PR. 

> If i need to Blink a led for infinite time, how to set blinktime/blinkcount? Blinkcount max is 32.000, exist a value for set infinite time?

If not happy with Blink, use RuleTimer