arendst/Tasmota #13535: Aliases for GPIO devices to use in commands/rules such as Power-fan instead of Power1

prabbit237, 2021-11-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have some multi-outlet devices. So I have Relay1, Relay2 and Relay 3. If I plug in three devices, such as a cable modem, a router and a printer, it would be much easier if I can set an alias for each relay.

There's the "BLEAlias mac=alias mac=alias" command that solves the same issue for multiple Bluetooth devices and a GPIOAlias could do the same here. So I could do something like:

````
GPIOAlias 2=Modem 14=Router
````

And then later, without trying to recall if the modem is on outlet 1 or 2, etc, I could use commands like:

````
backlog Power-Modem 0;delay 600;Power-Modem 1
````
The controls can be named in the WebUI but if I'm doing all this from rules/console/remote MQTT, it'd be a pain to load up the WebUI just to remind myself which is whine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): N/A
- [x ] Tasmota binary firmware version number used: N/A
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: N/A
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-11-01:
For enhancement suggestions, you should not open an issue here (which is to report bugs), but instead a discussion under that tab.

ascillato2, 2021-11-01:
Moving to discussions as this is not a bug report, but a feature request

closed 2021-11-01 by ascillato2