arendst/Tasmota #13543: Can't configure extra GPIO pins of Sonoff POW R2

TheDuchCobra1, 2021-11-02:
I've reflashed a Sonoff POW R2 with the tasmota firmware (latest version).
When I select "Configure Module" and turn it into a Sonoff POW R2, I can't select any options at extra GPIO pins.
If you take a look at the board, there are a few GPIO ports free. (GPIO1, GPIO3, GPIO4 and GPIO5)

Why can't the Tasmota Sonoff POW R2 firmware don't use these GPIO pins for extra configuration?

closed 2021-11-02 by close-issue-app[bot]

ascillato, 2021-11-02:
As designed. Those pins are referenced to mains AC. If you connect anything there, that will be connected to mains AC too. That is very dangerous. We don't give support to dangerous modifications.

This has been already addressed several times. Please, do a quick search for more information on old issues. Thanks.

TheDuchCobra1, 2021-11-02:
I understand, that's clear.
I was searching, but couldn't find it directly.
Thank you for your answer!