arendst/Tasmota #13566: Constant  MAX_SHUTTER_RELAYS = 8 interfere with devices like ESP32 or i2c extended devices

stefanbode, 2021-11-04:
### PROBLEM DESCRIPTION
Only in autodiscovery for HASS (xdrv12) the MAX_SHUTTER_RELAYS is used. The constant has no functionality on shutters itself.
On my ESP32 I now try to use autodiscovery with HASS and have defined the shutter on relay 7/8 and 12/13. The second shutter will not be exposed through HAS due to the relay number > 8.

My proposal for fixing: Remove the constant and remove the additional check on MAX_SHUTTER_RELAYS. I do not see a benefit. Additionally the "const" can easily also go up to 32 relays with i2c expansion board on esp8266. If the coding team agree I can make a pr for it. Or lets discuss why the discovery needs the max number.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: _____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-11-04:
> My proposal for fixing: Remove the constant and remove the additional check on MAX_SHUTTER_RELAYS.

I agree. That restriction is not needed.

Jason2866, 2021-11-04:
Since shutter is your work, you know best.

sfromis, 2021-11-04:
You need careful review of involved code for that. At least, going above 8 would break the use of `uint8_t shutter_mask` in [https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_12_home_assistant.ino](xdrv_12_home_assistant.ino)

stefanbode, 2021-11-05:
@sfromis : This is exactly the information I was waiting for from people knowing this driver (xdrv12). I will check how it interferes here and fix this, too. THX!

stefanbode, 2021-11-05:
@sfromis : Can you make a small test if my assumptions to fix it are ok?

sfromis, 2021-11-05:
@stefanbode Not being a user of shutters and/or HA, I have no practical way of doing any testing here.

ascillato2, 2021-11-05:
Wow, that was fast ! 👍 

Closing as the PR #13572 has been merged.

Thanks.

closed 2021-11-05 by ascillato2