arendst/Tasmota #13576: If statement command not executed when multiple commands are present

pestevez, 2021-11-05:
### PROBLEM DESCRIPTION
The statement list of an `IF` statement is not executed when the rule contains multiple commands.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4CH Pro R3
- [x] Tasmota binary firmware version number used: 10.0.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled (with USE_EXPRESSION and SUPPORT_IF_STATEMENT)
- [x] Flashing tools used: TasmoCompiler v7.0.0
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
03:27:34.941 MQT: pool-equipment-switch/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
03:27:35.155 MQT: pool-equipment-switch/stat/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
03:27:35.409 MQT: pool-equipment-switch/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
03:29:52.806 MQT: pool-equipment-switch/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":65,"Free":446,"Rules":"ON Power4#state=1 DO BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON"}}
03:29:53.022 MQT: pool-equipment-switch/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":71,"Free":440,"Rules":"ON Power1#state=1 DO Power3 1 ENDON ON Power2#state=1 DO Power3 1 ENDON"}}
03:29:53.272 MQT: pool-equipment-switch/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":86,"Free":425,"Rules":"ON Power3#state=1 DO IF ((Power1#state==0) AND (Power2#state==0)) Power3 0 ENDIF ENDON"}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
03:30:33.038 MQT: pool-equipment-switch/stat/STATUS = {"Status":{"Module":23,"DeviceName":"Pool Equipment Switch","FriendlyName":["Pool Pump High","Pool Pump Low","Pool Pump None","Pool Equipment Aux"],"Topic":"pool-equipment-switch","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
03:30:33.048 MQT: pool-equipment-switch/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:58","StartupUTC":"2021-10-29T02:19:35","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2021-10-18T03:04:01","SaveCount":730,"SaveAddress":"F8000"}}
03:30:33.057 MQT: pool-equipment-switch/stat/STATUS2 = {"StatusFWR":{"Version":"10.0.0.1(TasmoCompiler-esp8266generic)","BuildDateTime":"2021-10-29T02:15:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"427/699"}}
03:30:33.068 MQT: pool-equipment-switch/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd-wrt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AC009","2805C8000100060000005A0A000000000000","00000280","00006000","00004000"]}}
03:30:33.080 MQT: pool-equipment-switch/stat/STATUS4 = {"StatusMEM":{"ProgramSize":525,"Free":476,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F804582","04260001","00000003","00000000","C000E981","00000004","00001000","00000000"],"Drivers":"1,2,4,6,8,9,10,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2"}}
03:30:33.091 MQT: pool-equipment-switch/stat/STATUS5 = {"StatusNET":{"Hostname":"pool-equipment-switch-3666","IPAddress":"192.168.1.76","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"E8:68:E7:45:4E:52","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
03:30:33.097 MQT: pool-equipment-switch/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_454E52","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:30:33.108 MQT: pool-equipment-switch/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-10-29T02:30:33","Local":"2021-10-29T03:30:33","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:32","Sunset":"17:34"}}
03:30:33.117 MQT: pool-equipment-switch/stat/STATUS10 = {"StatusSNS":{"Time":"2021-10-29T03:30:33"}}
03:30:33.130 MQT: pool-equipment-switch/stat/STATUS11 = {"StatusSTS":{"Time":"2021-10-29T03:30:33","Uptime":"0T00:10:58","UptimeSec":658,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"dd-wrt","BSSId":"48:DD:0C:DC:68:85","Channel":1,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
03:32:22.326 MQT: pool-equipment-switch/tele/STATE = {"Time":"2021-10-29T03:32:22","Uptime":"0T00:12:47","UptimeSec":767,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"dd-wrt","BSSId":"48:DD:0C:DC:68:85","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
03:32:22.330 MQT: pool-equipment-switch/stat/RESULT = {"POWER4":"ON"}
03:32:22.334 MQT: pool-equipment-switch/stat/POWER4 = ON
03:32:22.367 RUL: POWER4#STATE=1 performs "BACKLOG IF (1==1) var1 1 ENDIF; var2 2"
03:32:22.369 SRC: Rule
03:32:22.371 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 30, Data 'IF (1==1) var1 1 ENDIF; var2 2'
03:32:22.374 SRC: Backlog
03:32:22.376 CMD: Grp 0, Cmnd 'IF', Idx 1, Len 19, Data '(1==1) var1 1 ENDIF'
03:32:22.384 MQT: pool-equipment-switch/stat/RESULT = {"If":"Done"}
03:32:22.625 SRC: Backlog
03:32:22.627 CMD: Grp 0, Cmnd 'VAR', Idx 2, Len 1, Data '2'
03:32:22.633 MQT: pool-equipment-switch/stat/RESULT = {"Var2":"2"}
```

### TO REPRODUCE
1. Set a rule to contain multiple commands, including an `IF` statement.
2. Perform an action that will trigger the rule.

Example:
```
rule1 ON Power4#state=1 DO BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON
```

### EXPECTED BEHAVIOUR
The statement list of the `IF` command is executed when its logical expression evaluates to true.

### ACTUAL BEHAVIOUR
[... 9 lines cut ...]
```
rule1 ON Power4#state=1 DO IF (1==1) var1 1 ENDIF ENDON
```
Weblog Output:
```lua
03:44:30.971 MQT: pool-equipment-switch/tele/STATE = {"Time":"2021-10-29T03:44:30","Uptime":"0T00:24:55","UptimeSec":1495,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"dd-wrt","BSSId":"48:DD:0C:DC:68:85","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
03:44:30.979 MQT: pool-equipment-switch/stat/RESULT = {"POWER4":"ON"}
03:44:30.983 MQT: pool-equipment-switch/stat/POWER4 = ON
03:44:31.016 RUL: POWER4#STATE=1 performs "IF (1==1) var1 1 ENDIF"
03:44:31.018 SRC: Rule
03:44:31.021 CMD: Grp 0, Cmnd 'IF', Idx 1, Len 19, Data '(1==1) var1 1 ENDIF'
03:44:31.028 MQT: pool-equipment-switch/stat/RESULT = {"If":"Done"}
03:44:31.240 SRC: Backlog
03:44:31.243 CMD: Grp 0, Cmnd 'VAR', Idx 1, Len 1, Data '1'
03:44:31.248 MQT: pool-equipment-switch/stat/RESULT = {"Var1":"1"}
```
Same issue as reported in the third scenario in #10569. Please note that Backlog is present. It is also a dup of #13506 , which was also closed because it was missing the screenshots section.
I asked for help in the rules Discord channel but the expected behavior was unclear.

The provided command is similar to the following, which is included in the [docs](https://tasmota.github.io/docs/Rules/#grammar):
```
Rule1 ON ENERGY#Current>10 Backlog Power1 0; IF (%var1%==1) Power1 1 ENDIF;Power 2 0;Power3 1 ENDON
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-11-05:
From the documentation:

When the `<if-statement>` is preceded by other Tasmota commands you should use Backlog rather than `Do` , e.g.
```Rule1 ON ENERGY#Current>10 Backlog Power1 0; IF (%var1%==1) Power1 1 ENDIF;Power 2 0;Power3 1 ENDON and not Rule1 ON ENERGY#Current>10 DO Power1 0; IF (%var1%==1) Power1 1 ENDIF ENDON```

pestevez, 2021-11-05:
@Jason2866 - what is your suggested way to re-write the example rule that I provide so that it sets Var1 and Var2?
```
rule1 ON Power4#state=1 BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON
```
Is it the one above? If so, it does not trigger the execution of the rule. Weblog below:
```
15:23:29.657 CMD: rule1 ON Power4#state=1 BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON
15:23:29.659 SRC: WebConsole from 192.168.1.124
15:23:29.662 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 62, Data 'ON Power4#state=1 BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON'
15:23:29.666 RUL: Stored uncompressed, would compress from 62 to 61 (-2%)
15:23:29.672 MQT: pool-equipment-switch/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":62,"Free":449,"Rules":"ON Power4#state=1 BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON"}}
15:23:34.854 WIF: Checking connection...
15:23:39.764 APP: Button4 level 1-0
15:23:39.767 SRC: Button
15:23:39.775 MQT: pool-equipment-switch/tele/STATE = {"Time":"2021-11-05T15:23:39","Uptime":"7T12:04:04","UptimeSec":648244,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"dd-wrt","BSSId":"48:DD:0C:DC:68:85","Channel":1,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:23:39.782 MQT: pool-equipment-switch/stat/RESULT = {"POWER4":"ON"}
15:23:39.785 MQT: pool-equipment-switch/stat/POWER4 = ON
15:23:39.864 APP: Button4 level 0-1
15:23:40.697 CFG: Saved to flash at F9, Count 777, Bytes 4096
```
Also, if the issue is just the precedence, should we then expect the any of the following to set Var1 to1?
```
rule1 ON Power4#state=1 DO BACKLOG var2 2; IF (1==1) var1 1 ENDIF ENDON
```
```
rule1 ON Power4#state=1 BACKLOG var2 2; IF (1==1) var1 1 ENDIF ENDON
```
None of them set Var1 to 1. There are additional examples in #10569. I don't think this is an issue of not following the documentation, but a bug. CC @barbudor

pestevez, 2021-11-05:
Also, the example that I provided is not **preceded** by any other commands, it is only **followed** by other commands:
```
rule1 ON Power4#state=1 DO BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON
```

Jason2866, 2021-11-05:
Just NO `DO` if you use `Backlog`

pestevez, 2021-11-05:
Yeah, but that doesn't work either because then the whole rule is not executed.

barbudor, 2021-11-05:
```
Rule1
  on event#test do backlog var2 %value%; var3 %value% ; if (1==1) var1 %value% endif ; var4 %value% endon

Rule2
  on event#test2 do if (1==1) var1 %value% endif endon

17:29:19.348 CMD: event test=5
17:29:19.479 MQT: sonnaz/unconf_18FE34A6D683/stat/EVENT = {"Event":"Done"}
17:29:19.565 RUL: EVENT#TEST performs "backlog var2 5; var3 5 ; if (1==1) var1 5 endif ; var4 5"
17:29:19.669 MQT: sonnaz/unconf_18FE34A6D683/stat/VAR = {"Var2":"5"}
17:29:19.913 MQT: sonnaz/unconf_18FE34A6D683/stat/VAR = {"Var3":"5"}
17:29:20.157 MQT: sonnaz/unconf_18FE34A6D683/stat/IF = {"If":"Done"}
17:29:20.400 MQT: sonnaz/unconf_18FE34A6D683/stat/VAR = {"Var4":"5"}

17:29:29.185 CMD: event test2=8
17:29:29.272 MQT: sonnaz/unconf_18FE34A6D683/stat/EVENT = {"Event":"Done"}
17:29:29.369 RUL: EVENT#TEST2 performs "if (1==1) var1 8 endif"
17:29:29.457 MQT: sonnaz/unconf_18FE34A6D683/stat/IF = {"If":"Done"}
17:29:29.700 MQT: sonnaz/unconf_18FE34A6D683/stat/VAR = {"Var1":"8"}
```

So Rule2 behave as expected
Rule1 doesn't process the IF correctly, but others commands work as expected
Rule1 with backlog but without do doesn't work at all => doc is wrong here.
The code in `RuleSetProcess()` definitively looks for a `DO`. Without it, rules is invalid.

barbudor, 2021-11-05:
I have found why the commands within IF/ENDIF block do not execute.
Now I am not ready to find a solution quickly by myself.
I need a call to a friend, or at least Master @arendst 

Theo, the problem is with a rule such as below where the block inside `IF/ENDIF` is not executed even when the condition is true
```
Rule 1
  ON event#test DO Backlog var1 %value% ; IF (1==1) var2 %value% ENDIF ENDON
```
Log:
```
19:22:08.368 CMD: event test=1
19:22:08.454 MQT: sonnaz/unconf_18FE34A6D683/stat/EVENT = {"Event":"Done"}
19:22:08.545 RUL: EVENT#TEST performs "backlog var1 1; if ( 1 == 1 ) var2 1 endif"
19:22:08.645 MQT: sonnaz/unconf_18FE34A6D683/stat/VAR = {"Var1":"1"}
19:22:08.808 Evaluate ' 1 == 1 '
19:22:08.810 Evaluate 1
19:22:08.812 => execute ' var2 1 '
19:22:08.814 ExecuteCommandBlock: backlog_mutex=1
19:22:08.899 MQT: sonnaz/unconf_18FE34A6D683/stat/IF = {"If":"Done"}
```
As pointed by my log, the problem lies here:
https://github.com/arendst/Tasmota/blob/4e4206af307df35e20dff30aa056eab31a4a2b87/tasmota/xdrv_10_rules.ino#L1944-L1953
and the commands within the `IF/ENDIF` are not executed because of the backlog mutex.

Doc says that 
> When the <if-statement> is preceded by other Tasmota commands you should use Backlog rather than Do , e.g.
`Rule1 ON ENERGY#Current>10 Backlog Power1 0; IF (%var1%==1) Power1 1 ENDIF;Power 2 0;Power3 1 ENDON`
> and not 
`Rule1 ON ENERGY#Current>10 DO Power1 0; IF (%var1%==1) Power1 1 ENDIF ENDON`

However this is wrong as `RuleSetProcess()` is looking for the `DO` and will fail without the `DO`
https://github.com/arendst/Tasmota/blob/4e4206af307df35e20dff30aa056eab31a4a2b87/tasmota/xdrv_10_rules.ino#L718-L719

And without `Backlog` : `ON event#test DO var1 %value% ; IF (1==1) var2 %value% ENDIF ENDON`
var1 gets assigned the whole string `%value%  ; IF (1==1) var2 %value% ENDIF`

Your advice would be appreciated.
Thanks.

arendst, 2021-11-06:
Read your comment.

For now I have no solution either. The mutex is needed to solve issues resulting in exceptions. Removing it is not an option.

I've never looked at the IF/ENDIF code as it was introduced by someone else and I never needed it.

If anyone wants to dive in pls do.

arendst, 2021-11-06:
Testing now.

arendst, 2021-11-06:
Using the latest commit at least the test rule works.

Pls try in your setup and see if backlog commands are excuted in expected order (without exceptions).

Gifford47, 2021-11-07:
i also tested it with the rule:
`on event#alarm backlog var16 %value%; IF (%value%==0) wifi 0;ruletimer2 1200 ENDIF
endon`

and it doesnt work either. i only get "{"Event":"Done"}" without any rule execution.
with following rule, the IF-block is executed, but not the var16 %value%
`on event#alarm do IF (%value%==0) wifi 0;ruletimer2 1200 ENDIF; var16 %value%
endon`

pestevez, 2021-11-07:
It is working fine for me when I build from the development branch and use the sample rule in the issue description. Please the details below:
Rule:
```
rule1 ON Power4#state=1 DO BACKLOG IF (1==1) var1 1 ENDIF; var2 2 ENDON
```
WebLog:
```
12:29:39.995 MQT: pool-equipment-switch/stat/RESULT = {"POWER4":"ON"}
12:29:39.999 MQT: pool-equipment-switch/stat/POWER4 = ON
12:29:40.011 RUL: POWER4#STATE=1 performs "BACKLOG IF (1==1) var1 1 ENDIF; var2 2"
12:29:40.013 SRC: Rule
12:29:40.015 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 30, Data 'IF (1==1) var1 1 ENDIF; var2 2'
12:29:40.021 SRC: Backlog
12:29:40.023 CMD: Grp 0, Cmnd 'IF', Idx 1, Len 19, Data '(1==1) var1 1 ENDIF'
12:29:40.029 MQT: pool-equipment-switch/stat/RESULT = {"If":"Done"}
12:29:40.071 APP: Button4 level 0-1
12:29:40.272 SRC: Backlog
12:29:40.274 CMD: Grp 0, Cmnd 'VAR', Idx 1, Len 1, Data '1'
12:29:40.280 MQT: pool-equipment-switch/stat/RESULT = {"Var1":"1"}
12:29:40.522 SRC: Backlog
12:29:40.524 CMD: Grp 0, Cmnd 'VAR', Idx 2, Len 1, Data '2'
12:29:40.530 MQT: pool-equipment-switch/stat/RESULT = {"Var2":"2"}
```
Thanks for the quick fix, @arendst !

pestevez, 2021-11-07:
@Gifford47 - in your first example I believe the issue is that you are missing `DO`. I am not sure what could be the problem with the second rule.

Gifford47, 2021-11-07:
ok, i found my failure;
it has to be:
`on event#alarm do backlog var16 %value%; IF (%value%==0) wifi 0;ruletimer2 1200 ENDIF endon`

so, the doc "https://tasmota.github.io/docs/Rules/#syntax" has to be corrected

barbudor, 2021-11-07:
One can update the doc by clicking on the pencil button on the top right

ascillato2, 2021-11-07:
Thanks for reporting that it is fixed.

closed 2021-11-07 by ascillato2