arendst/Tasmota #13634: Tasmota led controller blinking (turning off and on) during restart

Heisenberg2980, 2021-11-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have a led controller ESP8285 with the latest tasmota firmware (10.0.0) that turn off and on every time it restarts for any reason (the issue was already happening with the version 9.2.0). This just happens when it is on before restarting (if it is off then it restarts without turning on).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): led controller ESP8285
- [x] Tasmota binary firmware version number used: 10.0.0(tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Upgrade by web server
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18
"Module":{"18":"Generic"}
"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"416":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
{"Status":{"Module":18,"DeviceName":"Tasmota_ESP8285_1","FriendlyName":["Tasmota_ESP8285_1"],"Topic":"tasmota_33FBE8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:14:35","StartupUTC":"2021-11-10T18:16:24","Sleep":50,"CfgHolder":4617,"BootCount":382,"BCResetTime":"2020-12-29T22:41:32","SaveCount":3746,"SaveAddress":"F9000"}}
{"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:23:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"442/699"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["2ND-FLOOR-2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["800A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00004000"]}}
{"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
{"StatusNET":{"Hostname":"tasmota-33FBE8-7144","IPAddress":"192.168.0.173","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"50:02:91:33:FB:E8","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
{"StatusMQT":{"MqttHost":"192.168.0.31","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_33FBE8","MqttUser":"heisenberg","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
{"StatusTIM":{"UTC":"2021-11-10T18:30:59","Local":"2021-11-10T18:30:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+00:00","Sunrise":"06:51","Sunset":"16:15"}}
{"StatusSNS":{"Time":"2021-11-10T18:30:59"}}
{"StatusSTS":{"Time":"2021-11-10T18:30:59","Uptime":"0T00:14:35","UptimeSec":875,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":55,"Color":"139,1","White":55,"CT":154,"Channel":[55,1],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"2ND-FLOOR-2G","BSSId":"20:0D:B0:58:41:FA","Channel":6,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just restart the device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should restart without blinking

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-11-10:
On ESP8266/ESP8285, PWM is handled by software (there is no hardware PWM).
So when the software restart, it stops managing PWM and the level is set to LOW right before the restart for security.
Tasmota restarts PWM generation after reboot.
No way to do it differently.

It may be possible that this problem doesn't exist on ESP32 as it has an hardware PWM that should keep on running during reboot.

Heisenberg2980, 2021-11-10:
Thank you for the quick reply and the good explanation, closing the issue as it seems this is the expected behaviour with these chips.