arendst/Tasmota #13640: webui ota fw upgrade to version 10 fails with memory issue

rsteininger, 2021-11-11:
### PROBLEM DESCRIPTION
hi there,
Yesterday I tried upgrading / updating a number of my devices from tasmota 9.5 to the latest 10.0.0.2.

Here a list of device types I tried:
Gosund UP111
Sonoff 433Mhz RF Bridge
Shelly 1PM

Due to memory issues, I always have to go the tasmota-minimal step first, followed by the target tasmota firmware.
However, yesterday for the first time, the final step failed on every single device type reporting: insufficient memory.
I downgraded back to 9.5 via minimal, followed by regular tasmota.bin v9.5 without any problems. Repeated efforts to go to 10.0.0.2 via webui manual or on-button upgrade failed every single time.

What is however interesting is, that when going to the console, and initiating the upgrade from there, I was able to upgrade ALL my devices, no problem with the following command!
backlog OtaUrl http://ota.tasmota.com/tasmota/tasmota.bin; upgrade 1

Comparing file sizes from v9.5.0 Michael:
tasmota.bin 622,112 bytes
tasmota-minimal.bin 398,528 Byte

with v10.0.0 Norman:
tasmota.bin 627,248
tasmota-minimal.bin 403,056

it seems to me that in my case, these have somehow become just a tat to big for a webui OTA upgrade with minimal fw intermediate step. Not sure if anything can be down about this or if there is another workaround?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Gosund UP111, Sonoff 433Mhz RF Bridge, Shelly 1PM, Shelly Mini
- [ ] Tasmota binary firmware version number used: 9.5.0 upgraded to 10.0.0.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: web ui OTA, console OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: Not relevant

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: none

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:12:06.902 MQT: socket04/stat/STATUS = {"Status":{"Module":0,"DeviceName":"socket04","FriendlyName":["socket04"],"Topic":"socket04","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:12:06.910 MQT: socket04/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T17:20:41","StartupUTC":"2021-11-10T15:51:25","Sleep":50,"CfgHolder":4617,"BootCount":1813,"BCResetTime":"2020-05-26T08:38:09","SaveCount":4623,"SaveAddress":"F6000"}}
10:12:06.919 MQT: socket04/stat/STATUS2 = {"StatusFWR":{"Version":"10.0.0.2(tasmota)","BuildDateTime":"2021-11-10T14:36:16","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"389/699"}}
10:12:06.926 MQT: socket04/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["STONEY_2G","STONEY_TP"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
10:12:06.941 MQT: socket04/stat/STATUS4 = {"StatusMEM":{"ProgramSize":614,"Free":388,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:12:06.947 MQT: socket04/stat/STATUS5 = {"StatusNET":{"Hostname":"socket04","IPAddress":"192.168.10.177","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.10.2","DNSServer2":"192.168.10.3","Mac":"C4:4F:33:88:50:AB","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
10:12:06.953 MQT: socket04/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.198","MqttPort":1883,"MqttClientMask":"socket04","MqttClient":"socket04","MqttUser":"mqtt_ha","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:12:06.962 MQT: socket04/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-11-11T09:12:06","Local":"2021-11-11T10:12:06","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:53","Sunset":"17:14"}}
10:12:06.966 MQT: socket04/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:12:06.977 MQT: socket04/stat/STATUS10 = {"StatusSNS":{"Time":"2021-11-11T10:12:06","ENERGY":{"TotalStartTime":"2020-02-24T15:18:49","Total":918.516,"Yesterday":0.435,"Today":2.334,"Power":228,"ApparentPower":286,"ReactivePower":173,"Factor":0.80,"Voltage":235,"Current":1.220}}}
10:12:06.987 MQT: socket04/stat/STATUS11 = {"StatusSTS":{"Time":"2021-11-11T10:12:06","Uptime":"0T17:20:41","UptimeSec":62441,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"STONEY_2G","BSSId":"68:02:B8:2D:22:07","Channel":13,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1) log into webui on device
2) Select Firmware Upgrade
3a) use Start upgrade 
or
3b.1) Select tasmota-minimal.bin followed by Start Upgrade, 
3b.2) Repeat with tasmota.bin
 
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Version Upgrade (e.g. V9.5.0 to V10.0.0.2)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
For me, the console upgrade works great! This is actually simpler and it seems less troublesome was the browser based web ui upgrade!

I just wanted to share my experience in case I am not the only one encountering that issue.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-11-11:
Did you try to use the tasmota.bin.gz  version?

rsteininger, 2021-11-11:
@s-hadinger , No, I did not. I was not aware of the tasmota.bin.gz version! The only approach I knew was with the minimal followed by the "full" version with manual selection und upgrade. That worked fine up to V9.5.

Does the .bin.gz version "solve" the memory issue with "my" (older) devices?

THANKS

sfromis, 2021-11-11:
Always, always use the .gz files for OTA upgrading, if you want it to work, as long as you're at version 8.3 or later, even when using -minimal. If using the http links (not https) without downloading first, Tasmota will automatically use -minimal when relevant.

rsteininger, 2021-11-11:
ok. Thanks! I learned something new! I was not aware that .gz is preferable / possible over .bin! 
THANKS
This might (?) solve this issue I guess?

rsteininger, 2021-11-11:
I can confirm that the use of .gz files for manual upload (instead of the .bin files) solves this issue! 
THANKS for the pointers!

closed 2021-11-11 by rsteininger

tinel-c, 2022-01-02:
Have the same issue for the Sonoff bridge update over OTA for the .bin file. The .gz file works as expected.
The bug can be reproduced also for the custom setups using the usual NodeMcu Lua CH340G V3 Based on ESP8266.

Should be documented somehow in the release notes for tasmota.

barbudor, 2022-01-02:
@tinel-c it is already documented that OTA/upload shall use GZ and Bin is for serial flashing.
But feel free to propose a PR on the doc