arendst/Tasmota #13666: When configuring GPIO_DSB on GPIO21 it fails to boot with Core  1 panic'ed (LoadProhibited). Exception was unhandled.

mikep1998, 2021-11-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When configuring GPIO_DSB on GPIO21 (and other GPIOs) boot fails with 
```Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x400818ec  PS      : 0x00060f33  A0      : 0x8013aca2  A1      : 0x3ffb26e0  
A2      : 0x00000498  A3      : 0x00000015  A4      : 0x00000054  A5      : 0xbaad5678  
A6      : 0xb33fffff  A7      : 0x0000abab  A8      : 0x3f423a18  A9      : 0x06050304  
A10     : 0x00000507  A11     : 0x0000007c  A12     : 0x3ffc949c  A13     : 0x00060f23  
A14     : 0xb33fffff  A15     : 0x0000abab  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x06050304  LBEG    : 0x4008142c  LEND    : 0x40081439  LCOUNT  : 0x00000012  

Backtrace:0x400818e9:0x3ffb26e00x4013ac9f:0x3ffb2710 0x400f266b:0x3ffb2740 0x400f2744:0x3ffb2760 0x40113844:0x3ffb2780 0x400f90cb:0x3ffb27a0 0x400ff5a1:0x3ffb27c0 0x40140baa:0x3ffb2820 
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
 {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,0,1,1,1,0,1,1,1,1,1,0,1,0,1312,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-11-22:
Nice error. It fails on all GPIOs and after four restarts it just hangs.. A hard reset does finish the restart though.

Jason2866, 2021-11-22:
Used provided Template `{"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,0,1,1,1,0,1,1,1,1,1,0,1,0,1312,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}`
and get correct sensor readings now.
Bug fix confirmed.

closed 2021-11-22 by Jason2866

mikep1998, 2021-11-22:
nice recovery!

arendst, 2021-11-23:
@mikep1998 for my understanding how did you get the backtrace decodes? I know the map file but this info is way more useful!

mikep1998, 2021-11-23:
yes it is very useful. I'm using visual code and platformio. These have a ESP32 exception decoder. You can find all the information here. https://community.platformio.org/t/espressif-32-dev-platform-1-12-0/12610
You just install the filter and then run 'platformio.exe device monitor' in terminal. When it sees a backtrace it automatically decodes it.

barbudor, 2021-11-23:
Problem is that you need to have the ESP on a PC running vscode, waiting for the crash
It would be great if we can have this from a serial capture or from the details in the INFO message

mikep1998, 2021-11-23:
@barbudor How do you get a serial capture? Can this be done remotely? Or is this just from the web console?

barbudor, 2021-11-23:
No from the web, from the serial.
Sometime I have up to 2 or 3 ESP connected to my Linux box and running a serial terminal with logging to file.
You can use an old raspi
Or even another ESP such as tasmota using tcpbridge to get the serial log over network

mikep1998, 2021-11-24:
Wow, I looked at  tcpbridge but did not understand how to view the data on the side of the network. So I just wrote my own to read the serial port and log it using tasmota. then I just viewed it on tasmota web console. It work pretty good.

mikep1998, 2021-11-24:
Are you running Platformio on linus? or something else?