arendst/Tasmota #13667: VINDRIKTNING missing sinds last update?

Ferrarica, 2021-11-14:
### PROBLEM DESCRIPTION
VINDRIKTNING sensor missing sinds last update. Sinds this weekend i could not find this sensor in the module list.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-11-14:
Which build ESP32, ESP8266 ? Tasmota version?
There is a reason we have the template.

ascillato2, 2021-11-14:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-11-14 by ascillato2

Ferrarica, 2021-11-14:
Build 10? I used https://tasmota.github.io/install/ url.
To install to esp8266.

Last week this included the vindriktning sensor.
Today it disappeared.

Verstuurd vanaf mijn iPhone

barbudor, 2021-11-14:
VINDRIKNING has never been included in any of the standard build.
May be last week you flashed Tasmota-Allsensors ? 

[image]

Ferrarica, 2021-11-14:
Pretty sure it was there using the webinstall.
Today, exact the same procedure en not showing the sensor. 😿

Verstuurd vanaf mijn iPhone

barbudor, 2021-11-14:
Every firmware build used by the Web Installer are archived in Github here : https://github.com/tasmota/install/tree/main/firmware/development

Click on "History" on the top-right of the file-list
[image]

Navigate until you reach a date where you are pretty sure the feature was there and click on the right of the line on "Browse the repository at this point in the history"
[image]

You are now back in time
Go to `firmware/development` and find the firmware you want such as `tasmota.bin.gz` (for file upload from the WebGUI) or `tasmota.bin` for serial flashing with Tasmotizer or similar tool
[image]

DO NOT right-click to save the link as a file, it will not work
Click on the firmware name to go to the file page and there click on download
[image]

Good luck to find a firmware with VINDRIKTNING included

Ferrarica, 2021-11-14:
Thx I will try that. I let you know asap.

Could cost mena couple of days ending the weekend almost.

Regards richard 

Verstuurd vanaf mijn iPhone