arendst/Tasmota #13668: Problem NTC calculation

refael6, 2021-11-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Problem NTC calculation.
there are two issue related #10906 , #8628.
Both are closed but the problem remains.

I checked, there is a mistake in the formula, no need to multiply by 3.3V.

tasmota/xsns_02_analog.ino

double Rt = (adc * Adc[idx].param1) / (ANALOG_RANGE * ANALOG_V33 - (double)adc);
Calculation of units
resistance != (bit*resistance)/(bit*Voltage-bit)
resistance = (bit*resistance)/(bit-bit)

the Right formula:
double Rt = (adc * Adc[idx].param1) / (ANALOG_RANGE - (double)adc);

I test it on LilyGO TTGO ESP32.
after i change the formula it work.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-11-14:
The formula is adapted for the Shelly 2.5
It may not apply to a different board with a different schematic

What is the schematic of your LilyGO board ? Does it has a voltage divider like the Mini-D1 or NodeMCU ?

refael6, 2021-11-14:
I don't see why it will work for any board.
I went over the formula, and there is no reason to multiply by 3.3 or any voltage.

ascillato, 2021-11-14:
> I don't see why it will work for any board.

Because if the circuitry is different, the formula will be different.

For the case of Shelly, as it is now, it gives the correct temperature measurement. If you need to modify the ADC parameters for your own board, Tasmota can be adjusted using commands for ADC ( Docs: at https://tasmota.github.io/docs/Commands/#adcparam ). Thanks.

ascillato2, 2021-11-14:
For the case of Shelly, as it is now, it gives the correct temperature measurement. If you need to modify the ADC parameters for your own board, Tasmota can be adjusted using commands for ADC ( Docs: at https://tasmota.github.io/docs/Commands/#adcparam ). Thanks.

refael6, 2021-11-14:
Can you explain why you need to multiply by 3.3v?
I see no reason when I go over the formula of Steinhart-Hart equation.

as I said when I go through the units

[resistance] != ([bit]*[resistance])/([bit]*[Voltage]-[bit])
In terms of units it makes no sense [bit]*[Voltage]-[bit]

resistance = (bit*resistance)/(bit-bit)

Maybe the Shelly is the exception.

the AdcParam Can not help me because the multiplication by 3.3 is within the formula.
When I understand why multiplying by 3.3 I might be able to offer comprehensive solutions.
I used the formula without the multiplying at 3.3 and compil, The temperature was correct.

ascillato, 2021-11-14:
For the case of a Shelly, it is needed. If not, the value is wrong. For your circuitry you don't need it and that is ok.

> I used the formula without the multiplying at 3.3 and compil, The temperature was correct.

Great to know that you could adapt it for your device.

sfromis, 2021-11-14:
Just a guess. Multiplication by 3.3 looks like the difference between 1.0V (which is what the ESP8266 analog pin supports) and 3.3V (which due to a voltage divider is supported by the analog pin on standard ESP8266 dev boards). With you using ESP32 instead, the analog pin range is different, up to 3.3V without voltage divider, and may not be suitable for a formula specifically created for an ESP8266 device.

refael6, 2021-11-15:
I found the problem and the solution:
Because the adc max voltage in Shelly is 1 volt, it was Removed from the formula (No need to multiply by 1)
The full formula:
double Rt = (adc * Adc[idx].param1 * MAX_ADC_V) / (ANALOG_RANGE * ANALOG_V33 - (double)adc * MAX_ADC_V);

Now it gets along with the units:
resistance = (bit * resistance * Voltage) / (bit * Voltage - bit * Voltage)

in Shelly  (ESP8266)
MAX_ADC_V = 1
ANALOG_RANGE = 1023  

ESP32
MAX_ADC_V = 3.3
ANALOG_RANGE = 4095

no need to change in AdcParam.

Hope this helps

ascillato2, 2021-11-16:
Thanks.

PR https://github.com/arendst/Tasmota/pull/13703

refael6, 2021-11-17:
Thank you, happy to contribute my little part in a successful project that helps a lot of people.

closed 2021-11-17 by refael6