arendst/Tasmota #13683: switchmode 8 and setoption19

quantenschaum, 2021-11-15:
### PROBLEM DESCRIPTION
`switchmode 8` does not send `HOLD` via MQTT (switch flipped twice) if `setoption19 0`. 

Only when `setoption19 1` then an additional topic `stat/tasmota-xxxx/SWITCH1T` is published containing json with `TOGGLE/HOLD`.

Maybe this is the desired behavior, but I cannot find anything about this in the documentation.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: `{"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:23:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"362/699"}}`
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

18:09:33.009 MQT: stat/plug10/RESULT = {"NAME":"Gosund SP1","GPIO":[0,576,0,160,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":55}
18:09:33.231 MQT: stat/plug10/RESULT = {"Module":{"0":"Gosund SP1"}}
18:09:33.487 MQT: stat/plug10/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"576":"LedLink_i"},"GPIO2":{"0":"None"},"GPIO3":{"160":"Switch1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

18:10:19.744 MQT: stat/plug10/STATUS = {"Status":{"Module":0,"DeviceName":"plug10","FriendlyName":["plug10"],"Topic":"plug10","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[8,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:10:19.752 MQT: stat/plug10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:18:44","StartupUTC":"2021-11-15T16:51:35","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2021-11-15T17:37:55","SaveCount":98,"SaveAddress":"FA000"}}
18:10:19.762 MQT: stat/plug10/STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:23:49","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"362/699"}}
18:10:19.769 MQT: stat/plug10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["lorem","orre"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00004000"]}}
18:10:19.785 MQT: stat/plug10/STATUS4 = {"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:10:19.793 MQT: stat/plug10/STATUS5 = {"StatusNET":{"Hostname":"plug10","IPAddress":"192.168.222.167","Gateway":"192.168.222.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.222.10","DNSServer2":"0.0.0.0","Mac":"C8:2B:96:C9:BD:4C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:10:19.800 MQT: stat/plug10/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.222.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C9BD4C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:10:19.809 MQT: stat/plug10/STATUS7 = {"StatusTIM":{"UTC":"2021-11-15T17:10:19","Local":"2021-11-15T18:10:19","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:59","Sunset":"17:09"}}
18:10:19.816 MQT: stat/plug10/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:10:19.827 MQT: stat/plug10/STATUS10 = {"StatusSNS":{"Time":"2021-11-15T18:10:19","Switch1":"ON","ENERGY":{"TotalStartTime":"2021-11-15T17:37:55","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
18:10:19.835 MQT: stat/plug10/STATUS11 = {"StatusSTS":{"Time":"2021-11-15T18:10:19","Uptime":"0T00:18:44","UptimeSec":1124,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"lorem","BSSId":"98:9B:CB:A9:0A:49","Channel":11,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

using a device with a switch and relay, i.e. Shelly 1

- set `setoption19 0`
- set `switchmode 8`
- flip the switch twice withing 0.5s
- observe MQTT and find that *nothing* is sent

- set `setoption19 1`
- flip the switch twice withing 0.5s
- observe MQTT and find that `stat/tasmota-xxxx/SWITCH1T` is published

### EXPECTED BEHAVIOUR
Even with `setoption19 0` I would expect that some topic containing `HOLD` is published,as stated in https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode

It would be even better, if the device de/activated the relay on a single flip of the switch (or push of the button), but the the switch/button is flipped/pressed multiple times in a row, the relay stayed as it is, but an MQTT message is published containing the count of the switch flips/button presses. This would allow a switch/button to act as multi gesture input.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

If temporarily configured the button in the Gosund SP1 as switch to test the behavior. Holding the button = switch turned on, releasing the button = switch turned off.

**(Please, remember to close the issue when the problem has been addressed)**

quantenschaum, 2021-11-15:
https://discord.com/channels/479389167382691863/490244847568158751/909873748604252211

quantenschaum, 2021-11-15:
The [doc](https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode) says

>SwitchMode 8
>Set switch to multi change toggle mode (0 = TOGGLE, 1 = TOGGLE, 2x change = HOLD).
>Same as SwitchMode 0 but when the state of the circuit changes within 0.5s twice no TOGGLE commands are send but Tasmota sends HOLD (use Switch<x>#state=3 in rules).

"but Tasmota sends HOLD" does **not** refer to sending (publishing on MQTT).

A rule like `on Switch1#state=3 do publish stat/%topic%/SWITCH %value% endon`can be used to publish on MQTT when the switch is on `HOLD` .

`setoption19 1` enables the mentioned additional topic, but this option is deprecated.

closed 2021-11-15 by quantenschaum

barbudor, 2021-11-15:
SetOption114 to detach the switch from the relay would also do the job.
Using SetOption19 for that purpose would be a very bad idea considering the other side effects.

quantenschaum, 2021-11-16:
just for the record: I used Tasmota together with home assistant with SO19 enabled, the switch should trigger the relay but additionally a double switch flip would trigger an automation in HASS. After disabling SO19 (because the new HASS ´Tasmota integration made it obsolete) this was not working any more, because nothing got published on MQTT when the switch was flipped twice (with SO19 previously enabled it was).