arendst/Tasmota #13687: Feature Request - Shutter auto calibration

najdanovicivan, 2021-11-15:
### PROBLEM DESCRIPTION
This is feature request.

Regarding the Shutters and Blinds I have the motors which have mechanical limiters. Once the motor reaches the end position it will automatically stop. If will be great for devices which have means of measuring power consumption to implement some sort of auto calibration feature. Looking at the power consumption to automatically measure the time needed to open and close the shutters.

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-11-15:
Feature requests goes in Discussion/Ideas : https://github.com/arendst/Tasmota/discussions

Don't clutter Github Issues for something that is not a qualified bug

najdanovicivan, 2021-11-15:
Sorry. First time writing here. Will close this one and add it to the discussions

closed 2021-11-15 by najdanovicivan