arendst/Tasmota #13690: Tasmota32-webcam 10.0.0.3 reboots after any relay/pwn on/off

bob-tm, 2021-11-15:
### PROBLEM DESCRIPTION
Tasmota v10.0.0.3 reboots after turn on/turn off for any relays/pwm.
Tasmota v9.5 drops picture for half a second and then running fine with same board and config.

Template used from 
https://templates.blakadder.com/ai-thinker_ESP32-CAM.html

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [ ai-thinker_ESP32-CAM ]
- [x] Tasmota binary firmware version number used: 10.0.0.3
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP-Flasher-Windows-x64.exe
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
default config after initial reflash. 
```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
rule1 ON System#Boot DO backlog WcInit ENDON

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
default config after initial reflash. 

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
impossible due to unexpected reboots. 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
0. Problem reproduced on ESP32-CAM board without any additional sensors and modules with default firmware settings.
1. Flash Tasmota32-webcam
2. Apply template https://templates.blakadder.com/ai-thinker_ESP32-CAM.html
3. Turn ON/OFF build in Flash LED.
4. Device is going to reboot.
5. The same reboot happens after configure any free pin as relay and then turn it on/off
6. It's not power or hardware issue, because Tasmota 9.5 works fine with same settings.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
no unexpected reboot on pwm/relays turn on/off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-11-15:
Hi,

I can't reproduce your issue. Both symptoms seems to be a bad power supply. The ESP32CAM use more energy than an ESP32 alone

bob-tm, 2021-11-15:
I try to power from 6A lab power supply directly to 3.3v pin with same result. v10 reboots. v9.5 works fine. Problems happen while watching mjpeg stream and change led-flash state.

barbudor, 2021-11-15:
It's not only the power supply but the whole design of the board.
When the led is over ~70%, the voltage starts to drop on the board and the ESP32 doesn't have enough power.
Check the the RestartReason in the boot INFO message or the logs on the serial console

bob-tm, 2021-11-16:
I can assign (without connecting it) Relay to any free pin. Toggle it and device reboots. Voltage is not changed in that case. So it's not power problem. I even try to  increase voltage to 3.4 and 3.5. Maybe its bad ESP CHIP, but why Tasmota 9.5 works fine? 

It looks like unhandled exception in mjpeg streaming. Somehow it conflicts with events from on/off. 9.5 in that case just drops several frames, but then works fine.

SilentBob83, 2021-11-17:
did you use a freewheeling diode ? it is absolutely necessary ....

Jason2866, 2021-11-19:
The issue you have is not a software problem. EspCam is a very sensible board regarding power supply (glitches and spikes). The board design does not cover this good enough. So a slightly changed software can produce problems where a other version does not have. If you search you will find many issue about the esp32 cam.

closed 2021-11-19 by Jason2866