arendst/Tasmota #13694: Sensor AS608 RX/TX missing in sensor list

nicx, 2021-11-16:
### PROBLEM DESCRIPTION
I want to use a R503 fingerprint sensor which is supported as AS608 sensor (https://tasmota.github.io/docs/Supported-Peripherals/) since firmware version 9.2.0.
I flashed the Tasmota-sensors.bin to my ESB8266, but I cannot find the sensor "AS608 RX" and "AS608 TX" in the sensors dropdown.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x ] Device used (e.g., Sonoff Basic): ESP8266 Generic
- [ x ] Tasmota binary firmware version number used: 10.0.0
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ x ] Flashing tools used: Arduino IDE
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Go  via GUI to Configuration -> Configure Module -> click on an pulldown for any pin -> search for "AS608 RX" or "AS608 TX"

### EXPECTED BEHAVIOUR
Go  via GUI to Configuration -> Configure Module -> click on an pulldown for any pin -> search for "AS608 RX" or "AS608 TX" and find it :)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2021-11-16:
As per documentation, this is an I2C device.
So you have to assign the two I2C options to the appropriate GPIOs. The AS608 should be identified automatically then.

nicx, 2021-11-16:
@schrej-zz The sensor itself is no I2C device, it is a TTL serial device. See https://cdn-learn.adafruit.com/downloads/pdf/adafruit-optical-fingerprint-sensor.pdf
I think the documentation is just wrong here ;)

And as far as I understand this the problem should already be solved, but it isn't: https://github.com/arendst/Tasmota/pulls?q=is%3Apr+as608+is%3Aclosed

sfromis, 2021-11-16:
To enable AS608 support, you need [your own build](https://tasmota.github.io/docs/Compile-your-build/) with `USE_AS608`, as it is not included in precompiled tasmota-sensors.

schrej-zz, 2021-11-16:
...and/or you can use the unsupported [tasmota-allsensors.bin](https://github.com/tasmota/install/raw/main/firmware/unofficial/tasmota-allsensors.bin) for a first test.

Jason2866, 2021-11-16:
Closing since the question is answered.

closed 2021-11-16 by Jason2866

nicx, 2021-11-16:
@Jason2866 thanks for closing, but the documentation is still wrong ;) Maybe this issue should be used to get this updated by someone.

ozett, 2022-04-27:
installed the allsensors.bin and configured tx and rx
It is the R557 Model
[R557 fingerprint module user manual (1).pdf](https://github.com/arendst/Tasmota/files/8572512/R557.fingerprint.module.user.manual.1.pdf)

after changing rx and tx now it changes ringlight from blue to red,
while putting an unknown finger onto it.

BUT how do i enroll a new finger with tasmota?
cant see anything in the mqtt-logs.

How do i control the fingerprint reader with tasmota?

thanks for any information...

ozett, 2022-12-22:
i upgraded from 11.1.x to 12.3.x and now AS608 is missing as TX/RX.

did the all-sensors.bin change without notice?