arendst/Tasmota #13720: NTC not reporting correct value after change 'Fix NTC Temp support for ESP32'

mikep1998, 2021-11-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
NTC not reporting correct value after change 'Fix NTC Temp support for ESP32'
before the change my NTC temperature sensor defined as GPIO_ADC_TEMP was displaying a temp 60 F. 
After the change GPIO_ADC_TEMP is displaying a temp -41 F.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-11-18:
Hi, can you explain how is the circuit you are using for measuring the temperature?

mikep1998, 2021-11-18:
Sure, I set it up like this article says https://learn.adafruit.com/thermistor/using-a-thermistor
I did this on a ESP32.
I used   3.3v ---- 10K resister --- Analog pin --- 10K thermistor  --- ground.

refael6, 2021-11-18:
Hi, For me before the Fix the temperature showed 70 C (not F) , Maybe check in that direction...
If you switch to GPIO_ADC_INPUT what is the value? (in room temp)

mikep1998, 2021-11-18:
I tried GPIO_ADC_INPUT. Before the temp reading was 68.7 F. Now the INPUT reading is 3651 (+-20)

ascillato, 2021-11-18:
For reference:

The PR which this issue refers is: https://github.com/arendst/Tasmota/pull/13703
That PR aimed to solve: https://github.com/arendst/Tasmota/issues/13668
Previous related discussions: https://github.com/arendst/Tasmota/issues/10906, https://github.com/arendst/Tasmota/pull/8628

ascillato, 2021-11-18:
> I tried GPIO_ADC_INPUT. Before the temp reading was 68.7 F. Now the INPUT reading is 3651 (+-20)

What do you have in ADCPARAM?

barbudor, 2021-11-18:
ADC_INPUT mode is independant of AdcParams
It is hte raw ADC value without any scaling.
On ESP32 the range of the ADC is 0 ... 4095 for 0 ... 3.3V (raw range at the ESP pin input - some dev board may add voltage divider)

mikep1998, 2021-11-18:
Closing PR because the 10K thermistor was actually a 50K thermistor. Once I installed a 10k thermistor the reading were correct without the 3.3 voltage in the equation.

closed 2021-11-18 by mikep1998

ascillato, 2021-11-19:
Great. Thanks for sharing your solution.