arendst/Tasmota #13721: Aqara Motion Sensor reports Xiaomi_64 in periodic tele report

rebel7580, 2021-11-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The Aqara Motion Sensor (Model RTCGQ11LM, https://zigbee.blakadder.com/Aqara_RTCGQ11LM.html) reports Xiaomi_64 in periodic tele report.
I'm not 100% sure, but I think it is likely that it should be "Occupancy", although it's possible it could be "Illuminance"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [x ] Tasmota binary firmware version number used: 9.3.1.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua

  Configuration output here:
23:18:49.044 CMD: Backlog Template; Module; GPIO 255
23:18:49.105 MQT: stat/tasmota_zbBridge/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:18:49.312 MQT: stat/tasmota_zbBridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
23:18:49.564 MQT: stat/tasmota_zbBridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

23:20:53.126 CMD: status 0
23:20:53.131 MQT: stat/tasmota_zbBridge/STATUS = {"Status":{"Module":75,"DeviceName":"zbbridge","FriendlyName":["zbBridge"],"Topic":"tasmota_zbBridge","ButtonTopic":"0","Power":0,"PowerOnState":4,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:20:53.135 MQT: stat/tasmota_zbBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"Boston","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"22T20:19:50","StartupUTC":"2021-10-26T08:01:03","Sleep":50,"CfgHolder":4617,"BootCount":161,"BCResetTime":"2021-02-15T07:07:42","SaveCount":223,"SaveAddress":"FB000"}}
23:20:53.138 MQT: stat/tasmota_zbBridge/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(zbbridge)","BuildDateTime":"2021-04-04T20:53:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"400/699"}}
23:20:53.141 MQT: stat/tasmota_zbBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["beantown",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","40006080","00000030"]}}
23:20:53.146 MQT: stat/tasmota_zbBridge/STATUS4 = {"StatusMEM":{"ProgramSize":575,"Free":1216,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
23:20:53.151 MQT: stat/tasmota_zbBridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota_zbBridge-5317","IPAddress":"192.168.123.31","Gateway":"192.168.123.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.123.1","Mac":"84:CC:A8:9D:14:C5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:20:53.155 MQT: stat/tasmota_zbBridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.123.176","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D14C5","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:20:53.160 MQT: stat/tasmota_zbBridge/STATUS7 = {"StatusTIM":{"UTC":"2021-11-18T04:20:53","Local":"2021-11-17T23:20:53","StartDST":"2021-03-14T02:00:00","EndDST":"2021-11-07T02:00:00","Timezone":99,"Sunrise":"06:44","Sunset":"16:36"}}
23:20:53.163 MQT: stat/tasmota_zbBridge/STATUS10 = {"StatusSNS":{"Time":"2021-11-17T23:20:53"}}
23:20:53.168 MQT: stat/tasmota_zbBridge/STATUS11 = {"StatusSTS":{"Time":"2021-11-17T23:20:53","Uptime":"22T20:19:50","UptimeSec":1973990,"Vcc":3.386,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"beantown","BSSId":"3C:37:86:52:48:14","Channel":5,"RSSI":100,"Signal":-23,"LinkCount":2,"Downtime":"0T00:00:54"}}}
23:21:03.063 MQT: tele/tasmota_zbBridge/STATE = {"Time":"2021-11-17T23:21:03","Uptime":"22T20:20:00","UptimeSec":1974000,"Vcc":3.386,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"beantown","BSSId":"3C:37:86:52:48:14","Channel":5,"RSSI":100,"Signal":-23,"LinkCount":2,"Downtime":"0T00:00:54"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Periodic tele Report:

14:34:48.849 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","BatteryVoltage":3.2,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":79}

Triggered tele Report (due to motion detected):

22:58:01.189 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","Illuminance":0,"Occupancy":1,"Endpoint":1,"LinkQuality":79}

ZBInfo:

22:57:08.086 MQT: tele/tasmota_zbBridge/Motion5/ZbInfo = {"Device":"0xED02","Name":"Motion5","IEEEAddr":"0x00158D0006F7D6F3","ModelId":"lumi.sensor_motion.aq2","Endpoints":[1],"Config":["I01"],"Illuminance":6,"Occupancy":0,"Reachable":true,"BatteryPercentage":100,"LastSeen":756,"LastSeenEpoch":1637207072,"LinkQuality":81}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Pair Aqara motion sensor, wait for periodic tele report.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

"Xiaomi_64" shold be replaced by "Occupancy" (I think!)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-11-18:
Please use the latest actual version. Many changes / fixes since your used version

rebel7580, 2021-11-19:
Same in 10.0.0.3 lastest dev release.

Periodic tele Report:

19:39:48.581 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","BatteryVoltage":3.2,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":79}

s-hadinger, 2021-11-19:
Can you please paste here the result of `ZbInfo 0xED02`

rebel7580, 2021-11-19:
Already in the original:

ZBInfo:
```
22:57:08.086 MQT: tele/tasmota_zbBridge/Motion5/ZbInfo = {"Device":"0xED02","Name":"Motion5","IEEEAddr":"0x00158D0006F7D6F3","ModelId":"lumi.sensor_motion.aq2","Endpoints":[1],"Config":["I01"],"Illuminance":6,"Occupancy":0,"Reachable":true,"BatteryPercentage":100,"LastSeen":756,"LastSeenEpoch":1637207072,"LinkQuality":81}
```

s-hadinger, 2021-11-19:
Thanks I have the same device. The value of attribute 100 (0x64) is not documented when it is in a battery reporting message, which are different from the presence messages.

I don't think anyone ever correlated this value to occupancy. Again I'm talking about this very specific message of reporting battery state.

Just ignore this value.

ascillato2, 2021-11-19:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-11-19 by ascillato2

s-hadinger, 2021-11-20:
Surprisingly, it works if you set `DS18x20` on `GPIO4`.

I have the impression that the GPIO direct access from this module is not compatible with ESP32. The timing system is also not well suited for ESP32, which has RMT hardware instead.

I believe that we'll need to find a more recent lib.

s-hadinger, 2021-11-20:
@arlosi any hints?

rebel7580, 2021-11-21:
FWIW, for the record:
I monitored tele reports when the Illuminance is something other than 0, and the Xiaomi_64 remained 0:
```
09:48:37.542 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","Illuminance":51,"Occupancy":1,"Endpoint":1,"LinkQuality":76}
09:54:43.547 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","BatteryVoltage":3.17,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":73}
10:31:14.611 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","Illuminance":23,"Occupancy":1,"Endpoint":1,"LinkQuality":92}
10:48:40.898 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","BatteryVoltage":3.17,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":76}
11:27:32.355 MQT: tele/tasmota_zbBridge/Motion5/ZbReceived = {"Device":"0xED02","Name":"Motion5","Illuminance":39,"Occupancy":1,"Endpoint":1,"LinkQuality":68}
```
So we can probably rule out Illuminance for the battery report.  But anyway, isn't really important what it is. :)