arendst/Tasmota #13754: sonoff mini ghost switching when using vpn

heizie, 2021-11-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The sonoff mini was flashed with version 9.1 since a year. It works with Home Assistant via MQTT. Everything fines in this year.
But in these days, I’ve used a VPN (for speed up gaming), one of all tasmota devices start ghost switching. I’ve upgraded it to 10.0, the problem remain.
I’ve started watching the console in the specific device. It didn’t shows new command before ghost switching, but just restart without reason. And the relay release and the close again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [*] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [*] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [*] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [*] Device used (e.g., Sonoff Basic): Sonoff Mini
- [*] Tasmota binary firmware version number used: 9.1, 10.0
  - [*] Pre-compiled
  - [ ] Self-compiled
- [*] Flashing tools used: OTA
- [*] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
```
15:41:27.969 MQT: tasmota/stat/light_storage_02/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:41:28.193 MQT: tasmota/stat/light_storage_02/RESULT = {"Module":{"1":"Sonoff Basic"}}
15:41:28.449 MQT: tasmota/stat/light_storage_02/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

```
- [*] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

No rules

```
- [*] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```

15:42:47.815 CMD: Status 0
15:42:47.822 MQT: tasmota/stat/light_storage_02/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"light_storage_02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:42:47.828 MQT: tasmota/stat/light_storage_02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"1T00:10:58","StartupUTC":"2021-11-20T14:31:49","Sleep":50,"CfgHolder":4617,"BootCount":104,"BCResetTime":"2020-12-16T21:48:01","SaveCount":1074,"SaveAddress":"F9000"}}
15:42:47.834 MQT: tasmota/stat/light_storage_02/STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021-10-19T08:23:49","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"383/699"}}
15:42:47.839 MQT: tasmota/stat/light_storage_02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["2L",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00004000"]}}
15:42:47.851 MQT: tasmota/stat/light_storage_02/STATUS4 = {"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:42:47.857 MQT: tasmota/stat/light_storage_02/STATUS5 = {"StatusNET":{"Hostname":"light-storage-02-5537","IPAddress":"192.168.0.98","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"C8:2B:96:DB:B5:A1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:42:47.861 MQT: tasmota/stat/light_storage_02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.220","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DBB5A1","MqttUser":"MQTT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:42:47.868 MQT: tasmota/stat/light_storage_02/STATUS7 = {"StatusTIM":{"UTC":"2021-11-21T14:42:47","Local":"2021-11-21T15:42:47","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:08","Sunset":"17:02"}}
15:42:47.874 MQT: tasmota/stat/light_storage_02/STATUS10 = {"StatusSNS":{"Time":"2021-11-21T15:42:47","Switch1":"ON"}}
15:42:47.904 MQT: tasmota/stat/light_storage_02/STATUS11 = {"StatusSTS":{"Time":"2021-11-21T15:42:47","Uptime":"1T00:10:58","UptimeSec":87058,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"2L","BSSId":"7C:FF:4D:E9:A8:7D","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

```
- [*] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Difficult, the device restart itself when it meets the issue

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Relay closed
2. Using vpn and play games (lol mobile…) for about 15 min

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

It will ghost switching

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

N/A

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-11-21:
No offend, do you realy think we install china spy/bloat ware to test?

closed 2021-11-21 by Jason2866

ascillato, 2021-11-21:
Hi,

You said that a sonoff mini is restarting but the logs you show are from a sonoff basic that have been left without power (RESTART REASON: POWER ON).

So, seems that your VPN issue is unrelated to your issue. Seems that your sonoff basic is just left without power (bad wiring) or it is just failing its internal regulator.

We can't help from Tasmota side if you have wiring issues or hardware failures. Sorry.

heizie, 2021-11-21:
> No offend, do you realy think we install china spy/bloat ware to test?

china spy ware? i think this repo consider only technical but nothing racist

Jason2866, 2021-11-21:
Installing software tested from Apple is safe. Great to know! In other words we do not
install any not Open Source Software for testing Tasmota.
So it could have been American bloat/spyware too.

barbudor, 2021-11-21:
@heizie 
I understand it is not ghost switching as in receiving switching commands but Restarting with a side effect
Does the ghost switching also happens when you restart the unit manually from the GUI or console? 

This device could have different setting from the others?
Is it using different features?
Are you using Device Groups?

heizie, 2021-11-21:
@Jason2866 
let me guess, you are actually from Germany as you have written in your file. Sure. you guys worry about everything.
But I don't care whether my mask contains 5G antenna. 
If you say you won't test it, then don't. 
I just want to kindly support you by letting you know what could be wrong. But sadly, I'm now worrying about my devices with tasmota, which are developed by someone with misjudgment.

@barbudor 
So, basically, this device runs nice and clean this year. but since I've used VPN software, it starts ghost switching. but **only this single device** (I have several different sonoff devices running with tasmota). 
and these devices are using the default setting, which means, I've only mapped the gpio pins, nothing else.

Jason2866, 2021-11-21:
@heizie who is the racist? All Germans are worrying about anything. Ahh yes.

barbudor, 2021-11-21:
@heizie
Are you running the VPN only when you play or all the time ?
Have you really made a strong correlation between VPN and restart of the unit ?
May be you should run a Wireshark when the VPN is running to see what this software is doing. Possibly some conflicts on ports or something ?
I was thinking about DevicesGroup because it uses Multicast UDP which could be something the VPN is using too.
```
  #define DEVICE_GROUPS_ADDRESS 239,255,250,250  // Device groups multicast address
  #define DEVICE_GROUPS_PORT 4447                // Device groups multicast port
```

heizie, 2021-11-21:
> @heizie Are you running the VPN only when you play or all the time ? Have you really made a strong correlation between VPN and restart of the unit ? May be you should run a Wireshark when the VPN is [...]

@barbudor 
Yes. I need this VPN to minimize the latency. 
I've noticed this issue for some time. The most possible reason i think should be the VPN. But I'll try the Wireshark.

> I was thinking about DevicesGroup because it uses Multicast UDP which could be something the VPN is using too.
> 
> ```
>   #define DEVICE_GROUPS_ADDRESS 239,255,250,250  // Device groups multicast address
>   #define DEVICE_GROUPS_PORT 4447                // Device groups multicast port
> ```
probably no... I'll let you know once I catch the issue again with Wireshark.