arendst/Tasmota #13774: Sonoff Mini with Tasmota Restart an Reset after configure Button_i mode

talleraxn, 2021-11-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Sonoff Mini with Tasmota Restart few minutes after configure Button_i mode and save

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [+] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [+] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [+] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [+] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [+] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [+] Device used (e.g., Sonoff Basic): _____ Sonoff Mini
- [ +] Tasmota binary firmware version number used: _____ 10.0.0 Lite
  - [+] Pre-compiled
  - [ ] Self-compiled
- [+] Flashing tools used: _____ UPGRADE Option
- [+] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

00:00:37.043 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:00:37.249 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
00:00:37.455 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"97":"Button_i2"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [+] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

00:00:36.023 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
00:00:36.230 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
00:00:36.480 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [+] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:00:42.224 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Luz_BañoA_02","FriendlyName":["Luz_BañoA_02"],"Topic":"Luz_BañoA_02","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:00:42.250 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:17","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-08-01T00:40:30","SaveCount":8,"SaveAddress":"F8000"}}
00:00:42.282 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0(lite)","BuildDateTime":"2021-10-19T08:23:48","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"411/699"}}
00:00:42.300 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Javy_Movistar","Home_Moran927"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00004000"]}}
00:00:42.329 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":500,"Free":500,"Heap":30,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000","00000020"],"Drivers":"1,2,3,4,9,10,12,16,20,21,37","Sensors":"3"}}
00:00:42.357 RSL: STATUS5 = {"StatusNET":{"Hostname":"Luz-Ba--oA-02-6453","IPAddress":"192.168.100.113","Gateway":"192.168.1.1","Subnetmask":"255.255.0.0","DNSServer1":"186.130.128.250","DNSServer2":"0.0.0.0","Mac":"DC:4F:22:BE:59:35","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:00:42.382 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.1","MqttPort":1883,"MqttClientMask":"Luz_BañoA_02","MqttClient":"Luz_BañoA_02","MqttUser":"CasaJavy","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:00:42.404 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:00:42","Local":"1970-01-01T00:00:42","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:00:42.422 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:42"}}
00:00:42.431 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:42","Uptime":"0T00:00:17","UptimeSec":17,"Vcc":3.498,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":2,"SSId":"Home_Moran927","BSSId":"84:AA:9C:EE:0F:8C","Channel":11,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [+] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

00:00:37.176 CMD: weblog 4
00:00:37.180 RSL: RESULT = {"WebLog":4}
00:00:37.602 CFG: Saved to flash at F7, Count 9, Bytes 4096
00:00:41.040 MQT: Attempting connection...
00:00:41.052 MQT: Connect failed to 192.168.100.1:1883, rc -2. Retry in 20 sec
00:00:48.787 WIF: Checking connection...
00:01:02.004 MQT: Attempting connection...
00:01:02.015 MQT: Connect failed to 192.168.100.1:1883, rc -2. Retry in 30 sec
00:01:08.779 WIF: Checking connection...
00:01:11.048 SRC: Button
00:01:11.049 CMD: Grp 0, Cmnd 'RESET', Idx 1, Len 1, Data '1'
00:01:11.052 RSL: RESULT = {"Reset":"Reset and Restarting"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Configure GPIO4 as Button_i 2, save. test behavior of button (no necesarry) this work fine about any seconds or minutes. And not response anymore.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

That continue working

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[Documento sin título.pdf](https://github.com/arendst/Tasmota/files/7584793/Documento.sin.titulo.pdf)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-11-23:
Hi,

Please update to latest from development branch and test again ( http://ota.tasmota.com/tasmota/tasmota-lite.bin.gz ). You can do an OTAurl to this version. Thanks.

talleraxn, 2021-11-23:
Hi, no work, same problem.

00:00:00.002 HDW: ESP8285
00:00:00.051 CFG: Loaded from flash at F8, Count 12
00:00:00.056 QPC: Count 1
00:00:00.062 Project tasmota - Luz_BañoA_02 Version 10.0.0.3(lite)-2_7_4_9(2021-11-22T19:07:30)
00:00:00.551 WIF: Connecting to AP2 Home_Moran927 Channel 11 BSSId 84:AA:9C:EE:0F:8C in mode 11n as Luz-Ba--oA-02-6453...
00:00:01.752 WIF: Connected
00:00:02.004 HTP: Web server active on Luz-Ba--oA-02-6453 with IP address 192.168.100.113
21:41:57.023 MQT: Attempting connection...
21:41:57.033 MQT: Connect failed to 192.168.100.1:1883, rc -2. Retry in 10 sec
21:42:00.513 QPC: Reset
21:42:08.025 MQT: Attempting connection...
21:42:08.037 MQT: Connect failed to 192.168.100.1:1883, rc -2. Retry in 20 sec
21:42:14.892 RSL: RESULT = {"POWER":"OFF"}
21:42:14.893 RSL: POWER = OFF
21:42:16.105 RSL: RESULT = {"POWER":"ON"}
21:42:16.107 RSL: POWER = ON
21:42:24.580 CMD: time
21:42:24.584 RSL: RESULT = {"Time":"2021-11-22T21:42:24"}
21:42:29.026 MQT: Attempting connection...
21:42:29.034 MQT: Connect failed to 192.168.100.1:1883, rc -2. Retry in 30 sec
21:42:38.689 RSL: RESULT = {"Reset":"Reset and Restarting"}

ascillato, 2021-11-23:
Hi, please, go to the console and type:
```lua
SO1 1
```
Then try again. Seems that your button is being detected as continously pressed so, after 40 seconds of a button being pressed, Tasmota will reset everything to default. The `SetOption1 1` disables that feature.

talleraxn, 2021-11-23:
Thanks.

I already tried it and it is working fine. I will also continue testing it for a couple of continuous days to see that it is stable.

I really appreciate your help, it was very quick and easy to follow.

I apologize for not suspecting that, as the contact was open, I did not associate it with a button pressed.

ascillato, 2021-11-23:
Great to know that now it is working fine for you.

closed 2021-11-23 by ascillato2

barbudor, 2021-11-23:
Why would you configure the Button as inverted as the template says to use non-inverted?
You just doing it wrong