arendst/Tasmota #13809: Tasmota reports different 'fingerprint learned' value (after config reset) than configured in user_config_override.h

sander1988, 2021-11-26:
### PROBLEM DESCRIPTION
MQTT TLS fingerprint validation works with the defined value **MQTT_FINGERPRINT1** but Tasmota reports it's autolearned the first time after a configuration reset:
``` 
16:55:44.585 MQT: TLS connected in 328 ms, max ThunkStack used 2788
16:55:44.586 MQT: Fingerprint learned: AE C1 F6 0E 59 32 A6 88 AA 9C 00 C4 DF 80 59 80 4F 8E FE 47
16:55:44.635 MQT: Connected
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

NO RULES DEFINED
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

16:57:15.675 MQT: tasmota/XXXXXX/STATUS = {"Status":{"Module":1,"DeviceName":"Switch","FriendlyName":["Switch"],"Topic":"XXXXXX","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:57:15.710 MQT: tasmota/XXXXXX/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software/System restart","Uptime":"0T00:01:39","StartupUTC":"2021-11-26T15:55:36","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2021-11-26T16:55:36","SaveCount":7,"SaveAddress":"F5000"}}
16:57:15.743 MQT: tasmota/XXXXXX/STATUS2 = {"StatusFWR":{"Version":"9.3.0(tasmota)","BuildDateTime":"2021-11-26T16:26:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"299/699"}}
16:57:15.765 MQT: tasmota/XXXXXX/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.1.100","LogPort":10514,"SSId":["xxxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00206000","00000000"]}}
16:57:15.798 MQT: tasmota/XXXXXX/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","088004C6","04000001","00000047","010001C0","40000900","00000000","00001000","00000000"],"Drivers":"1,2,3,4,8,10,27,29,45","Sensors":"1,2,3,5"}}
16:57:15.832 MQT: tasmota/XXXXXX/STATUS5 = {"StatusNET":{"Hostname":"XXXXXX-1637","IPAddress":"192.168.1.149","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"xxxxxxx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:57:15.854 MQT: tasmota/XXXXXX/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.ta","MqttPort":8884,"MqttClientMask":"TA_%06X","MqttClient":"TA_XXXXXX","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:57:15.875 MQT: tasmota/XXXXXX/STATUS7 = {"StatusTIM":{"UTC":"2021-11-26T15:57:15","Local":"2021-11-26T16:57:15","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99}}
16:57:15.890 MQT: tasmota/XXXXXX/STATUS10 = {"StatusSNS":{"Time":"2021-11-26T16:57:15"}}
16:57:15.907 MQT: tasmota/XXXXXX/STATUS11 = {"StatusSTS":{"Time":"2021-11-26T16:57:15","Uptime":"0T00:01:39","UptimeSec":99,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"PNET_HC","BSSId":"xxxxxxx","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

17:01:36.327 MQT: tasmota/xxxxxx/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:01:36.557 MQT: tasmota/xxxxxx/RESULT = {"Module":{"1":"Sonoff Basic"}}
17:01:36.808 MQT: tasmota/xxxxxx/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

### TO REPRODUCE

1. Add MQTT TLS configuration and build the firmware:
``` 
  #undef USE_MQTT_TLS
  #define USE_MQTT_TLS

  #undef USE_MQTT_TLS_CA_CERT

  #undef MQTT_FINGERPRINT1
  #define MQTT_FINGERPRINT1 0x68,0x70,0xC5,0x3E,0xEA,0x3A,0x7D,0xD4,0x6D,0xA4,0xC6,0x14,0x5E,0x72,0x96,0x48,0x59,0x23,0xEF,0x53
  #undef MQTT_FINGERPRINT2
  #define MQTT_FINGERPRINT2 0x68,0x70,0xC5,0x3E,0xEA,0x3A,0x7D,0xD4,0x6D,0xA4,0xC6,0x14,0x5E,0x72,0x96,0x48,0x59,0x23,0xEF,0x53
```

2. Upload the firmware to a Tasmota device.
3. Reset the configuration (via webgui).
4. After the reboot it reports the fingerprint was learned
```
MQT: Fingerprint learned: AE C1 F6 0E 59 32 A6 88 AA 9C 00 C4 DF 80 59 80 4F 8E FE 47
```

### EXPECTED BEHAVIOUR
I would **not** expect to see the 'Fingerprint learned' message after a configuration reset. Atleast not a different fingerprint value.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
- **The fingerprint validation itself seems to work fine**. When I build the firmware with the wrong MQTT_FINGERPRINT1/MQTT_FINGERPRINT2 values it fails with TLS error 1 (bad fingerprint) as expected.
- The fingerprint of server MQTT certificate was generated using the recommended tool (https://github.com/issacg/tasmota-fingerprint).
- The learn action updates the config values:
```
17:10:57.494 CMD: mqttfingerprint1
17:10:57.505 MQT: tasmota/xxxxxx/RESULT = {"MqttFingerprint1":"AE C1 F6 0E 59 32 A6 88 AA 9C 00 C4 DF 80 59 80 4F 8E FE 47"}
17:11:00.197 CMD: mqttfingerprint2
17:11:00.208 MQT: tasmota/xxxxxx/RESULT = {"MqttFingerprint2":"AE C1 F6 0E 59 32 A6 88 AA 9C 00 C4 DF 80 59 80 4F 8E FE 47"}
```
- **Maybe this is only a cosmetic issue.** But I would like to be sure as it's an important part of the MQTT/IOT security of the device.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-11-26:
I suspect this is a compilation issue, your `  #define MQTT_FINGERPRINT1 0x68,0x70,0xC5,0x3E,0xEA,0x3A,0x7D,0xD4,0x6D,0xA4,0xC6,0x14,0x5E,0x72,0x96,0x48,0x59,0x23,0xEF,0x53` is placed in a location where it is actually replaced by the default value.

What is the value of `MqttFingerprint1` after a reset and before the first connection?

sander1988, 2021-11-26:
> I suspect this is a compilation issue, your ` #define MQTT_FINGERPRINT1 0x68,0x70,0xC5,0x3E,0xEA,0x3A,0x7D,0xD4,0x6D,0xA4,0xC6,0x14,0x5E,0x72,0x96,0x48,0x59,0x23,0xEF,0x53` is placed in a location [...]

I blocked the device using the firewall of the broker, performed a configuration reset and checked again...
```
23:12:12.437 MQT: Attempting connection...
23:12:23.043 MQT: TLS connection error: -1002
23:12:23.044 MQT: Connect failed to mqtt.ta:8884, rc -2. Retry in 20 sec
23:12:23.062 CMD: mqttfingerprint1
23:12:23.067 RSL: tasmota/xxxxxx/RESULT = {"MqttFingerprint1":"68 70 C5 3E EA 3A 7D D4 6D A4 C6 14 5E 72 96 48 59 23 EF 53"}
23:12:30.794 CMD: MqttFingerprint2
23:12:30.799 RSL: tasmota/xxxxxx/RESULT = {"MqttFingerprint2":"68 70 C5 3E EA 3A 7D D4 6D A4 C6 14 5E 72 96 48 59 23 EF 53"}
```

So it seems to match the fingerprint configured during compilation.

s-hadinger, 2021-11-27:
Thanks. That's definitely a bug. I will take the opportunity to enable dual mode (CA validation and Fingerprints) in the same binary.

sander1988, 2021-11-30:
I did some more testing. Based on these results I think it's not really a bug but a undocumented feature that should be documented here: https://tasmota.github.io/docs/TLS/

I found some fingerprint migration code in source that's by default enabled (when USE_MQTT_TLS_DROP_OLD_FINGERPRINT is not set). My fingerprint is generated with the [recommended tool ](https://github.com/issacg/tasmota-fingerprint), but this is the old insecure method?

There seems to be no issue when build with USE_MQTT_TLS_DROP_OLD_FINGERPRINT and using the new style fingerprint in the user_config_override.h:
```
#undef MQTT_FINGERPRINT1
  #define MQTT_FINGERPRINT1 0xAE,0xC1,0xF6,0x0E,0x59,0x32,0xA6,0x88,0xAA,0x9C,0x00,0xC4,0xDF,0x80,0x59,0x80,0x4F,0x8E,0xFE,0x47
  #undef MQTT_FINGERPRINT2
  #define MQTT_FINGERPRINT2 0xAE,0xC1,0xF6,0x0E,0x59,0x32,0xA6,0x88,0xAA,0x9C,0x00,0xC4,0xDF,0x80,0x59,0x80,0x4F,0x8E,0xFE,0x47
``` 

I got the right fingerprint value from the earlier log "Fingerprint learned". Any idea how to calculate it? E.g. using openssl.

s-hadinger, 2021-11-30:
Ah yes. There was a security flaw in the first version of fingerprint, the new is more secure and converts automatically fingerprints the first time

s-hadinger, 2021-11-30:
Honestly the simplest tool to compute the fingerprint would be to run Tasmota in auto learn mode and note the recorded fingerprint

ascillato2, 2021-12-02:
https://github.com/arendst/Tasmota/pull/13887

closed 2021-12-02 by ascillato2