arendst/Tasmota #13839: EnergyReset replaced, "set StartTotalTime" not possible anymore

Bascht74, 2021-11-28:
### PROBLEM DESCRIPTION
I replaced a device (a shelly 1PM) and restored a configuration backup. But the power values are not in the backup, so I had to use the "energy" commands to set the values again. 
I can set the energytotal value, but i cannot set the StartTotalTime, because `EnergyReset` is replaced by new commands. 
With the new commands there is no way to set the StartTotalTime to e.g. 2020-12-06 (epoch 1607275350) for the shelly replacement.
I haven't found any information how to do it with V10 of Tasmota

Thx 
Sebastian

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 1 PM
- [X] Tasmota binary firmware version number used: Tasmota 10.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
- [] Flashing tools used:
- [S] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

18:36:34.351 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/RESULT = {"NAME":"Shelly 1PM","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}
18:36:34.560 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/RESULT = {"Module":{"0":"Shelly 1PM"}}
18:36:34.817 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:37:28.275 CMD: Status 0
18:37:28.282 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS = {"Status":{"Module":0,"DeviceName":"Geraet_EG_Hauswirtschaftsraum_Router","FriendlyName":["Geraet_EG_Hauswirtschaftsraum_Router"],"Topic":"Geraet_EG_Hauswirtschaftsraum_Router","ButtonTopic":"","Power":1,"PowerOnState":4,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
18:37:28.288 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:16:56","StartupUTC":"2021-11-28T17:20:32","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-12-27T23:40:35","SaveCount":49,"SaveAddress":"F5000"}}
18:37:28.293 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS2 = {"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021.10.19 08:23:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"552/699"}}
18:37:28.297 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT-LOTZ",""],"TelePeriod":30,"Resolution":"558180F0","SetOption":["0000A029","0505C80001000600283C5A0A000000000000","00000280","00006000","00006000"]}}
18:37:28.308 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS4 = {"StatusMEM":{"ProgramSize":613,"Free":388,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:37:28.316 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS5 = {"StatusNET":{"Hostname":"Geraet-EG-Hauswirtschaftsraum-R","IPAddress":"192.168.175.100","Gateway":"192.168.175.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.175.1","DNSServer2":"0.0.0.0","Mac":"68:C6:3A:FA:EF:93","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:37:28.323 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.175.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FAEF93","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":5,"SOCKET_TIMEOUT":4}}
18:37:28.329 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS7 = {"StatusTIM":{"UTC":"2021-11-28T17:37:28","Local":"2021-11-28T18:37:28","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:55","Sunset":"16:22"}}
18:37:28.333 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:37:28.341 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS10 = {"StatusSNS":{"Time":"2021-11-28T18:37:28.335","Switch1":"ON","ANALOG":{"Temperature":56.0},"ENERGY":{"TotalStartTime":"2021-11-28T00:00:00","Total":895.124,"Yesterday":2.540,"Today":2.604,"Power":160},"TempUnit":"C"}}
18:37:28.346 MQT: stat/Geraet_EG_Hauswirtschaftsraum_Router/STATUS11 = {"StatusSTS":{"Time":"2021-11-28T18:37:28.343","Uptime":"0T00:16:56","UptimeSec":1016,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IOT-LOTZ","BSSId":"E2:63:DA:23:3C:3E","Channel":9,"Mode":"11g","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Replace the device
2. Restore backup (no energy data in the backup)
3. Manually restore energy values via energytotal, -today ...
4. Try to set the StartTotalTime for EnergyTotal --> not possible

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
EnergyStartTotalTime [epoch]
or
EnergyTotal [epoch]
to set the StartTotalTime

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
It is the first time that I need that. 
First I was a bit surprised that the energy values are not restored by the backup. 
But on second thought it makes sense -> you don't want that values to be overridden if you normally restore a backup.
It would be great if you could select that you are replacing a device. Than more data could be restored (must be saved before). 
But that would be a feature request...

**(Please, remember to close the issue when the problem has been addressed)**
OK

sfromis, 2021-11-28:
The commands to reset the energy values also take a 2nd parameter for start time.

ascillato2, 2021-11-28:
@Bascht74

Yes, in the docs is not clear that the explanations for the former `EnergyReset` command also applies to the new ones.

Please, help us and do those changes in the docs. Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-11-28 by ascillato2

Bascht74, 2021-11-28:
Done: https://github.com/tasmota/docs/pull/866