arendst/Tasmota #13862: Can't set MQTT Topic to Hostname

jeroenvermeulen, 2021-11-29:
### PROBLEM DESCRIPTION
I can set the MQTT Topic to anything but not to the hostname. This happens both in the web interface and in console.
When I use the hostname it applies `tasmota_%06X` as topic.
I have experienced this problem on many devices running 10.0.0.0, but I also repeated it on a fresh build on today's development branch. 
On rare occasions on some devices I was able to set the topic to the hostname, I don't know when this occurs.
On most devices I am able to use `FullTopic %prefix%/__hostname__/` as a workaroud but not on the `zbbridge` because then `ZbDeviceTopic 1` no longer works.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Zigbee Bridge
- [x] Tasmota binary firmware version number used: 10.0.0.3
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: upload via webif
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
21:42:33.125 CMD: Backlog Template; Module; GPIO 255
21:42:33.126 SRC: WebConsole from 192.168.1.227
21:42:33.128 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
21:42:33.139 SRC: Backlog
21:42:33.140 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
21:42:33.146 MQT: stat/tasmota_E2FABC/RESULT = {"NAME":"Sonoff ZBBridge","GPIO":[320,3552,0,3584,5312,0,0,0,640,576,608,0,32,0],"FLAG":0,"BASE":75}
21:42:33.371 SRC: Backlog
21:42:33.373 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
21:42:33.377 MQT: stat/tasmota_E2FABC/RESULT = {"Module":{"0":"Sonoff ZBBridge"}}
21:42:33.623 SRC: Backlog
21:42:33.625 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
21:42:33.635 MQT: stat/tasmota_E2FABC/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
21:42:34.023 WIF: Checking connection...
21:42:39.392 MQT: tele/tasmota_E2FABC/STATE = {"Time":"2021-11-29T21:42:39","Uptime":"0T00:17:09","UptimeSec":1029,"Vcc":3.468,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"GRS-IOT","BSSId":"F6:92:BF:19:EB:F4","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
not used
```
- [x] Provide the output of this command: `Status 0`:
```lua
21:43:53.820 CMD: Status 0
21:43:53.821 SRC: WebConsole from 192.168.1.227
21:43:53.822 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
21:43:53.827 MQT: stat/tasmota_E2FABC/STATUS = {"Status":{"Module":0,"DeviceName":"zbbridge","FriendlyName":["zbbridge"],"Topic":"tasmota_E2FABC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:43:53.833 MQT: stat/tasmota_E2FABC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:18:23","StartupUTC":"2021-11-29T20:25:30","Sleep":50,"CfgHolder":4617,"BootCount":813,"BCResetTime":"2020-10-20T22:31:31","SaveCount":1074,"SaveAddress":"1FB000"}}
21:43:53.838 MQT: stat/tasmota_E2FABC/STATUS2 = {"StatusFWR":{"Version":"10.0.0.3(zbbridge)","BuildDateTime":"2021-11-29T21:21:45","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"397/699"}}
21:43:53.843 MQT: stat/tasmota_E2FABC/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GRS-IOT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008209","2805C80001000600003C5A00000000000000","000000A0","00046080","00004000"]}}
21:43:53.852 MQT: stat/tasmota_E2FABC/STATUS4 = {"StatusMEM":{"ProgramSize":671,"Free":1120,"Heap":21,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000003","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50","Sensors":"1"}}
21:43:53.858 MQT: stat/tasmota_E2FABC/STATUS5 = {"StatusNET":{"Hostname":"zbbridge","IPAddress":"192.168.1.109","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"98:F4:AB:E2:FA:BC","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
21:43:53.863 MQT: stat/tasmota_E2FABC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"zbbridge","MqttClient":"zbbridge","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:43:53.869 MQT: stat/tasmota_E2FABC/STATUS7 = {"StatusTIM":{"UTC":"2021-11-29T20:43:53","Local":"2021-11-29T21:43:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:20","Sunset":"16:56"}}
21:43:53.876 MQT: stat/tasmota_E2FABC/STATUS10 = {"StatusSNS":{"Time":"2021-11-29T21:43:53"}}
21:43:53.884 MQT: stat/tasmota_E2FABC/STATUS11 = {"StatusSTS":{"Time":"2021-11-29T21:43:53","Uptime":"0T00:18:23","UptimeSec":1103,"Vcc":3.458,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"GRS-IOT","BSSId":"F6:92:BF:19:EB:F4","Channel":6,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
21:44:59.528 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '4'
21:44:59.531 MQT: stat/tasmota_E2FABC/RESULT = {"WebLog":4}
21:45:13.502 CMD: Topic zbbridge
21:45:13.504 SRC: WebConsole from 192.168.1.227
21:45:13.505 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 8, Data 'zbbridge'
21:45:13.509 MQT: stat/tasmota_E2FABC/RESULT = {"Topic":"tasmota_%06X"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Get the current hostname using `Hostname` in console
- Try to set the MQTT Topic to exactly the same value

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- I expect the hostname to be applied as Zigbee topic

### SCREENSHOTS
The device is called `zbbridge`, in console I enter `Topic zbbridge`, response in console:
```
21:26:30.760 CMD: Topic zbbridge
21:26:30.761 SRC: WebConsole from 192.168.1.227
21:26:30.762 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 8, Data 'zbbridge'
21:26:30.766 MQT: stat/tasmota_E2FABC/RESULT = {"Topic":"tasmota_%06X"}
```

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-11-29:
There is no problem to set hostname and topic identical.
However, as [written in the docs](https://tasmota.github.io/docs/Commands/#topic), it is forbidden to set Topic and **MqttClient** the same.
It is because your MqttClient is also equal to "zbbridge" that Topic is reset to default, not because of hostname.

jeroenvermeulen, 2021-11-29:
Thanks. I was already working to update the topic since I discovered the MqttClient value was the problem.

closed 2021-11-29 by ascillato2