arendst/Tasmota #13863: Unable to disable MQTT TLS when build with option USE_MQTT_TLS

sander1988, 2021-11-29:
### PROBLEM DESCRIPTION
I'm unable to disable MQTT TLS when build with option USE_MQTT_TLS. Disabling it temporary is sometimes needed to investigate other issues.

I tried it using the web interface checkbox and SetOption103 OFF/0 . The device restarts, but the setting is not applied (it stays enabled).

The test environment is exact the same as #13809. Please use that issue for more details on the settings and build options used.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic and Shelly 1
- [x] Tasmota binary firmware version number used: 9.3.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Web upload
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. Build the firmware with USE_MQTT_TLS.
2. Try to disable MQTT TLS with 'SetOption103 OFF'
3. Device reboots to apply setting.
4. SetOption103 is still ON

### EXPECTED BEHAVIOUR
SetOption103 (and the TLS checkbox) allows you to toggle MQTT TLS off (and on).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I also tried it with a build **without** TLS (and without the option USE_MQTT_TLS). In that case the checkbox and SetOption103 both work.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-11-29:
Did you change the MqttPort to 1883 or any suitable value on which you are sure that your broker will respond without TLS ?
If your broker immediately engage a TLS negociation, that could be the reason.

sander1988, 2021-11-29:
> Did you change the MqttPort to 1883 or any suitable value on which you are sure that your broker will respond without TLS ? If your broker immediately engage a TLS negociation, that could be the reason.

That's it. I see it in the source code now. I was using port 8884; in that case it always turns TLS on.

ascillato, 2021-11-29:
Besides there is a bug in the UI for TLS in your version 9.3.0. That is quite old. Updating will solve it.

closed 2021-11-29 by ascillato2