arendst/Tasmota #13882: ESP32-webcam 10.0.0.3 dev restarts or resets on power toggle after CAM: Initialized

nikvostro, 2021-12-01:
### PROBLEM DESCRIPTION
ESP32-webcam 10.0.0.3 dev restarts or even resets on power toggle after CAM: Initialized. Tried different power sources.
Issue doesn't appear for FW 10.0.0.0 and earlier for the same hardware.

### REQUESTED INFORMATION
- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues):
Similar case [Tasmota32-webcam 10.0.0.3 reboots after any relay/pwn on/off #13690]
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Ai-Thinker (ESP32-CAM)
- [ ] Tasmota binary firmware version number used: DEV 10.0.0.3
  - [ ] Pre-compiled: 
  - [ ] Self-compiled: the same behaviour for my own build
- [ ] Flashing tools used: VSCode/Platformio for sel-compiled + Tasmota web-installer for Pre-compiled
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

**Before switching to Module ESP32CAM:**
```lua
13:08:56.392 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:08:56.663 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:08:56.869 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

**After switching to Module ESP32CAM:**
```lua
13:44:06.983 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:44:07.231 RSL: RESULT = {"Module":{"2":"ESP32-Cam"}}
13:44:07.489 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"0":"None"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
~~- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:~~

- [ ] Provide the output of this command: `Status 0`:
**Before  switching to Module ESP32CAM:**
```lua
13:36:41.908 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_7882E8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:36:41.941 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Timer Group1 Watch dog reset digital core","Uptime":"0T00:09:54","StartupUTC":"2021-12-01T12:26:47","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2021-12-01T09:37:12","SaveCount":10}}
13:36:41.974 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0.3(webcam)","BuildDateTime":"2021-11-30T09:11:44","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":240,"Hardware":"ESP32-D0WD","CR":"386/699"}}
13:36:41.998 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domuk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
13:36:42.024 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1036,"Free":1856,"Heap":173,"PsrMax":4093,"PsrFree":4083,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","0F90058A","00040001","00000082","00000000","40000981","80000080","00200000","00000000"],"Drivers":"1,2,4,7,8,9,10,12,24,27,29,38,50,81","Sensors":"2,6,127"}}
13:36:42.060 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-7882E8-0744","IPAddress":"192.168.11.89","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.11.1","DNSServer2":"0.0.0.0","Mac":"40:F5:20:78:82:E8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:36:42.083 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7882E8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:36:42.107 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-01T12:36:42","Local":"2021-12-01T13:36:42","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:22","Sunset":"16:55"}}
13:36:42.120 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-01T13:36:42","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
13:36:42.135 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-01T13:36:42","Uptime":"0T00:09:55","UptimeSec":595,"Heap":173,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"domuk","BSSId":"F0:79:59:36:57:40","Channel":11,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:04"}}}
13:36:42.157 RSL: STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"IllegalInstruction","EPC":"400803c0","EXCVADDR":"00000000","CallChain":["400803bd"]}}
```

**After switching to ESP32CAM:**
```lua
13:42:00.180 CMD: status 0
13:42:00.183 RSL: STATUS = {"Status":{"Module":2,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_7882E8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:42:00.216 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:22","StartupUTC":"2021-12-01T12:41:38","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-12-01T09:37:12","SaveCount":13}}
13:42:00.248 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0.3(webcam)","BuildDateTime":"2021-11-30T09:11:44","Core":"2_0_1_1","SDK":"v4.4-dev-3728-g7cac8278e","CpuFrequency":240,"Hardware":"ESP32-D0WD","CR":"386/699"}}
13:42:00.260 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domuk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00006000","00004000"]}}
13:42:00.295 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1036,"Free":1856,"Heap":173,"PsrMax":4093,"PsrFree":4085,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","0F90058A","00040001","00000082","00000000","40000981","80000080","00200000","00000000"],"Drivers":"1,2,4,7,8,9,10,12,24,27,29,38,50,81","Sensors":"2,6,127"}}
[... cut ...]
13:48:31.946 RSL: SENSOR = {"Time":"2021-12-01T13:48:31","ESP32":{"Temperature":41.1},"TempUnit":"C"}
```

### TO REPRODUCE
1. Upload dev 10.0.0.3 tasmota32-webcam.bin
2. Switch module to 2 (ESP32CAM)
3. Open web GUI and move PWM (flash brightness) slider

### EXPECTED BEHAVIOUR
No reset/reboot

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

nikvostro, 2021-12-01:
Sample logs:
[20211201135357.log](https://github.com/arendst/Tasmota/files/7633988/20211201135357.log)
[20211201150622.log](https://github.com/arendst/Tasmota/files/7633989/20211201150622.log)
[20211201152432.log](https://github.com/arendst/Tasmota/files/7633991/20211201152432.log)

ascillato, 2021-12-01:
Just for testing. Please, change the GPIO from pwm to relay and try again.

nikvostro, 2021-12-01:
After reporting this issue I've obtained that 10.0.0.4 DEV has been released. Flashed it. Situation remained the same. 
Now I'm even trying `seriallog 4` (after cam init) and getting restart.
[20211201162334.log](https://github.com/arendst/Tasmota/files/7634229/20211201162334.log)
Switched gpio 4 to relay --> the same. Behaves very unstable probably after cam init.
See attached log. [20211201162334.log](https://github.com/arendst/Tasmota/files/7634229/20211201162334.log)

jaschkopf, 2021-12-06:
I can confirm the issue. Using a ESP32-Cam Module. When changing the brightness or toggel the state, the esp restarts. I'm using Tasmota 10.0.0 flashed via Chrome web flasher

cgomesu, 2021-12-07:
I can replicate this issue on `Tasmota Version 10.0.0.4(webcam)-2_0_1_1(2021-12-06T14:17:10)` as well:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.003 HDW: ESP32-D0WD (PSRAM)
00:00:00.027 UFS: FlashFS mounted with 304 kB free
00:00:00.066 CFG: Loaded from File, Count 46
00:00:00.068 CFG: CR 367/699, Busy 0
00:00:00.071 SPI: Hardware using GPIO14(CLK), GPIO15(MOSI) and GPIO02(MISO)
00:00:00.073 ROT: Mode 1
00:00:00.684 SRC: Restart
00:00:00.686 Project tasmota - Tasmota Version 10.0.0.4(webcam)-2_0_1_1(2021-12-06T14:17:10)
...
13:27:32.439 CAM: Stream init
13:27:32.441 CAM: User template
13:27:32.441 CAM: PSRAM found
13:27:32.800 CAM: Initialized
...
13:28:04.145 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '55'
13:28:04.146 SRC: Light
13:28:04.149 RSL: RESULT = {"POWER":"ON","Dimmer":55}
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.003 HDW: ESP32-D0WD (PSRAM)
00:00:00.008 UFS: FlashFS mounted with 308 kB free
00:00:00.023 CFG: Use defaults
00:00:00.184 QPC: Count 1
00:00:00.190 Project tasmota - Tasmota Version 10.0.0.4(webcam)-2_0_1_1(2021-12-06T14:17:10)
00:00:00.569 WIF: WifiManager active for 3 minutes
...
```

arendst, 2021-12-09:
Unable to reproduce:

```
00:00:00.003 HDW: ESP32-D0WDQ6 (PSRAM)
00:00:00.003 HDW: FoundPSRAM=1 CanUsePSRAM=1
00:00:00.020 UFS: FlashFS mounted with 48 kB free
00:00:00.047 CFG: Loaded from File, Count 578
00:00:00.055 QPC: Count 1
00:00:00.059 ROT: Mode 1
00:00:00.061 SRC: Restart
00:00:00.063 Project tasmota - Esp32Cam2 Version 10.1.0(webcam)-2_0_1_1(2021-12-08T14:50:31)
00:00:00.214 WIF: Checking connection...
00:00:00.214 WIF: Attempting connection...
00:00:01.268 WIF: Network (re)scan started...
00:00:01.268 WIF: Checking connection...
00:00:01.269 WIF: Attempting connection...
00:00:02.158 WIF: Checking connection...
00:00:02.158 WIF: Attempting connection...
00:00:03.158 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -48, Encryption 1
00:00:03.158 WIF: Network 1, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -87, Encryption 1
00:00:03.538 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as esp32cam2...
00:00:03.538 WIF: Checking connection...
00:00:03.539 WIF: Attempting connection...
00:00:04.406 WIF: Checking connection...
00:00:04.407 WIF: Attempting connection...
00:00:05.406 WIF: Checking connection...
00:00:05.407 WIF: Attempting connection...
00:00:06.403 WIF: Checking connection...
00:00:06.404 WIF: Connected
00:00:06.658 HTP: Web server active on esp32cam2 with IP address 192.168.2.113
00:00:06.912 QPC: Reset
00:00:06.991 NTP: Synch time...
00:00:07.014 RTC: UTC 2021-12-09T13:13:15, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
14:13:15.001 NTP: Synched
14:13:15.643 MQT: Attempting connection...
14:13:15.667 MQT: Connected
14:13:15.676 MQT: tele/esp32cam2/LWT = Online (retained)
14:13:15.687 MQT: cmnd/esp32cam2/POWER = 
14:13:15.688 MQT: Subscribe to cmnd/esp32cam2/#
14:13:15.692 MQT: Subscribe to cmnd/tasmotas/#
14:13:15.699 MQT: Subscribe to cmnd/DVES_1453B0_fb/#
14:13:15.712 MQT: tele/esp32cam2/INFO1 = {"Info1":{"Module":"AITHINKER CAM No SPI","Version":"10.1.0(webcam)","FallbackTopic":"cmnd/DVES_1453B0_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:13:15.728 MQT: tele/esp32cam2/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"esp32cam2","IPAddress":"192.168.2.113"}}
14:13:15.744 MQT: tele/esp32cam2/INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
14:13:15.753 MQT: stat/esp32cam2/RESULT = {"POWER":"ON"}
14:13:15.762 MQT: stat/esp32cam2/POWER = ON
14:13:15.768 TFS: File 'autoexec.bat' not found
14:13:16.910 APP: Boot Count 168
14:13:17.390 CFG: Saved, Count 579, Bytes 4096
14:13:19.916 MQT: tele/esp32cam2/STATE = {"Time":"2021-12-09T14:13:19.905","Uptime":"0T00:00:12","UptimeSec":12,"Heap":169,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":9,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:07"}}
14:13:19.955 MQT: tele/esp32cam2/SENSOR = {"Time":"2021-12-09T14:13:19.948","ESP32":{"Temperature":43.9},"TempUnit":"C"}
14:13:24.961 MQT: tasmota/discovery/246F281453B0/config = {"ip":"192.168.2.113","dn":"Esp32Cam2","fn":["Esp32Cam2",null,null,null,null,null,null,null],"hn":"esp32cam2","mac":"246F281453B0","md":"AITHINKER CAM No SPI","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"10.1.0","t":"esp32cam2","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":1,"sho":[0,0,0,0],"ver":1} (retained)
14:13:24.971 MQT: tasmota/discovery/246F281453B0/sensors = {"sn":{"Time":"2021-12-09T14:13:24.963","ESP32":{"Temperature":43.3},"TempUnit":"C"},"ver":1} (retained)
14:13:30.353 CMD: dimmer 25
14:13:30.354 SRC: WebConsole from 192.168.2.1
14:13:30.355 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '25'
14:13:30.364 MQT: stat/esp32cam2/RESULT = {"POWER":"ON","Dimmer":25}
14:13:30.558 CFG: Saved, Count 580, Bytes 4096
14:13:34.659 WIF: Checking connection...
14:13:38.731 CMD: dimmer 60
14:13:38.732 SRC: WebConsole from 192.168.2.1
14:13:38.734 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '60'
14:13:38.742 MQT: stat/esp32cam2/RESULT = {"POWER":"ON","Dimmer":60}
14:13:39.636 CFG: Saved, Count 581, Bytes 4096
14:13:44.154 CMD: dimmer 55
14:13:44.155 SRC: WebConsole from 192.168.2.1
14:13:44.157 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '55'
14:13:44.165 MQT: stat/esp32cam2/RESULT = {"POWER":"ON","Dimmer":55}
14:13:44.567 CFG: Saved, Count 582, Bytes 4096
14:13:50.191 CMD: dimmer 10
14:13:50.193 SRC: WebConsole from 192.168.2.1
14:13:50.194 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '10'
14:13:50.201 MQT: stat/esp32cam2/RESULT = {"POWER":"ON","Dimmer":9}
14:13:50.794 CFG: Saved, Count 583, Bytes 4096
```
Notice difference in ESP32 hardware (D0WDQ6) and partition layout.

nikvostro, 2021-12-09:
@arendst 
> Notice difference in ESP32 hardware (D0WDQ6) and partition layout.

Not sure what you did mean here.
I'm using the same settings as for standard env:tasmota32-webcam + some defs/undefs

[image]

arendst, 2021-12-09:
No idea where the -nav version comes from but it's not from the Tasmota repository.

This is what we support:

```
[env:tasmota32-webcam]
extends                 = env:tasmota32_base
board                   = esp32-cam
build_flags             = ${env:tasmota32_base.build_flags} -DFIRMWARE_WEBCAM
lib_extra_dirs          = lib/libesp32
```

nikvostro, 2021-12-09:
> No idea where the -nav version comes from but it's not from the Tasmota repository.

Sorry for confusing you. It's in _cenv.ini and _NAV is my initials to selected features (def/undef) in _ovveride.h
Basically, I declared this issue for the standard pre-compiled fw and for mine custom as well.

Now uploaded v10.1.0, `reset 1` --> the same behavior :(
Attaching log: [20211209173740.log](https://github.com/arendst/Tasmota/files/7685917/20211209173740.log)

arendst, 2021-12-09:
Thx for the clarification. In your latest log the same error occurs indeed:

``TG1WDT_SYS_RESET``

This is a watchdog error most likely caused by the underlying idf/arduino core code. Core has been changed between 9.5 and 10.0 (and again in 10.1). That makes debugging a bit of an issue.

I'll see where to start.

BTW. My cam does also do the TG1WDT_SYS_RESET but not that often as yours.

nikvostro, 2021-12-09:
One more attempt from the scratch.
1. Erasing (`esptool.py --chip esp32 -p COM13 -b 1500000 erase_flash`)
2. Uploading FW via web installer.
[image]
3. Switch module to ESP32-Cam
4. Toggle `power`

Magic (now it's reset to defaults) happens right after `16:06:33.087 CAM: Initialized`
```lua
Terminal log file
Date: 09.12.2021 - 18:05:43
-----------------------------------------------
16:05:53.923 CMD: restart 1
16:05:53.925 RSL: RESULT = {"Restart":"Restarting"}
16:05:55.118 APP: Restarting
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.002 HDW: ESP32-D0WD (PSRAM)
00:00:00.019 UFS: FlashFS mounted with 304 kB free
00:00:00.046 CFG: Loaded from File, Count 15
00:00:00.053 QPC: Count 1
00:00:00.057 SPI: Hardware using GPIO14(CLK), GPIO15(MOSI) and GPIO02(MISO)
00:00:00.671 Project tasmota - Tasmota Version 10.1.0(webcam)-2_0_1_1(2021-12-08T14:50:31)
00:00:01.307 WIF: Connecting to AP1 domuk Channel 6 BSSId F0:79:59:36:57:40 in mode 11n as tasmota-7882E8-0744...
00:00:06.119 WIF: Connected
00:00:06.373 HTP: Web server active on tasmota-7882E8-0744 with IP address 192.168.11.89
00:00:06.926 QPC: Reset
16:06:04.372 RSL: INFO1 = {"Info1":{"Module":"ESP32-Cam","Version":"10.1.0(webcam)","FallbackTopic":"cmnd/DVES_7882E8_fb/","GroupTopic":"cmnd/tasmotas/"}}
16:06:04.384 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-7882E8-0744","IPAddress":"192.168.11.89"}}
16:06:04.395 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
16:06:04.396 RSL: RESULT = {"POWER":"ON"}
16:06:04.397 RSL: POWER = ON
16:06:08.924 RSL: STATE = {"Time":"2021-12-09T16:06:08","Uptime":"0T00:00:13","UptimeSec":13,"Heap":174,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":9,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"domuk","BSSId":"F0:79:59:36:57:40","Channel":6,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:07"}}
16:06:08.958 RSL: SENSOR = {"Time":"2021-12-09T16:06:08","ESP32":{"Temperature":42.8},"TempUnit":"C"}
16:06:09.408 CMD: power 2
16:06:09.411 RSL: RESULT = {"POWER":"OFF"}
16:06:09.412 RSL: POWER = OFF
16:06:10.662 CMD: power 2
16:06:10.665 RSL: RESULT = {"POWER":"ON"}
16:06:10.666 RSL: POWER = ON
16:06:11.310 CMD: power 2
16:06:11.313 RSL: RESULT = {"POWER":"OFF"}
16:06:11.314 RSL: POWER = OFF
16:06:11.692 CMD: power 2
16:06:11.694 RSL: RESULT = {"POWER":"ON"}
16:06:11.695 RSL: POWER = ON
16:06:15.351 CMD: dimmer 10
16:06:15.354 RSL: RESULT = {"POWER":"ON","Dimmer":9}
16:06:16.829 CMD: dimmer 50
16:06:16.832 RSL: RESULT = {"POWER":"ON","Dimmer":50}
16:06:18.869 CMD: dimmer 90
16:06:18.872 RSL: RESULT = {"POWER":"ON","Dimmer":91}
16:06:20.765 CMD: dimmer 0
16:06:20.769 RSL: RESULT = {"POWER":"OFF","Dimmer":0}
16:06:33.087 CAM: Initialized
16:07:02.086 CMD: power 2
16:07:02.089 RSL: RESULT = {"POWER":"ON"}
16:07:02.090 RSL: POWER = ON
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.002 HDW: ESP32-D0WD (PSRAM)
00:00:00.009 UFS: FlashFS mounted with 308 kB free
00:00:00.041 CFG: Use defaults
00:00:00.192 QPC: Count 1
00:00:00.198 Project tasmota - Tasmota Version 10.1.0(webcam)-2_0_1_1(2021-12-08T14:50:31)
00:00:00.397 WIF: WifiManager active for 3 minutes
00:00:00.138 HTP: Web server active on tasmota-7882E8-0744 with IP address 192.168.4.1

-----------------------------------------------
Date: 09.12.2021 - 18:07:00
End log file
```

arendst, 2021-12-09:
Yes mine did a reset to defaults too after a total re-install and starting the stream followed by the TG1WDT_SYS_RESET.

So for now, to use the webcam, stay at 9.5 until we manage to solve the webcam issue.

arendst, 2021-12-10:
As an alternative you might want to use v10.1.0 BUT set ``savedata 0`` tot stop saving changes to the settings into NVS and/or filesystem. My (short) tests show the webcam keeps running without WDT's.

I suspect the NVS writes are causing the timeouts due to unavailable space in the NVS area. I noticed the webcam library tries to save it's settings into NVS too which is already very tight.

I will investigate this further in the next days.

nikvostro, 2021-12-10:
@arendst I'm using webcam as a smart device with a scripting and not saving configuration is not the case for me :)
But for others who are using only streaming or other fixed (non-configurable) features maybe OK indeed.

Jason2866, 2021-12-10:
@arendst  Can confirm that `savedata 0` stops TG1WDT_SYS_RESET and webcam runs stable.

ascillato2, 2021-12-12:
Hi @nikvostro

Please, try latest Tasmota from the development branch. Jason and Theo have investigated this and Theo has pushed a fix. Thanks

cgomesu, 2021-12-13:
I can confirm that the issue is fixed on `Tasmota Version 10.1.0.1(webcam)-2_0_1_1(2021-12-12T20:09:47)`:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

00:00:00.002 HDW: ESP32-D0WD (PSRAM)
00:00:00.029 UFS: FlashFS mounted with 304 kB free
00:00:00.071 CFG: Loaded from File, Count 61
00:00:00.079 QPC: Count 1
00:00:00.081 CFG: CR 377/699, Busy 0
00:00:00.083 SPI: Hardware using GPIO14(CLK), GPIO15(MOSI) and GPIO02(MISO)
00:00:00.085 ROT: Mode 1
00:00:00.696 SRC: Restart
00:00:00.698 Project tasmota - TasmotaCam02 Version 10.1.0.1(webcam)-2_0_1_1(2021-12-12T20:09:47)
...
14:12:30.789 CAM: Initialized
...
14:13:41.998 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '22'
14:13:42.001 SRC: Light
14:13:42.003 RSL: RESULT = {"POWER":"ON","Dimmer":22}
14:13:42.970 CFG: Saved, Count 63, Bytes 4096
14:13:47.644 SRC: WebGui from 192.168.10.30
14:13:47.645 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '48'
14:13:47.648 RSL: RESULT = {"POWER":"ON","Dimmer":48}
14:13:47.938 CFG: Saved, Count 64, Bytes 4096
14:13:50.013 WIF: Checking connection...
14:13:55.597 SRC: WebGui from 192.168.10.30
14:13:55.598 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '78'
14:13:55.601 RSL: RESULT = {"POWER":"ON","Dimmer":78}
14:13:55.966 CFG: Saved, Count 65, Bytes 4096
14:14:05.504 SRC: WebGui from 192.168.10.30
14:14:05.505 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '19'
14:14:05.508 RSL: RESULT = {"POWER":"ON","Dimmer":19}
14:14:05.965 CFG: Saved, Count 66, Bytes 4096
14:14:10.033 WIF: Checking connection...
14:14:11.572 SRC: WebGui from 192.168.10.30
14:14:11.574 RSL: RESULT = {"POWER":"OFF"}
14:14:11.576 RSL: POWER = OFF
14:14:11.958 CFG: Saved, Count 67, Bytes 4096
```

Jason2866, 2021-12-13:
Not completely fixed. When using higher resolutions it still stalls.
Theo (arendst) is on it. No easy fix!

nikvostro, 2021-12-13:
Just fetched the dev. Now I can't compile by own build :(

```
...
|-- <Wire> 2.0.0
|-- <XPT2046_Touchscreen>
|   |-- <SPI> 2.0.0
Building in release mode
*** use provided user_config_override.h as planned ***
*** use provided platformio_override.ini as planned ***
*** use provided platformio_tasmota_cenv.ini as planned ***
Use manually specified: COM13
Compiling .pio\build\tasmota32-webcam_nav\src\homekit.c.o
xtensa-esp32-elf-gcc: error: CreateProcess: No such file or directory
Compiling .pio\build\tasmota32-webcam_nav\src\tasmota.ino.cpp.o
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\tasmota32-webcam_nav\lib383\LVGL8\core\lv_event.c.o
*** [.pio\build\tasmota32-webcam_nav\src\homekit.c.o] Error 1
Compiling .pio\build\tasmota32-webcam_nav\lib383\LVGL8\core\lv_group.c.o
*** [.pio\build\tasmota32-webcam_nav\src\tasmota.ino.cpp.o] Error 1
======================================================== [FAILED] Took 27.63 seconds ========================================================

Environment           Status    Duration
--------------------  --------  ------------
tasmota32-webcam_nav  FAILED    00:00:27.630
=================================================== 1 failed, 0 succeeded in 00:00:27.630 ===================================================
The terminal process "C:\Users\nikvostro\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'tasmota32-webcam_nav'" terminated with exit code: 1.
```

My cenv isn't far from the standard env32:
```lua
[env:tasmota32-webcam_nav]
extends                 = env:tasmota32_base
board                   = esp32-cam
build_flags             = ${env:tasmota32_base.build_flags} -DFIRMWARE_WEBCAM_NAV
board_build.f_cpu       = 240000000L
upload_port             = COM13
upload_speed            = 1500000 ; NAV: in case of issues try 2000000 -> 1500000 -> 921600 -> 460800
lib_extra_dirs          = ${env:tasmota32_base.lib_extra_dirs}
```
User override for FIRMWARE_WEBCAM_NAV:
[user_config_override.txt](https://github.com/arendst/Tasmota/files/7705258/user_config_override.txt)

Other files are not changed.

Jason2866, 2021-12-13:
@nikvostro  For the webcam you have enabled a lot. BLE is definitely a no go here!

cgomesu, 2021-12-13:
> Not completely fixed. When using higher resolutions it still stalls. Theo (arendst) is on it. No easy fix!

i can confirm this as well.  it seems that any resolution above 9 now causes a `TG1WDT_SYS_RESET` but anything equal or below 9 is working as intended.

- firmware version:  `10.1.0.1(webcam)-2_0_1_1(2021-12-12T20:09:47)`
- hw: `ESP32-D0WDQ6`
- related log: [tasmota32-webcam-wcresolution-tests.log](https://github.com/arendst/Tasmota/files/7705600/tasmota32-webcam-wcresolution-tests.log)

arendst, 2021-12-17:
Should be fixed now.

Jason2866, 2021-12-18:
It is fixed :-)

closed 2021-12-18 by Jason2866