arendst/Tasmota #13884: Home Assistant discovery is not putting the energy sensors in the right device_class and state_class

bigcookie, 2021-12-01:
### PROBLEM DESCRIPTION
When using SetOption19=1 and thus MQTT discovery in Home Assistant, the energy sensors like Sonoff Pow R2 related energy sensors are marked as device_class=power and no state_class. To use them in the new energy dashboard, the sensors need to be attributed correctly as device_class=energy and state_class=total or total_increasing or measurement.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [Sonoff Pow R2] Device used (e.g., Sonoff Basic): _____
- [10.0.0] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: webbased installer from tasmota project
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:15:53.428 CMD: Backlog Template; Module; GPIO 255
16:15:53.478 MQT: powermeter2/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:15:53.679 MQT: powermeter2/stat/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
16:15:53.887 MQT: powermeter2/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

- [x] Provide the output of this command: `Status 0`:
```
16:16:42.614 CMD: status0
16:16:42.646 MQT: powermeter2/stat/STATUS0 = {"Status":{"Module":43,"DeviceName":"SH-Powermeter2","FriendlyName":["SH-Powermeter2"],"Topic":"powermeter2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:45","StartupUTC":"2021-12-01T15:06:57","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-12-01T01:59:26","SaveCount":45,"SaveAddress":"F7000"},"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021.10.19 08:23:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"397/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndroLAN",""],"TelePeriod":300,"Resolution":"558380C0","SetOption":["00028009","2805C8000100068000005A0A000000000000","00000280","00006000","00004000"]},"StatusMEM":{"ProgramSize":613,"Free":388,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":2,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"SH-Powermeter2","IPAddress":"192.168.20.242","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.20.1","DNSServer2":"0.0.0.0","Mac":"80:7D:3A:68:23:BF","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.20.253","MqttPort":1883,"MqttClientMask":"SH-Powermeter2","MqttClient":"SH-Powermeter2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-12-01T15:16:42","Local":"2021-12-01T16:16:42","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:22","Sunset":"16:55"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2021-12-01T16:16:42","ENERGY":{"TotalStartTime":"2021-12-01T02:29:03","Total":0.004,"Yesterday":0.000,"Today":0.004,"Power": 1,"ApparentPower":17,"ReactivePower":17,"Factor":0.09,"Voltage":234.8,"Current":0.073}},"StatusSTS":{"Time":"2021-12-01T16:16:42","Uptime":"0T00:09:45","UptimeSec":585,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"AndroLAN","BSSId":"7C:FF:4D:FF:AF:2C","Channel":13,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Enable MQTT discovery in HA, SetOption19=1, check entities and device_class for energy sensors.

### EXPECTED BEHAVIOUR
Expect that energy sensors get teh following attributes:
device_class=energy (not power as today)
state_class=total or total_increasing or measurement - depending on sensor

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-12-01:
The PR https://github.com/arendst/Tasmota/pull/13053 that intends to fix this has been already submitted but it is under review before merging it. I couldn't have the time to check it.

arendst, 2021-12-01:
Just merged it.

arendst, 2021-12-02:
Should be fixed now.

bigcookie, 2021-12-02:
Hi, checking the PR, it seems to cover the device_class only. That would not be sufficient to my understanding as state_class also need to be set...
At least I manually fixed this customizing device_class only and it was not sufficient... For the total energy sensor, state_class "total_increasing" would be required...

I will check the fix above as well, but can only do this over the weekend.