arendst/Tasmota #13885: Tasmota-minimal too big for Sonoff POW R2 or Shelly 1 reflashing Tasmota 10.0 via OTA

bigcookie, 2021-12-01:
### PROBLEM DESCRIPTION
With Tasmota 10.0 I had one issue with one device and wanted to reflash. Device feedback was "not enough memory" when trying to flash tasmota-minimal.
See status from Sonoff (similar for Shelly 1)
[image]

According to ota.tasmota.com, in comparison for 9.5.0 - 
- tasmota-DE.bin increased by 9kb (609kb to 618kb)
-  tasmota-minimal.bin increased by 5kb (390kb to 395kb)

According to device information (see screenshot)
program takes 613kb
free flash is 388kb

Is there any plan to reduce size of tasmota-minimal.bin further (is this possible?).

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW R2, Shelly 1
- [x] Tasmota binary firmware version number used: 10.0.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
18:33:35.604 MQT: powermeter1/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
18:33:35.805 MQT: powermeter1/stat/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
18:33:36.011 MQT: powermeter1/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
18:34:06.236 MQT: powermeter1/stat/STATUS0 = {"Status":{"Module":43,"DeviceName":"SH-Waschmaschine","FriendlyName":["SH-Waschmaschine"],"Topic":"powermeter1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:56:13","StartupUTC":"2021-12-01T16:37:53","Sleep":50,"CfgHolder":4617,"BootCount":46,"BCResetTime":"2020-04-24T11:55:23","SaveCount":1511,"SaveAddress":"F5000"},"StatusFWR":{"Version":"10.0.0(tasmota)","BuildDateTime":"2021.10.19 08:23:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"489/699"},"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndroLAN","a17"],"TelePeriod":300,"Resolution":"558380C0","SetOption":["00028009","2805C8000100068000005AFF000000000000","00000280","00006000","00004000"]},"StatusMEM":{"ProgramSize":613,"Free":388,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"SH-Powermeter1","IPAddress":"192.168.20.228","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.20.1","DNSServer2":"0.0.0.0","Mac":"CC:50:E3:54:8C:B0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.20.253","MqttPort":1883,"MqttClientMask":"SH-Powermeter1","MqttClient":"SH-Powermeter1","MqttUser":"DVES_USER","MqttCount":41,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-12-01T17:34:06","Local":"2021-12-01T18:34:06","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:22","Sunset":"16:55"},"StatusPTH":{"PowerDelta":[80,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2021-12-01T18:34:06","ENERGY":{"TotalStartTime":"2019-03-26T14:43:24","Total":324.998,"Yesterday":0.000,"Today":0.000,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage":233.3,"Current":0.000}},"StatusSTS":{"Time":"2021-12-01T18:34:06","Uptime":"0T00:56:13","UptimeSec":3373,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":41,"POWER":"ON","Wifi":{"AP":1,"SSId":"AndroLAN","BSSId":"44:4E:6D:5F:22:03","Channel":13,"Mode":"11n","RSSI":100,"Signal":-33,"LinkCount":73,"Downtime":"0T00:01:53"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Goto device webpage, try to update firmware 10.0.0 with newest tasmota-minimal.bin on Sonoff POW R2 or Shelly 1.

### EXPECTED BEHAVIOUR
Tasmota minimal can be installed

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-01:
For upgrading use `tasmota-minimal.bin.gz`

closed 2021-12-01 by Jason2866

bigcookie, 2021-12-01:
I used "tasmota-minimal.bin". Or do you ask to directly upload the zipped and not unpacked version?

bigcookie, 2021-12-01:
ok, understood - just tried the zipped version. Didnt know that this would be working. Sorry for the fuzz, I really didnt find this question/Answer while searching...

ISsue can be closed - and many thanks for the quick reply