arendst/Tasmota #13894: RF24_MI_BLE works incorrect with ATC devices with negative temperatures

mrekin, 2021-12-02:
### PROBLEM DESCRIPTION
_RF24_MI_BLE works incorrect with ATC devices with negative temperatures._
Device shows "-1 C"
Tasmota recognize it as "655534 C"

Looks like unsigned char used in
xsns_61_MI_NRF24.ino 
-> MINRFhandleATCPacket(void)
--> MIBLEsensors[_slot].temp = (float)(__builtin_bswap16(_packet->temp))/10.0f;

As I see, problem solved for HM10 
xsns_62_MI_HM10.ino
->HM10parseATC(char * _buf, uint32_t _slot)
--> MIBLEsensors.at(_slot).temp = (float)(__int16_t__(__builtin_bswap16(_packet->temp)))/10.0f;

So, rebuild with the same type conversion (int16_t) solved the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP8266+ NNRF24+
- [ ] Tasmota binary firmware version number used: v9 and v10.0.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_Recognize negative values._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-12-02:
Since you have already the solution, can you please make a PR for? Thx!

arendst, 2021-12-02:
This is a quicky. I'll do it now.

mrekin, 2021-12-02:
Thanks @arendst  @Jason2866 !

closed 2021-12-02 by ascillato2