arendst/Tasmota #13901: Can't connect to mqtt.cloud.yandex.net:8883

arikon, 2021-12-02:
### PROBLEM DESCRIPTION

Sonoff 4CH could not connect to mqtt.cloud.yandex.net:8883. The error is:

```
22:02:10.044 MQT: Attempting connection...
22:02:10.529 MQT: TLS connection error: 49
22:02:10.533 MQT: Connect failed to mqtt.cloud.yandex.net:8883, rc -2. Retry in 10 sec
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4CH ver 2.0
- [x] Tasmota binary firmware version number used: Version 10.0.0.4(tasmota)-STAGE (86628c9f5e5ef2cf5692f47eb45e3902ed9098fe)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:13:21.442 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:13:21.695 RSL: RESULT = {"Module":{"7":"Sonoff 4CH"}}
23:13:21.935 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

23:14:02.920 RSL: STATUS = {"Status":{"Module":7,"DeviceName":"Rigs power module #1","FriendlyName":["Power 1","Power 2","Power 3","Power 4"],"Topic":"tasmota_9CF661","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:14:02.053 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T01:11:57","StartupUTC":"2021-12-02T21:02:05","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2021-12-02T14:11:24","SaveCount":70,"SaveAddress":"F4000"}}
23:14:02.245 RSL: STATUS2 = {"StatusFWR":{"Version":"10.0.0.4(tasmota)","BuildDateTime":"2021-12-03T00:00:53","Boot":6,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"419/699"}}
23:14:02.346 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CRIMEA-MINE","CRIMEA"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A000000000000","00000080","00206000","00004000"]}}
23:14:02.367 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":661,"Free":340,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC5C7","04268001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
23:14:02.558 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9CF661-5729","IPAddress":"192.168.1.58","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:9C:F6:61","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:14:02.658 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.cloud.yandex.net","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9CF661","MqttUser":"are50873d3rhrrf9dra4","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:14:02.670 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-12-02T22:14:02","Local":"2021-12-02T23:14:02","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:23","Sunset":"16:55"}}
23:14:02.801 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-12-02T23:14:02"}}
23:14:02.847 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-12-02T23:14:02","Uptime":"0T01:11:57","UptimeSec":4317,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":2,"SSId":"CRIMEA","BSSId":"10:7B:EF:54:EF:C8","Channel":6,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}
2
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Build with `#define USE_MQTT_TLS` & upload
2. Specify MQTT server host & port: `mqtt.cloud.yandex.net:8883`
3. See logs in console

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Connection ok

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-12-03:
TLS is not included in standard 'tasmota.bin' and requires self compile as stated in the documentation
https://tasmota.github.io/docs/TLS/

Have you followed those instructions?

Alternatively you can use the special build 'Tasmota TLS' from the web flasher at https://tasmota.github.io/install/
Or download the binary from https://github.com/tasmota/install/tree/main/firmware/unofficial

arikon, 2021-12-03:
> Have you followed those instructions?

@barbudor  Yes, I compiled it myself. I mentioned that in the issue.

arikon, 2021-12-03:
> Or download the binary from https://github.com/tasmota/install/tree/main/firmware/unofficial

@barbudor I tried this one with no luck. Error messages are the same.

s-hadinger, 2021-12-03:
Hi,
According to BearSSL, error 49 is `BR_ERR_X509_INDEFINITE_LENGTH`. Anyways, the CA from yandex is not recognized by Tasmota. The only way to connect is to use Fingerprint validation. With the latest development version (from no more than 2 days ago), you can enable Fingerprint with `SetOption132 1`

arikon, 2021-12-03:
> you can enable Fingerprint with `SetOption132 1`

this Helped, thank you!

closed 2021-12-03 by arikon